hoylabs/OpenDTU-OnBattery #1897: [Request] Force browsers to reload web app

schlimmchen, 2025-04-22:
### Is your feature request related to a problem? Please describe.

Web applications incompatible with the firmware keep causing issues.

### Describe the solution you'd like

I know there are ways to make the browser fetch a new copy of the web app. I used this in some application some time ago. The idea is to request the resources with an URL parameter, and the backend will only answer with the respective HTTP response code if the parameter matches the backend firmware version. Something like that... Not sure if that still works, though. Modern browsers might be even more "smart".

schlimmchen, 2025-04-22:
@tbnobody Is this something you explored already? If so, you might have insight regarding the feasibility. If not, I think I want to have a look at this, as telling users to clear their browser cache and such, or even having these problems in the first place, becomes cumbersome.

tbnobody, 2025-04-22:
I don't know why it happens.... the etag header is sent, a full reload is performed after firmware upgrade. I don't know why some browsers seem to ignore it..

schlimmchen, 2025-04-22:
Had to look at the code, as I was unaware that you put something like that in already.

I agree, that should do it. At least because you also put in a Header `"Cache-Control", "public, must-revalidate"`. 

Using what I have in mind is a nightmare... The idea is that rather than `js/app.js`, `js/app.js?v=<git sha>` is requested.

Oooooh, wait! `index.html` often (mostly?) is the same among releases, right? Since all the other resources are merely loaded through `index.html`, the browser might actually be checking for a new version of `index.html`, as it should, but it's the same, so the important resources like `js/app.js` are not actually reloaded. The browser assumes that they are also still valid, as index.html did not change...

Also, assuming that the etag is not only working, but actually used (using it is optional, as far as I can tell), I think there is an easy solution. Let me draft a PR and I will mention you.

tbnobody, 2025-04-22:
Just doublechecked it... Even if I just press F5 (so without clearing any cache) the browser asks for the app.js and gets the right answer (in this case 304)

[image]

the page reload is executed here (imho there is no way to say the browser to ignore the cache)
https://github.com/tbnobody/OpenDTU/blob/f0db2a51daf349fa9ae7919883b6a9ea279b85dc/webapp/src/views/WaitRestartView.vue#L50

schlimmchen, 2025-05-03:
Let's assume #1898 fixed this. We will have to observe this for a couple of months after the next release.

closed 2025-05-03 by schlimmchen

stefan123t, 2025-05-27:
Yes, IMO the HEAD request was introduced together with the WebDAV extensions [RFC-2518](https://datatracker.ietf.org/doc/html/rfc2518) until its latest incarnation as [RFC-4918](https://datatracker.ietf.org/doc/html/rfc4918), but has also seen widespread use-cases with other technologies, like Caching / Proxies, etc.

In principle it allows a client to check whether a Resource / URL and the "data collection" / Folder behind it has changed its (WebDAV) content. Before requesting the complete content it requests only the HTTP `HEAD`ers, which can then be checked by comparing the [`ETag`](https://developer.mozilla.org/de/docs/Web/HTTP/Reference/Headers/ETag) header and the [`If-None-Match`](https://developer.mozilla.org/de/docs/Web/HTTP/Reference/Headers/If-None-Match) Headers on either/both Server and/or Client side before responding with the full Body or requesting only a partial update, e.g. a single changed block from byte X to Y, as e.g. used in Resuming Downloads and/or some Block Filesystems available via HTTP transport (think e.g. Dropbox).

<details><summary>Details of `ETag` and `If-None-Match`</summary>
<p>

The [ESP8266WebServer](https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WebServer/examples/WebServer#etag-support) describes it as follows and last made changes in PR [#8227](https://github.com/esp8266/Arduino/pull/8227):
> When enabled by default the server reads the file content and creates a checksum using the md5 and base64 algorithm her called the ETag value that changes whenever the file contains something different.
> Once a browser has got the content of a file from the server including the ETag information it will add that ETag value is added in the following requests for the same resource. Now the server can answer with a 'use the version from the cache' when the new calculated ETag value is equal to the ETag value in the request.
> The calculation of the ETag value requires some time and processing but sending content is always slower. So when you have the situation that a browser will use a web server multiple times this mechanism saves network and computing and makes web pages more responsive.

Namely the following change was made amongst others to the libraries/ESP8266WebServer/src/detail/[RequestHandlersImpl.h](https://github.com/mathertel/Arduino/blob/master/libraries/ESP8266WebServer/src/detail/RequestHandlersImpl.h#L274-L277)

```c++
            if (server.header("If-None-Match") == eTagCode) {
                server.send(304);
                return true;
            }
...
        if (SRH::_cache_header.length() != 0)
            server.sendHeader("Cache-Control", SRH::_cache_header);

        if ((server._eTagEnabled) && (_eTagCode.length() > 0)) {
            server.sendHeader("ETag", _eTagCode);
        }
```

Here is another eplanation [BCsabaEngine/svelteesp32](https://github.com/BCsabaEngine/svelteesp32) that uses the PsychicHttpServer or ESP32AsyncWebServer to serve Svelte, ReactJS, Vue or Angular apps from an ESP32:

> **ETag**
> The ETag HTTP header can be used to significantly reduce network traffic. If the server sends ETag information, the client can check the integrity of the file by sending back this ETag (in `If-None-Match` header) without sending the data back again. All browsers use this function, the 304 HTTP response code is clearly visible in the network traffic.
> Since microcontroller data traffic is moderately expensive, it is an individual decision whether to use the ETag or not. We recommend using ETag, which adds a bit more code (about 1-3%) but results in a much cleaner operation.
> The use of ETag is not enabled by default, ...
> 
> **Cache-control**
> By default (when using the ETag), we send `no-cache` in the `Cache-Control` header of the HTTP response. Pages, subpages and other elements are downloaded every time. This is perfectly acceptable when serving small pages with ESP.
> At the same time, it can be an advantage that the content is cached by the browser and not even the ETag check is performed. For this, you can specify how many seconds the `max-age` value sent instead of `no-cache` should be. In the case of `--cachetime=86400` (`max-age=86400`), the page (and other elements) will not be downloaded by the browser for one day.

</p>
</details> 

I found the pictures on Werner Rothschopfs page about [Arduino WebServer Caching](https://werner.rothschopf.net/microcontroller/202011_arduino_webserver_caching_en.htm) with ESP8266 very instructive.

So coming back to OpenDTU and the ESP32AsyncWebServer which answers to your HEAD request with a full GET response body.
This would not be necessary and could be spared from the precious CPU cycles of our ESP32-S3 or whatever the code is running on.

Here is a request together with an `If-None-Match` header with the current `ETag` which is returning the `304 Not Modified` Response

```bash
$ curl -v -H 'If-None-Match: "22a34b79871413b73b71840875cadb58"' http://opendtu-fusion/js/app.js --output app.js.gz 
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
* Host opendtu-fusion:80 was resolved.
* IPv6: (none)
* IPv4: 192.168.xxx.xxx
*   Trying 192.168.xxx.xxx:80...
* Connected to opendtu-fusion (192.168.xxx.xxx) port 80
> GET /js/app.js HTTP/1.1
> Host: opendtu-fusion
> User-Agent: curl/8.5.0
[... cut ...]
      return;
    }
``` 

</p>
</details> 

So from my understanding we may still have some work to do in order to resolve the Browser issue.
But first we should get a HAR (HTTP Archive) to analyse the situation when someone experiences this issue again.
So anyone who faces this issue should reload the Live View page using **F12 > Network**, reloading the OpenDTU Homepage e.g. using F5 or **Reload** button and then use the **Download HAR archive** button to attach it to this or either a new github issue.

stefan123t, 2025-05-28:
Also note that we are returning a different `ETag` hash-sum when responding to the `HEAD` request rather than in the case of a `GET` request.

```
$ curl -v -I -H 'If-None-Match: "22a34b79871413b73b71840875cadb58"' http://opendtu-fusion/js/app.js
* Host opendtu-fusion:80 was resolved.
* IPv6: (none)
* IPv4: 192.168.xxx.xxx
*   Trying 192.168.xxx.xxx:80...
* Connected to opendtu-fusion (192.168.xxx.xxx) port 80
> HEAD /js/app.js HTTP/1.1
> Host: opendtu-fusion
> User-Agent: curl/8.5.0
> Accept: */*
> If-None-Match: "22a34b79871413b73b71840875cadb58"
> 
< HTTP/1.1 200 OK
< Content-Encoding: gzip
< Cache-Control: public, must-revalidate
< ETag: "1e878dc7185fcc444783a57f38861dc8"
< connection: close
< accept-ranges: none
< content-length: 392
< content-type: text/html

< 
* Excess found: excess = 392 url = /js/app.js (zero-length body)
* Closing connection
```

Together with the following three caveats from the above depicted explanation by Werner Rothschopf this may be the cause for the oddities we see on some devices / browsers:

* The value of the ETag has to be sent in double quotes.
* Even if you send the ETag with double quotes you can't rely that a browser sends back the value in double quotes. I see several browsers which are NOT returning the ETag in double quotes (for example the OEM Samsung browser) which makes the server comparison more difficult.
* Don't rely on that you will receive the HTTP Header field exactly as "If-None-Match", be prepared to get it in small or capital letters also.

When returning the `ETag` value here `"22a34b79871413b73b71840875cadb58"` with double quotes in the `If-None-Match` header without the double quotes, will deliver the gzipped content again.

```
$ curl -v -H 'If-None-Match: 22a34b79871413b73b71840875cadb58' http://opendtu-fusion/js/app.js --output app.js.gz 
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
* Host opendtu-fusion:80 was resolved.
* IPv6: (none)
* IPv4: 192.168.xxx.xxx
*   Trying 192.168.xxx.xxx:80...
* Connected to opendtu-fusion (192.168.xxx.xxx) port 80
> GET /js/app.js HTTP/1.1
> Host: opendtu-fusion
> User-Agent: curl/8.5.0
> Accept: */*
> If-None-Match: 22a34b79871413b73b71840875cadb58
> 
< HTTP/1.1 200 OK
< Content-Encoding: gzip
< Cache-Control: public, must-revalidate
< ETag: "22a34b79871413b73b71840875cadb58"
< connection: close
< accept-ranges: none
< content-length: 233529
< content-type: text/javascript
< 
{ [1216 bytes data]
100  228k  100  228k    0     0   632k      0 --:--:-- --:--:-- --:--:--  631k
* Closing connection
```

stefan123t, 2025-05-30:
@schlimmchen yes will create a new issue with the following points:

* Static files should have the `Cache-Control: public, max-age=2678400` which would cache for 31 days.
* In case of static files like `/js/app.js.gz` and css files we should keep precalculated md5sums of the files somewhere to make calculation less effort for the ESP32AsyncWebServer. I do not know whether you would already hash it together with the githash of the current OpenDTU-OnBattery firmware version.
* I think about language files, as well as `pin_mapping.json` and `config.json`, which may persist an OpenDTU update and therefor we would need the md5sum of the file to calculate the md5sum combined with the githash.
* `HEAD` should verify the `If-None-Match` request header with the `ETag` for the resource being requested.
* So we will have to make some sanitation, e.g. all lowercase and remove quotes if existing.
   * Even if we send the `ETag` with double quotes we can't rely that every browser sends back the value in double quotes. We have seen several browsers which are NOT returning the `ETag` in double quotes (for example the OEM Samsung browser) which makes the server comparison more difficult.
   * Don't rely on that you will receive the HTTP Header field exactly as `If-None-Match`, be prepared to get it in small or capital letters also.
* `GET` on static files should also verify the `If-None-Match` request header with the `ETag` for the resource.
* In case they match the response should be `HTTP/1.1 304 Not Modified`
```
            if (server.header("If-None-Match") == eTagCode) {
                server.send(304);
                return true;
            }
```