hoylabs/OpenDTU-OnBattery #1901: HMS hängt nach Neustart bei 100% fest, liefert aber das letzte Limit

KarlHeinzz1, 2025-04-23:
### What happened?

Nach dem nächtlichen Neustart kommt der HMS2000 WR (FW 2.0.4) mit 100% zurück in OPENDTU, liefert aber den letzten gesetzten Wert (im Bild 60W). OpenDTU regelt jetzt nicht mehr, da die Steuerung irrtümlich annimmt, dass der WR schon bei 100% ist und nicht mehr leisten kann. Das bleibt dann so lange der Fall, bis man eingreift.
Stellt man kurz selbst einen Wert ein, regelt die Steuerung wieder normal.

Mögliche Lösung:
Nach einem Neustart sollte OpenDTU kurz das ein temporäres Limit übermitteln auch wenn es nicht nötig ist, um einen echten Limit Wert zu erhalten.

### To Reproduce Bug

HMSXXXX auf niedriges Limit (52W) einstellen und neu starten, passiert sporadisch.

### Expected Behavior

Der WR sollte nach dem Neustart wieder normal laufen.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

25.04.17 

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Anything else?

[image]

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

KarlHeinzz1, 2025-04-23:
Habs installiert. Rückmeldung kann aber etwas dauern, passiert nicht immer.
Für mein Verständnis: 
OpenDTU zeigt ja 100%, die nicht stimmen. Wird der Wert vom Wechselrichter mit jeder Aktualisierung neu abgefragt, also alle paar Sekunden? Dann wäre es ein Bug beim WR, den OpenDTU aber leicht fangen könnte.

AndreasBoehm, 2025-04-23:
So einfach ist die Frage nicht zu beantworten.

Da wir den WR mit 'alten' Kommandos ansprechen kommt irgendwann quatsch zurück. Das ist zumindest meine derzeitige Annahme.

KarlHeinzz1, 2025-04-23:
> So einfach ist die Frage nicht zu beantworten.
> 
> Da wir den WR mit 'alten' Kommandos ansprechen kommt irgendwann quatsch zurück. Das ist zumindest meine derzeitige Annahme.

Heute früh war es so: Neustart 0:00, als ich es gesehen habe war es so gegen um 7. 0:00 bis um 7 hin der WR. Also habe ich dann 3% per Hand temporär gesetzt und sofort gings es wieder normal. 
Es müssen also 2 Sachen zusammen kommen: WR ist auf sehr niedrigem Limit und wird neu gestartet. WR meldet nach dem Neustart 100% zurück, ist aber in Wirklichkeit auf den niedrigen Limit.
Für OpenDTU heißt das jetzt richtiger Weise: WR ist schon auf 100% und liefert nicht mehr als die z.B. 60W im Bild. Damit kein Grund ein Kommando zu schicken. Dadurch löst sich das nicht auf, da kein höheres Limit gesetzt werden kann als 100% und kein niedrigeres als die z.b. 60W gebraucht wird.
Davon und den häufigen Neustarts wegen Standby abgesehen, läuft es super.

AndreasBoehm, 2025-04-27:
Bitte nutzt vorerst wieder die Release Version, aktuell sind alle Limits persistent mit der Test Version.

RealNBB, 2025-05-04:
Moin Moin...gilt das nach wie vor auch für die neuesten Testversionen, wie z.B. für die hier?

https://github.com/hoylabs/OpenDTU-OnBattery/actions/runs/14813108427

AndreasBoehm, 2025-05-04:
Nein.
Aber wieso willst du spezielle diese Version testen?
Wenn du mit Test Versionen arbeiten willst dann mit denen vom development branch. Die Versionen der PRs sind nicht immer nützlich oder sinnvoll.

MichaMEG, 2025-07-07:
Wo liegt das Problem bei der HMS Version 2.0.4?
Bis jetzt sieht es bei meinem so aus, als ob er korrekt auf -800 Watt geregelt wird. Wobei ich das auch erst zwei Tage im Einsatz habe und es sehe selten überhaupt so viel Leistung gibt.
Wie kann ich den Fehler bemerken?

MichaMEG, 2025-07-11:
> > Wo liegt das Problem bei der HMS Version 2.0.4? Bis jetzt sieht es bei meinem so aus, als ob er korrekt auf -800 Watt geregelt wird. Wobei ich das auch erst zwei Tage im Einsatz habe und es sehr selten überhaupt so viel Leistung gibt. Außerdem habe ich den HMS-1600-4T mit der Hoymiles DTU auf 88 % permanent eingestellt. Wie kann ich den Fehler bemerken?
> 
> Ich habe auch keine Probleme bemerkt.
> 
> Was aber jetzt besser ist, ohne die power distribution logic ist das ganze System agiler geworden. Mir wurden teils Eingänge mit 50% weniger genutzt als möglich waren. Vermutlich funktioniert das mit Speichern nicht gut. OpenDTU selber macht das besser und verlässlich.

Welche Version benutzt du den jetzt?