hoylabs/OpenDTU-OnBattery #1905: Inverter Total Yield Day not daily resetted

gitisgreat2023, 2025-04-24:
### What happened?

Two inverters connected. With newest release the daily yield of the inverters is not resetted daily.
Before it worked without an issue.
Settings
[image]

[image]

[image]

[image]

### To Reproduce Bug

See settings above.

### Expected Behavior

Resets daily the yield of the inverters.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

	2025.04.17

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

michihuf, 2025-04-24:
Why you don't toggle the second Button in your screenshot?

gitisgreat2023, 2025-04-24:
> Why you don't toggle the second switch in your screenshot?

Because of this:

[image]

That's why I also posted the screenshots that the inverters truly are restarted at 01:00.

gitisgreat2023, 2025-04-29:
Nobody else has this issue?
Can I use zero daily yield at midnight? The inverters are on then... info says only if it unreachable...

schlimmchen, 2025-05-03:
I never had this issue. At least I did not see it. Until now I did not react to this issue as I was confused. You clearly showed that the inverter does restart at 1 AM, as it tells you that. However, that should make sure that the values reported by the inverter are actually reset to zero at the inverter, and fetching it's stats should then show the correct values. You also have "yield day correction" off, which is correct. So this is puzzling.

[image]

gitisgreat2023, 2025-05-03:
> I never had this issue. At least I did not see it. Until now I did not react to this issue as I was confused. You clearly showed that the inverter does restart at 1 AM, as it tells you that. [...]

Yeah it worked flawlessly til this release. Apparently no one has the same issue…

Feuerossi, 2025-05-07:
Same problem Two inverters, Tsun Tsol 800, Hoymiles HM 300. The Tsun with Marstek b2500 is controlled, the Tsol is not. At 00:00, the Tsol is reset to 0, but the Hoymiles is not. Both have the same settings.

gitisgreat2023, 2025-05-07:
Did it worked correctly with the previous release? (I assume you have the latest installed now)

Feuerossi, 2025-05-07:
I can't say, as I've only just observed it. I noticed the difference in the daily yield of the DTU compared to Home Assistant. Can I downgrade to the previous version normally? I'll give it a try.

mgraupner, 2025-05-08:
I'm seeing the same thing, daily production never resets. It started with the Marstek battery being connected to the HMS 1600, before that I only had solar panels connected and it automatically reset at every night. I have just started using OpenDTU-OnBattery, so I have never used a firmware version other than the current one.

These are my settings:

[image]

Feuerossi, 2025-05-08:
You have to activate "Nulle Tagesertrag um Mitternacht". Perhaps than it works for you 🤷🏼‍♂️

spcqike, 2025-05-08:
Wasn’t there an „issue“ with timing? 

Like resetting the daily yield before or after restarting the inverter? 

I don’t have a battery so I can’t test this, but afaik it was something like 

„Reset daily yield at X, inverter still reports Y. So Y is used again. After that restart inverter. So inverter reports 0, but because of this day yield correction and the last known yield was Y, we will just use Y+0 (and summing up till next reset)“ kind of thing. 

So, when I remember correctly one should restart the inverter just before midnight and reset daily yield at midnight. You will have a small error , but it should be nearly 0.

mgraupner, 2025-05-08:
> You have to activate "Nulle Tagesertrag um Mitternacht". Perhaps than it works for you 🤷🏼‍♂️

The info says that this only works when the inverter is inaccessible. Having the Marstek battery connected and using it at night means that the inverter is always accessible. So this should not work, right?

AndreasBoehm, 2025-05-08:
> Having the Marstek battery connected and using it at night means that the inverter is always accessible. So this should not work, right?

The inverter will be forcefully restarted by the DTU at midnight (default setting) or any other time that you have configured in the DPL settings and will be shortly unreachable because of this.

Feuerossi, 2025-05-08:
> The info says that this only works when the inverter is inaccessible. Having the Marstek battery connected and using it at night means that the inverter is always accessible. So this should not [...]

Yes, you are right. I didnt read the context 🙈

Feuerossi, 2025-05-08:
> The inverter will be forcefully restarted by the DTU at midnight (default setting) or any other time that you have configured in the DPL settings and will be shortly unreachable because of this.

Yes, it seems to work, but the DTU didnt realize that. When I restart the DTU shortly after midnight, the Tagesertrag is 0

mgraupner, 2025-05-08:
> The inverter will be forcefully restarted by the DTU at midnight (default setting) or any other time that you have configured in the DPL settings and will be shortly unreachable because of this.

This might be my issue, I'm not using DPL right now (I just started with the battery and use a timed output right now). And there is no other setting to restart the inverter except in the DPL settings.

AndreasBoehm, 2025-05-08:
@mgraupner as a workaround you can enable the DPL and use it to cover the `base load`. This works without a powermeter. Basically it will output a fixed amount of power 24/7. But as a benefit you will get the restart of the inverter :)

gitisgreat2023, 2025-05-09:
@AndreasBoehm which setting do I need to set to get the daily yield reset? Currently all four are inactive (see my first post).

AndreasBoehm, 2025-05-09:
> which setting do I need to set to get the daily yield reset? Currently all four are inactive (see my first post).

I use the following settings (taken from my previous comment)
> Working fine [image]

Feuerossi, 2025-05-09:
But without daily yield correction, the Dtu deletes the yield of the storage from the night, if the storage is empty in the morning and the inverter goes to standby, just had

gitisgreat2023, 2025-05-16:
@schlimmchen @AndreasBoehm I manually restarter Invert 1: daily yield immediately dropped by 2/3. Restarted Inverter 2: voila, daily yield 0. So apparently the inverters were never restarted.

What settings do I have to set to make sure they are daily restarted? The current don't do the job apparently anymore... :-(

schlimmchen, 2025-05-16:
Please select a different time for the automatic inverter restart in the DPL settings and save the settings, then observe.

gitisgreat2023, 2025-05-18:
Changed it from 01:00 to 02:00. Still no reset. In the logs I don't see the inverters restarting anymore...

Feuerossi, 2025-05-18:
I've tried all the settings, but the DTU won't go back to 0 without a reboot. I restart the inverter at 11:58 PM with Home Assistant, and I restart the DTU at 11:59 PM with a smart plug. This is the only way to reset everything to 0.

gretl3000, 2025-05-18:
Thanks for the Idea for this workaround. For me it is ok to restart the inverter with an HA Automation without an additional smart plug. In my case it wasn't a big issue before because HA corrected the faulty statistics in the Energy Dashboard by itself.

Feuerossi, 2025-05-18:
Home Assistant and onbattery are two different systems. You have to tinker to make them work together. 😄

AndreasBoehm, 2025-05-18:
@gretl3000 But why do you care about the daily yield if you use HA? The total yield is the only value that you need for the energy dashboard.

Feuerossi, 2025-05-18:
Yes, but the restart of the inverter is too late 

[image]
Restart at midnight

[image]
Restart at 11:59 pm

AndreasBoehm, 2025-05-18:
Use yield total an its gonna be fine :)

Feuerossi, 2025-05-18:
No, I wanna see both inverter. But this is not an Issue for me. Driving wheels from a VW at a BMW I have to tinker.

AndreasBoehm, 2025-05-18:
Yield total is available for each individual inverter

gretl3000, 2025-05-18:
> [@gretl3000](https://github.com/gretl3000) But why do you care about the daily yield if you use HA? The total yield is the only value that you need for the energy dashboard.

As I said it wasn't really an issue for me. If I use daily or total yield the result in HA is the same. It would just be nice to have the correct data in the Opendtu on Battery site. But nothing really important for me.

gitisgreat2023, 2025-05-18:
@AndreasBoehm in my case the daily yield isn't resetting because the inverters aren't restarted in the night. Manually they restart and the yield goes to zero.

AndreasBoehm, 2025-05-21:
hmm, that is very odd as this works perfectly fine for me with the latest release.
You have all relevant inverters setup as battery- or smart-buffer-powered, right?

Feuerossi, 2025-05-21:
Hallo Andreas
Ich schreibe jetzt mal auf deutsch. Mein 50 Jahre altes Schulenglisch reicht nicht.
Ich nutze 2 Hoymiles WR, einen HM 300 ohne Regelung durch OpenDTU-OnBattery. Einen HM 600, welcher den Marstek B2500 speist. Die Dtu Firmware ist die 4.17. Definitiv hat die Dtu Probleme. Heute war gegen 15 Uhr die Regelung der Nulleinspeisung ausgefallen. Ich habe es 2 Minuten beobachtet, der WR war nicht erreichbar. Die Dtu wollte dem WR 600w entlocken, er war aber nicht erreichbar. Nach manuellem Neustart der Dtu ging es wieder sofort.
Und warum ist bei der Dtu Yield.Day und Yield.Total das gleiche? Ich habs beobachtet, beide geben dieselben Werte des Tages aus.
Ich kann damit leben, weil ich jetzt alle 3 Stunden über HA und Shelly einen Dtu Neustart erzwinge.
Gruß Oppa 👋

spcqike, 2025-05-22:
> Und warum ist bei der Dtu Yield.Day und Yield.Total das gleiche? 

Das sollte nur nach dem aller ersten Start des WEchselrichter, also an seinem aller ersten Tag, der Fall sein.

die Metriken "yield.day" und "yield.total" werden vom Wechselrichter erfragt. "Yield.Day" wird ggf. noch "korrigiert", je nachdem, was eingestellt ist (Stichwort Tagesertragkorrektur, wenn aktiviert). Nach einem Neustart der openDTU ist diese Korrektur aber auch wieder bei "0" und man sieht nur aktuellen "yield.day" Wert.

Feuerossi, 2025-05-22:
Da stimmt was nicht mit den Entitäten. Habe für gestern mal Day und Total nebenher aufzeichnen lassen. Die Werte in der Dtu selbst stimmen 

[image]

AndreasBoehm, 2025-05-22:
> Da stimmt was nicht mit den Entitäten. Habe für gestern mal Day und Total nebenher aufzeichnen lassen. Die Werte in der Dtu selbst stimmen

Für mich sieht das richtig aus. Wo siehst du denn das Problem?

Feuerossi, 2025-05-22:
[image]

Feuerossi, 2025-05-22:
Ok, dann ist das alles wohl richtig. Danke für deine Geduld 😉

AndreasBoehm, 2025-05-22:
Can you share some logs @gitisgreat2023 and your config.json please?

gitisgreat2023, 2025-05-22:
> Can you share some logs [@gitisgreat2023](https://github.com/gitisgreat2023) and your config.json please?

Sure. Which logs should be detailed? DPL? I can set the restart time such that it is included in the log time.

AndreasBoehm, 2025-05-23:
Yes, DPL logs that show the next restart plus what you mentioned that the log timeframe matches the restart time 👍

schlimmchen, 2025-05-30:
This is the relevant part:

```
11:00:03.919 > [DPL] up 101207 s, it is day, next inverter restart at 101262 s (set to 11)
11:00:03.919 > [DPL] battery interface enabled, SoC 88.0 % (ignored), age 0 s (valid)
11:00:03.919 > [DPL] BMS: 53.83 V, MPPT: -1.00 V, inverter 116183773414: 54.10 
11:00:03.919 > [DPL] battery voltage 53.83 V, load-corrected voltage 53.83 V @ 0 W, factor 0.00030 1/A
11:00:03.919 > [DPL] battery discharge allowed, start 49.60 V or 25 %, stop 48.80 V or 20 %
11:00:03.919 > [DPL] start reached, stop NOT reached, solar-passthrough disabled, use at night disabled and dormant
11:00:03.979 > [DPL] total max AC power is 3725 W, conduction losses are 0 %
11:00:03.979 > [DPL] targeting -15 W, base load is 65 W, power meter reads 244.8 W (valid)
11:00:03.979 > [DPL] DC power bus usage blocked by HuaweiCan auto power
11:00:03.979 > [DPL] requesting 0 W from 2 battery-powered inverters currently producing 0 W (diff 0 W, hysteresis 1 W)
11:00:03.979 > [DPL] will cover 0 W using 2 battery-powered inverters
11:00:03.979 > [DPL inverter 138291601012]:
11:00:03.979 >     battery-powered, standing by at 0 W, output included in power meter reading
11:00:03.979 >     lower/current/upper limit: 90/533/2225 W, output capability: 2250 W
11:00:04.026 >     sending commands enabled, reachable, eligible
11:00:04.026 >     max reduction production/standby: 0/0 W, max increase: 2225 W
11:00:04.026 >     target limit/output/state: -1 W (unchanged)/0 W/unchanged, 0 update timeouts
11:00:04.026 >     MPPTs AC power/DC voltage: a: 0 W/54.5 V b: 0 W/53.8 V c: 0 W/54.1 V
11:00:04.026 > [DPL inverter 116183773414]:
11:00:04.026 >     battery-powered, standing by at 0 W, output included in power meter reading
11:00:04.026 >     lower/current/upper limit: 60/285/1500 W, output capability: 1500 W
11:00:04.026 >     sending commands enabled, reachable, eligible
11:00:04.026 >     max reduction production/standby: 0/0 W, max increase: 1500 W
11:00:04.110 >     target limit/output/state: -1 W (unchanged)/0 W/unchanged, 0 update timeouts
11:00:04.110 >     MPPTs AC power/DC voltage: a: 0 W/54.1 V b: 0 W/54.1 V
11:00:04.110 > [PowerMeters::Json::Http] New total: 244.80
11:00:04.110 > Limit Current: 0.000000 A
11:00:04.110 > Power limiter normal operation
11:00:04.110 > [DPL] Localtime read 11:00 / configured RestartHour 11
11:00:04.110 > [DPL] dayMinutes 660 / targetMinutes 660
11:00:04.110 > [DPL] next inverter restart in 1440 minutes
11:00:04.110 > [DPL] next inverter restart @ 187607596 millis
11:00:04.110 > [DPL] reloading DPL configuration
11:00:04.110 > [DPL] waiting for sufficiently recent power meter reading
11:00:04.382 > Success
11:00:04.871 > Fetch inverter: 138291601012
11:00:05.087 > Queue size - NRF: 0 CMT: 1
11:00:05.087 > [ 101208.421] JK BMS: request for data sent
11:00:05.087 > [ 101208.426] JK BMS: busy waiting for or reading a message from the BMS
11:00:05.087 > [ 101208.455] JK BMS: a whole frame was received
11:00:05.087 > [ 101208.466] JK BMS: waiting for poll interval to elapse
11:00:05.087 > [DPL] up 101208 s, it is day, next inverter restart at 187607 s (set to 11)
```

My log looks different:

```
2025-05-26T01:00:26.965196+02:00  [DPL] up 73683 s, it is night, next inverter restart at 73683 s (set to 1)
2025-05-26T01:00:26.966034+02:00  [DPL] battery interface enabled, SoC 0.0 % (ignored), age 2 s (valid)
2025-05-26T01:00:26.966855+02:00  [DPL] BMS: 49.19 V, MPPT: -1.00 V, inverter 116493100759: -1.00 
2025-05-26T01:00:26.967698+02:00  [DPL] battery voltage 49.19 V, load-corrected voltage 49.19 V @ 0 W, factor 0.00040 1/A
2025-05-26T01:00:26.968687+02:00  [DPL] battery discharge restricted, start 52.00 V or 45 %, stop 51.00 V or 40 %
2025-05-26T01:00:26.969740+02:00  [DPL] full solar-passthrough dormant, start 55.04 V or 90 %, stop 54.40 V
2025-05-26T01:00:26.970627+02:00  [DPL] start NOT reached, stop reached, solar-passthrough enabled, use at night enabled and dormant
2025-05-26T01:00:26.971585+02:00  [DPL] total max AC power is 3500 W, conduction losses are 5 %
2025-05-26T01:00:26.972419+02:00  [DPL] targeting -20 W, base load is 400 W, power meter reads 251.0 W (valid)
2025-05-26T01:00:26.973388+02:00  [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 271 W AC
2025-05-26T01:00:26.974269+02:00  [DPL inverter 116493100759]:
2025-05-26T01:00:26.975166+02:00     battery-powered, standing by at 0 W, output included in power meter reading
2025-05-26T01:00:26.975983+02:00     lower/current/upper limit: 60/508/2000 W, output capability: 2000 W
2025-05-26T01:00:26.976772+02:00     sending commands enabled, offline, disqualified (unreachable)
2025-05-26T01:00:26.977684+02:00     max reduction production/standby: 0/0 W, max increase: 0 W
2025-05-26T01:00:26.978579+02:00     target limit/output/state: -1 W (unchanged)/0 W/unchanged, 0 update timeouts
2025-05-26T01:00:26.979621+02:00     MPPTs AC power: a: 0 W b: 0 W c: 0 W d: 0 W
2025-05-26T01:00:26.980442+02:00  [DPL inverter 116183125666]:
2025-05-26T01:00:26.981465+02:00     battery-powered, standing by at 0 W, output included in power meter reading
2025-05-26T01:00:26.982430+02:00     lower/current/upper limit: 52/120/1500 W, output capability: 1500 W
[... 13 lines cut ...]
2025-05-26T01:00:27.211799+02:00  [PowerMeters::Json::Mqtt] Topic 'grid/power/phase1/watts': new value: 54.00, total: 251.00
[... cut ...]
It seems that in your case calcNextInverterRestart() is executed in the context of reloadConfig() rather than in autoRestartInverters(), and autoRestartInverters() has no chance to issue the restart commands as the next restart time already advanced.

Can you give an explanation to why the DPL reloads the config?

I acknowledge that the auto-restart should not be missed even though this race occurs, but it seems to me that somehow you manage to lose this race in all cases, which is quite a complicated thing, so I am wondering how that can happen.

gitisgreat2023, 2025-05-30:
Interesting observation, I missed that it reloaded the config.
Aha... wait, I control with node red this DTU. It's a bit stupid currently, it doesn't check whether the current state is the desired state, that I have to fix. Let me do that and probably it's gone then.
Note though the state is not changed, but apparently it nevertheless reloads.
Now I worry, these changes to the DPL state (via MQQT) are only in RAM, right?

schlimmchen, 2025-06-02:
@gitisgreat2023 Changes to the DPL settings (also via MQTT) are persisted in flash. Always have been. Changing that would be a regression, which is why I am not inclined to change that. Also, one update per minute is not an issue, not even over years. Unless you are bombarding the DPL with updates in the range of a few seconds, you are good.

gitisgreat2023, 2025-06-02:
Thanks! Unfortunately I had an update rate every 5 secs... :-( Though the value didn't change, but probably it wrote the none-change nevertheless I guess...

I now fixed it in Node red, I read the DPL status via "powerlimiter/status/mode" and only if the new setting is different it updates "OpenDTU-onBattery2/powerlimiter/cmd/mode".

With that this issue was actually caused by my Node Red routine updating via MQTT so was a non-issue, accordingly it can be closed

One final question on the Huawei: are the changes to "/huawei/cmd/mode", "huawei/cmd/limit_online_current" and "huawei/cmd/limit_online_voltage" only stored in RAM?

gitisgreat2023, 2025-06-08:
Yeah you're right, it should be robust against it...