hoylabs/OpenDTU-OnBattery #1906: [Request] Huawei Efficiency correction curve

gitisgreat2023, 2025-04-24:
### Is your feature request related to a problem? Please describe.

I measured the actual efficiency vs the reported efficiency of my Huawei R4850G2. The reported efficiency is used to calculate the request charging current. In the example curve below it charges too much at low wattage because the efficiency is lower than assumed.
With two calibration points, for example here number 1 at 54W with -22% number 2 at 230W with 0% it would fix it. In the testing phase we could check which defaults values work okay. @Manos1966 are you able to measure such a curve as well? If the numbers are very similar it would be really an improvement for the Huawei users...

[image]

### Describe the solution you'd like

The Huawei achieves the target value during charging. A stable solution is using a correction curve as proposed.

Solarteur, 2025-04-24:
I use a Huawei R4850G2 to charge a Pylontech US2000C battery. The battery is charged when there is a PV surplus of at least 50W (= minimum power of the R4850G2). The PV surplus is measured with a Shelly 3EM.

If there is more PV surplus, charging is carried out with a higher current (on the AC side), which also increases the output current on the DC side. In this respect, the system is "stable" from my perspective. I realize that the efficiency of the R4850G2 is lower in the low power range than above approximately 250W, but it's better to use it at all than to waste it.

My explanations (hopefully) make it clear that the R4850G2's efficiency isn't necessary in the system I've described. I'm sorry, but I don't see the benefit of a "Huawei Efficiency Correction Curve."

gitisgreat2023, 2025-04-24:
Yes, what you describe, that it isn't reaching its target value, is exactly the issue.
If we have a target value, it should aim for it. It doesn't currently.

Solarteur, 2025-04-24:
I assume we're talking about different "target values." In my system, as described above, the target value is an AC power consumption of the R4850G2 that is just large enough to achieve zero feed-in. The resulting DC power is used to charge the battery. Any further correction would make **this system** unstable.

gitisgreat2023, 2025-04-24:
Okay, we're talking different things.

The target target feed-in is not reached. That is because the algorithm uses the reported efficiency, but the actual efficiency is different, up to about 230W. That can be corrected with the proposed solution.

Manos1966, 2025-04-24:
Hi @gitisgreat2023 

Ich muss auch erwähnen (wie Solarteur bereits gesagt hat), der/die/das (deutsche Sprache, schwere Sprache) HUAWEI R4850G2 ist ein 3.000W Gerät.
Die Treffgenauigkeit bei 50W zu monieren (es ist nicht Mal 2% der R4850G2 Leistung) ist etwas "uebertrieben".
Ich bin eigentlich sehr erfreut zu hoeren, dass bei 230W  (8% Leistung), der R4850G2 so präzis ist 👍 

Ich habe eine Shelly Plus 1PM am 230V Eingang und ein SHUNT am 52V Ausgang der HUWAEI
Was und wie soll ich messen? (genaue Anleitung bitte)

[image]

gitisgreat2023, 2025-04-24:
> Hi [@gitisgreat2023](https://github.com/gitisgreat2023)
>
> Ich muss auch erwähnen (wie Solarteur bereits gesagt hat), der/die/das (deutsche Sprache, schwere Sprache) HUAWEI R4850G2 ist ein 3.000W [...]

Measure: three values have to be measured, 1) AC by Shelly, 2) AC as reported by Huawei, 3) DC as reported by Huawei and optionally DC as reported by the Shunt (I checked, my Huawei is very accurate with the DC power).

In my case it gave following result:
1), 2) and 3) are the first three columns. The fourth column is the efficiency as measured by the Huawei (it should be very close to the Huawei reported efficiency), the 5th is the difference in AC power (not used here), the 6th the "real" efficiency, the 7th contains the difference in this "real efficiency" and the "Huawei efficiency".
[image]

Manos1966, 2025-04-24:
So, here is what I did:

- set the Max output power (lowest allowed setting is 100W)
- waited for the reset of the HUAWEI
- waited for the data to become stable
- reported the numbers  **(My Shelly has a 5W base consumption from the 5V DTU and 12V fans, all Shelly numbers in the table have 5W subtracted)**
[image]

Here the results:
<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File href="file:///C:/WINDOW~1/msohtmlclip1/01/clip.htm">
<link rel=File-List href="file:///C:/WINDOW~1/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

OpenDTU     Eingabe | Shelly     AC | HUAWEI     AC | HUAWEI     DC
-- | -- | -- | --
100 | 108 | 96,22 | 85,5
110 | 120 | 107,75 | 98
120 | 131 | 119,2 | 106,5
130 | 141 | 128 | 117
140 | 153 | 140 | 127,4
150 | 160 | 146,9 | 131,7
160 | 171 | 158,5 | 144,3
170 | 174 | 167,7 | 152,6
180 | 189 | 179,2 |  
200 | 207 | 197,7 | 180
250 | 255 | 249 | 224
300 | 303 | 300 | 272
400 | 397 | 398 | 362

</body>

</html>

gitisgreat2023, 2025-04-25:
@Manos1966 
That was superfast!
Here's the graph:

[image]

Your efficiency is a bit better, but also here with a two point calibration it would work like a charm: 100W -10% and 275W 0% (eyeballing).

Manos1966, 2025-04-25:
Ich musste nach jeder Eingabe, ein paar Minuten warten bis die Daten sich stabilisiert haben.
Ausserdem, bei den niedrigen Einstellungen, konnte ich leichte Schwankungen bemerken in Zusammenhang mit Schwankungen der 230V Spannung passiert sind. Vielleicht sind es Schwankungen der 230V Spannung die das Problem verursachen.

Haben wir die gleiche Board Version?
[image]

Solarteur, 2025-04-25:
Nein, wir haben (natürlich) verschiedene Board Versionen - guter Punkt!

[image]

Bei mir schwankten die Messwerte auch nach einiger Zeit des Wartens noch. Ich habe mich bemüht, die jeweils "zusammengehörenden" Wertepaare (Shelly EM und HUAWEI) in die Tabelle einzutragen.

gitisgreat2023, 2025-06-18:
@Manos1966 @Solarteur 
I discovered that my Huawei is very cool (55°C) at 3kW with fan control at "full". At "auto" it gets 90°C or more and the fan speed doesn't change at all from 100W charging to 3000W charging. Did you both also observe that?
It strikes me that all people complain about high temperatures, so it seems to me the fan control isn't properly working. [Here is a bug-report](https://github.com/hoylabs/OpenDTU-OnBattery/issues/2061), if you have experiences please report them there. A work-around could be to implement some logic to switch from auto to full if the charging power exceeds 2.2kW or so (temperature is a bad idea because it cools so effectively down to low values, so it would switch on and off then).

Solarteur, 2025-06-19:
I cannot contribute to this because my Huawei R4850G2 is operated at a maximum of 1100W (available PV power, only one Pylontech US2000C can be charged, in addition to the settings in the software, appropriate fuses on both the AC and DC sides).

gitisgreat2023, 2025-06-19:
Okay. At lower power it's too hard to distinguish whether the fan ramps up. At 3kW it's very clear.