hoylabs/OpenDTU-OnBattery #1907: Zendure AIO2400 UI Bugs

traxanos, 2025-04-24:
### What happened?

Hi

i have a Zendure AIO2400 and i have some problems/enhancments with UI.

the mppt1 and mppt2 are displayed in the solar input box. sometimes the correct values are displayed here and sometimes the total power is displayed on one of the two and the other value is 0.
as i monitor the values myself via mqtt, i can say that the values have never been delivered incorrectly.
instead, the values are not always delivered and i suspect that some fallback logic takes effect and then delivers the total value instead of waiting for a message with mppt1 and mppt2.

something similar happens in the box above with the battery data. sometimes the battery voltage (which is system voltage and not battery voltage) and sometimes a cell voltage (3.xxx V) are output.

it would also make sense to hide model-dependent fields. the Wh of the batteries are not supplied with the AIO2400. and therefore the SoH is also wrong (is always output with 15).

i have add an image above.

### To Reproduce Bug

have an aio2400 and look at ui

### Expected Behavior

fix ui

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU-OnBattery?

bb55e91

### What firmware variant (PIO Environment)?

generic_esp32_4mb_no_ota

### Relevant log/trace output

```Shell

```

### Anything else?

[image]

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

AndreasBoehm, 2025-04-24:
@vaterlangen

vaterlangen, 2025-04-24:
Thanks for the feedback, I do not have an AOI - so nothing could ne tested from my side. 

First of all, we need to get the battery integrated. Can you please share the first four digits of the battery serial?

Am I correct that AOI has two MPPTs?

traxanos, 2025-04-24:
Korrekt. Die AIO2400 hat 2 MPPTs und einen integrierten Akku  der auch nicht erweiterbar ist.

Der Name des AkkuPack wird als UNKOWN angezeigt. Das würde ich Integrated oder so nennen beim AIO.
Die Seriennummer des AkkuPack begint mit ABB3.
Die Firmwareversionen (AIO und AkkuPack) werden korrekt angezeigt.

Ausblenden würde ich.

Status:
Total capacity	0	Wh
Available capacity	0	Wh
Useable capacity	0	Wh
Remaining discharge time	N/A	
Remaining charge time	N/A	

Akku:
State of Health	1.0	%
Total capacity	0	Wh
Available capacity	0	Wh

Übrigens flappen noch mehr Werte.

Minimal State of Charge
Charge power
Discharge power
Total output power

Vermute ich das richtig das du immer alle Werte im MQTT je Report erwartest. Aktuell sieht es so aus als würde die AIO nicht immer alle Werte senden sondern Bedarfsorientiert.

Da du aus Deutschland bist hab ich mal in Deutsch geantwortet - besser verständlich.

traxanos, 2025-04-24:
Vielleicht noch eine Kleinigkeit. Das Produkt heißt AIO2400 und nicht SolarFlow AIO 2400. Und auch nicht mit dem SolarFlow 2400 AC verwechseln.

traxanos, 2025-04-24:
iot ist ja schreiben. das sind nur die sachen die du sendest. ich hab dir die daten aus  meinem überwachung gelesen. nicht 100% aber es sollte alles drinne stehen was dich interessiert:

```
/yWF7hV/XXX/event/device/data/electric: 0
/yWF7hV/XXX/event/device/data/endTime: 1745520429
/yWF7hV/XXX/event/device/data/eventId: 3
/yWF7hV/XXX/event/device/data/inout: 2
/yWF7hV/XXX/event/device/data/mode: 2
/yWF7hV/XXX/event/device/data/startTime: 1745520242
/yWF7hV/XXX/event/device/deviceId: XXX
/yWF7hV/XXX/event/device/messageId: 123
/yWF7hV/XXX/event/device/method: device
/yWF7hV/XXX/event/device/offData: 0
/yWF7hV/XXX/event/device/timestamp: 1745520429
/yWF7hV/XXX/log/deviceId: XXX
/yWF7hV/XXX/log/log/params/Index (1): 11
/yWF7hV/XXX/log/log/params/Index (2): 1
/yWF7hV/XXX/log/log/params/Index (3): 101
/yWF7hV/XXX/log/log/params/Index (4): 0
/yWF7hV/XXX/log/log/params/Index (5): 0
/yWF7hV/XXX/log/log/params/Index (6): 0
/yWF7hV/XXX/log/log/params/Index (7): 4805
/yWF7hV/XXX/log/log/params/Index (8): 0
/yWF7hV/XXX/log/log/params/Index (9): 0
/yWF7hV/XXX/log/log/params/Index (10): 0
/yWF7hV/XXX/log/log/params/Index (11): -1
/yWF7hV/XXX/log/log/params/Index (12): 0
/yWF7hV/XXX/log/log/params/Index (13): 0
/yWF7hV/XXX/log/log/params/Index (14): 0
/yWF7hV/XXX/log/log/params/Index (15): 320
/yWF7hV/XXX/log/log/params/Index (16): 0
/yWF7hV/XXX/log/log/params/Index (17): 0
/yWF7hV/XXX/log/log/params/Index (18): 0
/yWF7hV/XXX/log/log/params/Index (19): 321
/yWF7hV/XXX/log/log/params/Index (20): 0
/yWF7hV/XXX/log/log/params/Index (21): 0
/yWF7hV/XXX/log/log/params/Index (22): 0
/yWF7hV/XXX/log/log/params/Index (23): 0
/yWF7hV/XXX/log/log/params/Index (24): 0
/yWF7hV/XXX/log/log/params/Index (25): 0
/yWF7hV/XXX/log/log/params/Index (26): 0
/yWF7hV/XXX/log/log/params/Index (27): 0
/yWF7hV/XXX/log/log/params/Index (28): 0
/yWF7hV/XXX/log/log/params/Index (29): 0
/yWF7hV/XXX/log/log/params/Index (30): 0
/yWF7hV/XXX/log/log/params/Index (31): 257
/yWF7hV/XXX/log/log/params/Index (32): 0
/yWF7hV/XXX/log/log/params/Index (33): 0
/yWF7hV/XXX/log/log/params/Index (34): 0
/yWF7hV/XXX/log/log/params/Index (35): 14
/yWF7hV/XXX/log/log/params/Index (36): 0
/yWF7hV/XXX/log/log/params/Index (37): 0
/yWF7hV/XXX/log/log/params/Index (38): 0
/yWF7hV/XXX/log/log/params/Index (39): 481
/yWF7hV/XXX/log/log/params/Index (40): 0
/yWF7hV/XXX/log/log/params/Index (41): 0
/yWF7hV/XXX/log/log/params/Index (42): 0
/yWF7hV/XXX/log/log/params/Index (43): 2
/yWF7hV/XXX/log/log/params/Index (44): 0
/yWF7hV/XXX/log/log/params/Index (45): 0
/yWF7hV/XXX/log/log/params/Index (46): 100000
/yWF7hV/XXX/log/log/params/Index (47): 3
/yWF7hV/XXX/log/log/params/Index (48): 1
/yWF7hV/XXX/log/log/params/Index (49): 0
/yWF7hV/XXX/log/log/params/Index (50): 0
/yWF7hV/XXX/log/log/params/Index (51): 0
/yWF7hV/XXX/log/log/params/Index (52): 0
/yWF7hV/XXX/log/log/params/Index (53): 0
/yWF7hV/XXX/log/log/params/Index (54): 0
/yWF7hV/XXX/log/log/params/Index (55): 0
[... 79 lines cut ...]
/yWF7hV/XXX/properties/report/properties/loraModuleState: 0
/yWF7hV/XXX/properties/report/properties/lowTemperature: 0
/yWF7hV/XXX/properties/report/properties/solarInputPowerCycle2: 0
/yWF7hV/XXX/properties/report/properties/autoModelValue: 0
/yWF7hV/XXX/properties/report/properties/autoRecover: 1
/yWF7hV/XXX/properties/report/properties/heatState: 0
/yWF7hV/XXX/properties/report/properties/localState: 0
/yWF7hV/XXX/properties/report/properties/outputInversePower: 0
/yWF7hV/XXX/properties/report/properties/passMode: 0
/yWF7hV/XXX/properties/report/properties/restState: 0
/yWF7hV/XXX/properties/report/timestamp: 1745520539
/yWF7hV/XXX/properties/report/packData/Index (1)/maxVol: 321
/yWF7hV/XXX/properties/report/packData/Index (1)/power: 0
/yWF7hV/XXX/properties/report/packData/Index (1)/sn: ABB3xxx
/yWF7hV/XXX/properties/report/packData/Index (1)/totalVol: 4807
/yWF7hV/XXX/properties/report/packData/Index (1)/minVol: 320
/yWF7hV/XXX/properties/report/packData/Index (1)/state: 0
/yWF7hV/XXX/properties/report/packData/Index (1)/maxTemp: 2873
/yWF7hV/XXX/properties/report/packData/Index (1)/socLevel: 11
/yWF7hV/XXX/properties/report/packData/Index (1)/softVersion: 32783
/yWF7hV/XXX/properties/report/ip: 192.168.13.70
/yWF7hV/XXX/properties/report/loraAddr: 000000000000
/yWF7hV/XXX/properties/report/mac: 94:c9:60:::
/yWF7hV/XXX/properties/report/wifiName: xxx
```

PR #1908 (2025-05-21): [Zendure] Fixes for AIO 2400
Various fixes for AIO 2400

Fixes #1907 

Tasks:
- [x] Add 'ABB3' as AIO2400 battery
- [x] Remove Solarflow from name
- [x] Check why SOH is always 1
- [x] Verify LOG messages
- [x] Verify REPORT messages
- [x] Fix flapping values
  - [x] SolarPower MPPT1 
  - [x] SolarPower MPPT2
  - [x] Minimal State of Charge
  - [x] Charge power
  - [x] Discharge power
  - [x] Total output power
- [x] Hide SOH if not reported
- [x] Find alternative source for BatteryVoltage
closed 2025-05-21 by AndreasBoehm