hoylabs/OpenDTU-OnBattery #1911: Minor: German word in MPPT VE.Direct verbose log

Squint5471, 2025-04-27:
### What happened?

When using a Victron MPPT via VE.Direct there is a german word "Frame kaputt" in the verbose logging (see log example).

### To Reproduce Bug

- Have broken frames in your VE.Direct communication
- activate MPPT VE.Direct verbose logging
- see console log (e.g., 11:14:14.231 > **Frame _kaputt_**)

### Expected Behavior

Frame kaputt -> Broken frame

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.04.17

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell
11:14:13.852 > [VE.Direct MPPT 10/9] Text Data 'PID' = '0XA058'
11:14:13.852 > [VE.Direct MPPT 10/9] Text Data 'FW' = '164'
11:14:13.852 > [VE.Direct MPPT 10/9] Text Data 'SER' = 'HQ23486MH7H'
11:14:13.852 > [VE.Direct MPPT 10/9] Text Data 'V' = '27580'
11:14:13.852 > [VE.Direct MPPT 10/9] Text Data 'I' = '34900'
11:14:13.852 > [VE.Direct MPPT 10/9] Text Data 'VPV' = '92570'
11:14:13.852 > [VE.Direct MPPT 10/9] Text Data 'PPV' = '986'
11:14:13.852 > [VE.Direct MPPT 10/9] Text Data 'CS' = '3'
11:14:13.852 > [VE.Direct MPPT 10/9] Text Data 'MPPT' = '1'
11:14:13.852 > [VE.Direct MPPT 10/9] Text Data 'OR' = '0X00000000'
11:14:14.231 > [VE.Direct MPPT 10/9] Text Data 'ERR' = '0'
11:14:14.231 > [VE.Direct MPPT 10/9] Text Data 'LOAD' = 'OFF'
11:14:14.231 > [VE.Direct MPPT 10/9] Text Data 'H19' = '53766'
11:14:14.231 > [VE.Direct MPPT 10/9] Text Data 'H20' = '189'
11:14:14.231 > [VE.Direct MPPT 10/9] Text Data 'H21' = '989'
11:14:14.231 > [VE.Direct MPPT 10/9] Text Data 'H22' = '495'
11:14:14.231 > [VE.Direct MPPT 10/9] Text Data 'H23' = '1012'
11:14:14.231 > [VE.Direct MPPT 10/9] Text Data 'HSDS' = '189'
11:14:14.231 > Frame kaputt
```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

schlimmchen, 2025-04-27:
As far as I know "kaputt" is used colloquially to express exactly what a German might expect. So I think this is a nice touch and should stay.

After doing some research, it turns out that the word is usually (but not always) used with a single `t`, but that's splitting hairs.

Also, this is part of the upstream log messages (Hoymiles driver), and not part of VE.Direct.

Let's keep it like this.

closed 2025-04-27 by schlimmchen

Squint5471, 2025-04-27:
Thanks for your fast reply and analysis.
But mixing languages is not really nice imo. Should I open an issue in OpenDTU project then?

schlimmchen, 2025-04-27:
Are you asking me if you can get this changed in OpenDTU-OnBattery if it is changed in the upstream project? Well, yes :wink: We will not maintain this difference. If you want to open an issue upstream, you can do so. However, I am *not* asking you to do so. That's your motivation, then.

schlimmchen, 2025-05-05:
There you go: https://youtu.be/WA26heMOAk0?si=GFy6QB_piZVz_vyu&t=1225 :grin: