hoylabs/OpenDTU-OnBattery #1914: DPL: Disabling full solar passthrough in GUI not working

Squint5471, 2025-04-27:
### What happened?

DPL is enabled and active. The full solar passthrough is set to 98% SOC but is deactivated in the OpenDTU GUI:

[image]

[image]

While checking the MQTT values I saw that it was active (although being disabled in the GUI):
OpenDtuFusion/powerlimiter/status/threshold/soc/full_solar_passthrough = 98
OpenDtuFusion/powerlimiter/status/full_solar_passthrough_active = 1

[image]

### To Reproduce Bug

- DPL is enabled and active
- The full solar passthrough is set to 98% SOC
- deactivate full solar passthrough in the OpenDTU GUI
- check MQTT values

### Expected Behavior

If full solar passthrough is disabled in the GUI it should never be active in MQTT.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.04.17

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell
Log while activating and deactivating solar passthrough in the GUI:
[PowerMeters::Json::Http] New total: -293.27
15:40:15.615 > Websocket: [/console][14] connect
15:40:15.615 > Fetch inverter: [SERIAL]
15:40:16.188 > Queue size - NRF: 0 CMT: 1
15:40:16.188 > [DPL inverter [SERIAL]]: sending limit of 14.3 % (257 W respectively), max output is 1800 W
15:40:16.188 > [DPL] waiting for a start/stop/restart/limit command to complete
15:40:16.188 > Success
15:40:16.806 > [PowerMeters::Json::Http] New total: -294.19
15:40:16.806 > Fetch inverter: [SERIAL]
15:40:17.295 > Queue size - NRF: 1 CMT: 1
15:40:17.295 > [PowerMeters::Json::Http] New total: -296.26
15:40:17.809 > Success
15:40:17.809 > Fetch inverter: [SERIAL]
15:40:18.423 > Queue size - NRF: 0 CMT: 2
15:40:18.423 > Success
15:40:18.839 > [DPL inverter [SERIAL]]: limit update succeeded, actual limit is 14.2 % (256 W respectively), effective 2801 ms after update started, requested were 14.3 %
15:40:18.839 > [DPL] waiting for sufficiently recent inverter data
15:40:18.839 > [PowerMeters::Json::Http] New total: -296.48
15:40:18.839 > Fetch inverter: [SERIAL]
15:40:19.141 > Queue size - NRF: 1 CMT: 1
15:40:19.141 > Middle missing
15:40:19.877 > Request retransmit: 4
15:40:19.877 > [PowerMeters::Json::Http] New total: -296.46
15:40:19.877 > Success
15:40:19.877 > [DPL] waiting for sufficiently recent power meter reading
15:40:19.877 > Success
15:40:19.877 > Fetch inverter: [SERIAL]
15:40:20.227 > Queue size - NRF: 0 CMT: 1
15:40:20.227 > [PowerMeters::Json::Http] New total: -296.83
[... 287 lines cut ...]
15:41:07.654 > [DPL inverter [SERIAL]]: limit update succeeded, actual limit is 8.5 % (153 W respectively), effective 2153 ms after update started, requested were 8.5 %
15:41:07.654 > [DPL inverter [SERIAL]]: limit update succeeded, actual limit is 21.5 % (129 W respectively), effective 2151 ms after update started, requested were 21.5 %
15:41:07.654 > [DPL] waiting for sufficiently recent inverter data
15:41:07.654 > [PowerMeters::Json::Http] New total: -69.71
15:41:08.133 > Fetch inverter: [SERIAL]
15:41:08.133 > Queue size - NRF: 1 CMT: 1
15:41:08.133 > Last missing
15:41:08.393 > Request retransmit: 3
15:41:08.393 > Success
15:41:08.393 > Middle missing
15:41:08.393 > Request retransmit: 1
15:41:08.393 > Success
15:41:08.643 > [DPL] waiting for sufficiently recent power meter reading
15:41:08.643 > [PowerMeters::Json::Http] New total: -65.66
15:41:09.331 > Fetch inverter: [SERIAL]
15:41:09.331 > Queue size - NRF: 0 CMT: 1
15:41:09.331 > Success
```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

schlimmchen, 2025-04-27:
Please enable DPL verbose logging next time. To hunt down bugs, that's actually important.

Have you disabled solar-passthrough in the UI while full solar-passthrough was active? If that's true, I can see the problem. That would only concern the display to MQTT, as full solar-passthrough would not actually be active. You should be able to confirm the latter by the DPL's behavior: Is it choosing to send limits that confirm full solar-passthrough is active, or is it not?

Squint5471, 2025-04-27:
> Have you disabled solar-passthrough in the UI while full solar-passthrough was active? If that's true, I can see the problem. That would only concern the display to MQTT, as full solar-passthrough [...]

Sure, I will provide the verbose logs next time.

Yes, I disabled solar-passthrough in the UI while full solar-passthrough was active.
After doublechecking, the behavior is as follows:
- Passthrough is active and enabled
- Disable passthrough in the GUI
- Passthrough gets deactivated in MPPT/Inverter but NOT in the MQTT topic (OpenDtuFusion/powerlimiter/status/full_solar_passthrough_active)

Thanks for the fast reply and directions. 👍

schlimmchen, 2025-04-27:
Okay, thanks for confirming my suspicion. The quick fix would be simple, but I am not a fan, as I discovered other problems while looking at this. I will address this.

Squint5471, 2025-04-27:
FYI: I just added the logs above.

Your fast response is highly appreciated 🥇

PR #1915 (2025-05-03): Fix: DPL: full solar-passthrough state
Closes #1914.

There is a second commit in this PR which has barely anything to do with #1914, but became quite obvious to me when looking at the code to understand said issue.
closed 2025-05-03 by schlimmchen