hoylabs/OpenDTU-OnBattery #1922: Grid feed after Inverter restart

greymda, 2025-04-29:
### What happened?

Hi all, I have this strange behaviour couple of times now. Not that often but still, maybe someone faced this and can help out with a solution/advice.

After Inverter restart (1:00AM) all my battery capacity is fed into Grid, although it should be not the case as I am not allowed and as I have a shelly and all the settings, usually it's not the case.

But this time, see the screenshot- it fed all into grid...

[image]

### To Reproduce Bug

Set inverter restart to 1AM, after restart all battery is fed into grid.

### Expected Behavior

After inverter restart,  the systems still takes into account the "Target Grid Consumption" setting and not feed all into grid.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2024.06.03

### What firmware variant (PIO Environment)?

generic_esp32

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

Manos1966, 2025-04-29:
2024.06.03  ?

You should probably install a newer version and test.
(which will probably be painful, as you will need to do it via cable (the main reason to upgrade to an ESP32-S3 = be able to do Firmware upgrades OTA)

spcqike, 2025-04-29:
why dont you set a lower persistant limit? like 50W or lower? so, after a restart, the inverter only produces little power.

if the DTU and communication works as expected, the DTU should either stop the inverter or increase its limit (non persistant) afterwards.

Manos1966, 2025-04-29:
> however i was under impression using persistent limit can damage the inverter?

If you write persistent a few thousand times.

greymda, 2025-04-30:
oh, cool! thanks all for input and help, I believe it's best route to go in my particuar case.

Polo2Turbo, 2025-04-30:
Hey.
Same here.
With Fusion Board and Firmware pr1756-202504070621
Its once a day, not exactly the same time. Once at 7pm, other day at 5am.

AndreasBoehm, 2025-04-30:
Please make sure that you use the latest release version and try to catch logs when this happens. Otherwise we can‘t help.

Polo2Turbo, 2025-04-30:
Websocket: [/console][1] connect
22:10:14.003 > [DPL] up 205409 s, it is night, next inverter restart at 212046 s (set to 0)
22:10:14.307 > [DPL] targeting 25 W, base load is 150 W, power meter reads -900.6 W (valid)
22:10:14.307 > [DPL] requesting 0 W from 1 smart-buffer-powered inverter currently producing 0 W (diff 0 W, hysteresis 20 W)
22:10:14.307 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 0 W AC
22:10:14.307 > [DPL inverter 1164a00aa3bc]:
22:10:14.307 >     solar-powered, standing by at 0 W, output included in power meter reading
22:10:14.307 >     lower/current/upper limit: 52/1600/1600 W, output capability: 1600 W
22:10:14.307 >     sending commands enabled, offline, disqualified (nighttime)
22:10:14.403 >     max reduction production/standby: 0/0 W, max increase: 0 W
22:10:14.403 >     target limit/output/state: -1 W (unchanged)/0 W/unchanged, 1 update timeouts
22:10:14.403 >     MPPTs AC power/DC voltage: a: 0 W/13.1 V b: 0 W/13.1 V c: 0 W/13.2 V d: 0 W/13.2 V
22:10:14.403 > [DPL inverter 1164a00b2811]:
22:10:14.403 >     smart-buffer-powered, standing by at 0 W, output included in power meter reading
22:10:14.403 >     lower/current/upper limit: 52/1600/1600 W, output capability: 1600 W
22:10:14.403 >     sending commands enabled, reachable, eligible
22:10:14.403 >     max reduction production/standby: 0/0 W, max increase: 1600 W
22:10:14.421 >     target limit/output/state: -1 W (unchanged)/0 W/unchanged, 0 update timeouts
22:10:14.421 >     MPPTs AC power/DC voltage: a: 0 W/46.8 V b: 0 W/46.7 V c: 0 W/46.7 V d: 0 W/46.7 V
22:10:14.421 > [DPL inverter 1164a00b1923]:
22:10:14.421 >     smart-buffer-powered, producing 1503 W, output included in power meter reading
22:10:14.421 >     lower/current/upper limit: 52/1483/1600 W, output capability: 1600 W
22:10:14.421 >     sending commands enabled, offline, disqualified (unreachable)
22:10:14.421 >     max reduction production/standby: 0/0 W, max increase: 0 W
22:10:14.421 >     target limit/output/state: -1 W (unchanged)/1503 W/unchanged, 3 update timeouts
22:10:14.818 >     MPPTs AC power/DC voltage: a: 377 W/45.8 V b: 377 W/45.8 V c: 371 W/44.9 V d: 377 W/45.0 V
22:10:14.818 > Fetch inverter: 1164A00B1923
22:10:14.926 > Request SystemConfigPara
22:10:14.927 > Resend ActivePowerControl
22:10:14.927 > Queue size - NRF: 0 CMT: 4
22:10:14.927 > TX ChannelChangeCommand 868.00 MHz --> 56 A0 0B 19 23 80 14 18 20 02 15 21 14 14 5D 
22:10:14.927 > RX Period End
22:10:14.927 > All missing
22:10:14.927 > Nothing received, resend whole request
22:10:14.927 > TX ChannelChangeCommand 868.00 MHz --> 56 A0 0B 19 23 80 14 18 20 02 15 21 14 14 5D 
22:10:14.927 > RX Period End
22:10:14.927 > All missing
22:10:14.927 > Nothing received, resend whole request
22:10:14.927 > TX ChannelChangeCommand 868.00 MHz --> 56 A0 0B 19 23 80 14 18 20 02 15 21 14 14 5D 
22:10:14.927 > RX Period End
22:10:14.927 > All missing
22:10:14.927 > Nothing received, resend whole request
22:10:14.963 > TX ChannelChangeCommand 868.00 MHz --> 56 A0 0B 19 23 80 14 18 20 02 15 21 14 14 5D 
22:10:14.964 > RX Period End
22:10:14.964 > All missing
22:10:14.964 > Nothing received, resend whole request
22:10:14.964 > TX ChannelChangeCommand 868.00 MHz --> 56 A0 0B 19 23 80 14 18 20 02 15 21 14 14 5D 
22:10:14.964 > [DPL] up 205410 s, it is night, next inverter restart at 212046 s (set to 0)
22:10:14.964 > [DPL] targeting 25 W, base load is 150 W, power meter reads -900.6 W (valid)
22:10:14.964 > [DPL] requesting 0 W from 1 smart-buffer-powered inverter currently producing 0 W (diff 0 W, hysteresis 20 W)
22:10:14.964 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 0 W AC
22:10:14.977 > [DPL inverter 1164a00aa3bc]:
22:10:14.977 >     solar-powered, standing by at 0 W, output included in power meter reading
22:10:14.977 >     lower/current/upper limit: 52/1600/1600 W, output capability: 1600 W
22:10:14.977 >     sending commands enabled, offline, disqualified (nighttime)
22:10:14.977 >     max reduction production/standby: 0/0 W, max increase: 0 W
22:10:14.977 >     target limit/output/state: -1 W (unchanged)/0 W/unchanged, 1 update timeouts
22:10:14.977 >     MPPTs AC power/DC voltage: a: 0 W/13.1 V b: 0 W/13.1 V c: 0 W/13.2 V d: 0 W/13.2 V
22:10:14.977 > [DPL inverter 1164a00b2811]:
22:10:14.995 >     smart-buffer-powered, standing by at 0 W, output included in power meter reading
22:10:14.996 >     lower/current/upper limit: 52/1600/1600 W, output capability: 1600 W
22:10:14.996 >     sending commands enabled, reachable, eligible
22:10:14.996 >     max reduction production/standby: 0/0 W, max increase: 1600 W
22:10:14.996 >     target limit/output/state: -1 W (unchanged)/0 W/unchanged, 0 update timeouts
22:10:14.996 >     MPPTs AC power/DC voltage: a: 0 W/46.8 V b: 0 W/46.7 V c: 0 W/46.7 V d: 0 W/46.7 V
22:10:14.996 > [DPL inverter 1164a00b1923]:
22:10:14.996 >     smart-buffer-powered, producing 1503 W, output included in power meter reading
22:10:15.012 >     lower/current/upper limit: 52/1483/1600 W, output capability: 1600 W
22:10:15.012 >     sending commands enabled, offline, disqualified (unreachable)
22:10:15.012 >     max reduction production/standby: 0/0 W, max increase: 0 W
[... 719 lines cut ...]
22:10:38.783 > [PowerMeters::Json::Http] New total: -1131.59
22:10:38.783 > [DPL] up 205434 s, it is night, next inverter restart at 212046 s (set to 0)
[... cut ...]
22:10:38.914 >     MPPTs AC power/DC voltage: a: 0 W/46.8 V b: 0 W/46.7 V c: 0 W/46.7 V d: 0 W/46.7 V
22:10:38.914 > [DPL inverter 1164a00b1923]:
22:10:38.914 >     smart-buffer-powered, producing 1503 W, output included in power meter reading
22:10:38.914 >     lower/current/upper limit: 52/1483/1600 W, output capability: 1600 W
22:10:38.914 >     sending commands enabled, offline, disqualified (unreachable)
22:10:38.914 >     max reduction production/standby: 0/0 W, max increase: 0 W
22:10:38.914 >     target limit/output/state: -1 W (unchanged)/1503 W/unchanged, 3 update timeouts

AndreasBoehm, 2025-04-30:
Das ist ein Kommunikations Problem zwischen WR und DTU. (Anzeige Rot im Screenshot) Deshalb können wir das Limit des WR nicht reduzieren.

Polo2Turbo, 2025-05-01:
Hallo. Das passiert mal immer so einmal am Tag. Woran kann das liegen und hat das noch wer ausser ich?

schlimmchen, 2025-05-03:
@Polo2Turbo Da kann der DPL leider nichts für dich tun, wenn die Kommunikaiton so schlecht ist. Leider ist das ein schwieriges Thema und wurde schon von vielen diskutiert. Entweder es klappt ganz gut, oder die Lösungen sind sehr kreativ, z.B. ne Schlauchschelle um die Antenne. Ist dieser WR denn besonders nah an der DTU? Dann musst du vielleicht die RF-Leistung reduzieren, oder Dämpfungsglieder an die Antenne einbauen. Ist er eher weiter weg, musst du die RF-Leistung vielleicht erhöhen. Vielleicht musst du auch die Spannugsquelle deines Boards überprüfen. Oder die Antennenausrichtung prüfen. Oder die Frequenz wechseln (CMT2300A). Bitte probiere aus. Falls du weiterhin Probleme hast, starte bitte eine *neue* **Diskussion**.

@greymda The requirement to set a low persistent limit manually is [documented](https://opendtu-onbattery.net/firmware/configuration/dpl/#automatic-restart-time). Hard to find, I agree, but it's there. Please set a low persistent limit. Once you did, I also insist that you try a newer firmware. I don't see a reason why the problem you describe should be an issue. I certainly don't experience it. If you see it again using the current firmware release, please provide a DPL verbose log while the issue presents itself, and then I am happy to dig into the code and try to find the root cause. Until then, I close this issue. Please comment again (or open a new issue) if the problem persists.

closed 2025-05-03 by schlimmchen