hoylabs/OpenDTU-OnBattery #1925: Request: Größere Trägheit beim herunterregeln für schnelle repetitive Stromverbräuche wie z.B. Waschmaschine

Mart-Muc, 2025-04-30:
### Is your feature request related to a problem? Please describe.

Hallo,
Ich bin letzte Woche von ahoy auf diese Version umgestiegen....und bin bislang begeistert.

Heute Wasche ich zum ersten Mal und die Nulleinspeisung reagiert beim herunterregeln "zu schnell", sodass sie eigentlich sofort wieder hochregeln muss für das nächste "Drehereignis".

Deswegen würde ich mir hier eine unsymetrische Reaktionsgeschwindigkeit wünschen.
Hochregeln soll die Logik natürlich as fast as possible, aber beim herunterregeln sollte es einen Variable gebe, damit die DTU z.B. x minuten den höheren Stand beibehält und träger reagiert. (Vermeidung der blauen Spitzen)

Macht natürlich nur Sinn, wenn man es wie ich (noch) ohne Batterie betreibt. ( Ich habe das -OnBattery im Titel schon gelesen, wäre aber ein einfach zu implementierend Feature für eine Optimierung der Nulleinspeisung)

Danke für jegliche Kommentare:-)
Mart

[image]

### Describe the solution you'd like

Zusätzlich zum Hysteresewert in W noch ein Intervall in Sekunden für den Abfall

spcqike, 2025-04-30:
so etwas wurde schon mal angefragt und besprochen. ich finde es leider auf die Schnelle nicht.

@schlimmchen da war mal was, oder? im Sinne von im Sommer eher verzögert runterregeln, im Winter eher verzögert hoch. so dass, je nach Jahreszeit, mehr PV Strom genutzt werden kann (sommer) bzw. weniger durch kurze spitzen als Überschuss verschenkt wird (winter)

Mart-Muc, 2025-04-30:
genau das wäre der Idealfall :-))

Oh- ich sehe das wurde schon als slope mode in Pull request Surplus-Power-Mode #1167 erwähnt.
Zudem viel besser beschrieben, als ich in der Lage war....

Da hat sich also schon jemand Gedanken gemacht und den größten Teil gecodet:-)
Ich würde begrüßen, wenn das in der UI und auch unabhängig von Batterie aktivierbar wäre

Mart-Muc, 2025-04-30:
Danke:_)
Genau das war mein Optimierungsvorschlag :-))

Tolle Arbeit