hoylabs/OpenDTU-OnBattery #1928: Inverter Gesamtleistung passt nicht

hex2647, 2025-05-01:
### What happened?

Hallo zusammen, 
die Gesamtleistung des Inverters passt leider nicht wenn die Eingangsspannung bei nur 16V liegt... er scheint dann irgendwie nur einen Eingang anstatt beide zu werten... Ist das Problem schon bekannt?

[image]

### To Reproduce Bug

Eingangsspannung bei 16V mit Growatt Noah 2000

### Expected Behavior

richtige Ausgabe der Leistung

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

	2025.04.17

### What firmware variant (PIO Environment)?

generic_esp32_8mb

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

AndreasBoehm, 2025-05-02:
Die Gesamtleistung ist die AC Leistung des WR, an den Eingängen wird die DC Leistung gemessen.

Problem bei dieser extrem niedrigen Eingangsspannung ist das die Effizienz dann in den Keller geht, laut deinem Screenshot nur ca 47%.

hex2647, 2025-05-02:
Heißt der Report des WR ist an der Stelle falsch? Weil tatsächlich ist die Effizienz nicht so schlecht… hab da noch ein Shelly dran hängen der hat 47W Reportet.

AndreasBoehm, 2025-05-02:
Es gibt User mit 24V Akkus, da ist die Anzeige des WRs auch totaler quatsch

spcqike, 2025-05-02:
Ja. Bei solch niedrigen Spannungen kommen die WR ab und an durcheinander. Da hilft ein Neustart von WR für ne gewisse Zeit.

closed 2025-05-03 by schlimmchen