hoylabs/OpenDTU-OnBattery #1930: HM-1500 first reachable again after reboot

gitisgreat2023, 2025-05-02:
### What happened?

I just noticed my HM-1500 wasn't reachable after 48hr... tried to tune the only knob available, transmittance power. Didn't do anything. As a last resort I rebooted, that worked as a charm, it was immediately available again.

Minimum power limit is 60W. "put into standby" the info tip says, could this cause the observed behavior?
Would it be possible to avoid a reboot and reinitialize (or whatever happens with the reboot?

### To Reproduce Bug

Huawei R4850G2, JK BMS transceiver on MCU, HMT-2250 and HM-1500. It happened after 3 days of uptime.
I didn't observe this before (for example with 2025.02.10).

### Expected Behavior

If it loses the connection to the HM-1500, it re-establishes like it does apparently with a reboot.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.04.17

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

AndreasBoehm, 2025-05-05:
What did you reboot? The DTU or the HM-1500?
You mentioned a HMT-2250 in the ticket as well, is it connected to the same DTU? Was it reachable while the HM-1500 was unreachable?

schlimmchen, 2025-05-06:
Well, yeah, in particular the SPI subsystem and the RF chip(s) are initialized. AFAIK there is no formal reset/re-initialization unless triggered by an event like saving the RF settings.

gitisgreat2023, 2025-05-06:
Okay... would it be an idea to formally reset/re-initialize after a certain time-out has been reached (say 5 minutes or so), and a log line is written in the inverters event log? It ran 48 hours in my case before I discovered it... rebooted the DTU and it worked immediately.

wiesi667, 2025-05-13:
My HM-800 shows the same behavior. At some point it can no longer be reached. A reboot of the DTU solves the problem immediately.

closed 2025-06-16 by github-actions[bot]