hoylabs/OpenDTU-OnBattery #1932: [Request] Publish Inverter Log to HomeAssistant

Dani1802, 2025-05-02:
### Is your feature request related to a problem? Please describe.

Would it be possible to make the event log of the inverter available in HomeAssisant?
Or at least the most recent entry?
That would be a good way to be actively informed of unexpected events.

### Describe the solution you'd like

Inverter Log is available in HA

schlimmchen, 2025-05-03:
This is more of an upstream feature. We will only merge a respective feature implementation if we are confident that the change will also be included upstream.

Dani1802, 2025-05-03:
> This is more of an upstream feature. We will only merge a respective feature implementation if we are confident that the change will also be included upstream.

What do you mean with that?
Should I open this feature request here -> https://github.com/tbnobody/OpenDTU   ?

schlimmchen, 2025-05-03:
Preferrably, yes.

Dani1802, 2025-05-03:
If anyone else is interested in this feature, please comment/like/vote here  😄 -> https://github.com/tbnobody/OpenDTU/issues/2656

closed 2025-06-25 by AndreasBoehm