hoylabs/OpenDTU-OnBattery #1940: DPL not shutting down inverter in the mornings with Use Battery At Night Even If Only Partially Charged set

wusel42, 2025-05-05:
### What happened?

I have one HMS-1600 connected to a battery, one HMS-800 to solar. The battery is charged via SmartSolar controllers (one of them is connected to another OnBattery-instance that publishes NetworkTotalDcInputPower via MQTT, read by this instance as "Ausgangsleistung Topic").  I have "Batterie nachts sogar teilweise geladen nutzen" active; according to the documentation, discharging should stop in the morning:

> Discharging the battery always ends when reaching the stop threshold. If a discharge cycle was previously started because of this option being enabled, discharging also ends roughly at sunrise, even if the battery still stores a significant amount of energy.

But it does not. Log states the change from night to day at 05:42, but the HMS-1600 isn't shut down even at 06:46, I had to disable DPL to stop discharging/allow charging the battery. This happens every day so far, this time I made sure to have logging enabled.

### To Reproduce Bug

During daytime, enable DPL, enable "Use Battery At Night Even If Only Partially Charged", save. Wait until sunset, DPL should start an inverter and discharge the battery even if not fully changed. Wait until sunrise, DPL does not stop the inverter.

### Expected Behavior

DPL should stop the inverter at/around sunrise as documented.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.04.17

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell
May  5 05:41:56 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] up 231468 s, it is night, next inverter restart at 296942 s (set to 0)
May  5 05:41:56 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] battery interface enabled, SoC 8.4 % (ignored), age 1 s (valid)
May  5 05:41:56 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] BMS: 50.84 V, MPPT: 50.85 V, inverter 11xxxxxxxx60: 50.70 
May  5 05:41:56 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] battery voltage 50.84 V, load-corrected voltage 51.38 V @ 540 W, factor 0.00100 1/A
May  5 05:41:56 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] battery discharge allowed, start 54.80 V or 80 %, stop 50.80 V or 20 %
May  5 05:41:56 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] start NOT reached, stop NOT reached, solar-passthrough disabled, use at night enabled and dormant
May  5 05:41:56 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] total max AC power is 2400 W, conduction losses are 5 %
May  5 05:41:56 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] targeting 400 W, base load is 900 W, power meter reads 449.1 W (valid)
May  5 05:41:56 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] granting 589 W from DC power bus (no battery discharge limit), solar power is 0/0 W DC/AC
May  5 05:41:56 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] requesting 589 W from 1 battery-powered inverter currently producing 540 W (diff 49 W, hysteresis 20 W)
May  5 05:41:56 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] will cover 593 W using 1 battery-powered inverter
May  5 05:41:56 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL inverter 11xxxxxxxx60]:
May  5 05:41:56 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     battery-powered, producing 540 W, output included in power meter reading
May  5 05:41:56 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     lower/current/upper limit: 80/544/1200 W, output capability: 1600 W
May  5 05:41:56 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     sending commands enabled, reachable, eligible
May  5 05:41:56 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     max reduction production/standby: 460/540 W, max increase: 656 W
May  5 05:41:56 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     target limit/output/state: 593 W (update)/593 W/production, 0 update timeouts
May  5 05:41:56 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     MPPTs AC power/DC voltage: a: 137 W/50.7 V b: 135 W/50.5 V c: 135 W/50.6 V d: 133 W/50.7 V
May  5 05:41:56 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL inverter 11xxxxxxxx97]:
May  5 05:41:56 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     solar-powered, standing by at 0 W, output included in power meter reading
May  5 05:41:56 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     lower/current/upper limit: 50/49/800 W, output capability: 800 W
May  5 05:41:56 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     sending commands enabled, offline, disqualified (nighttime)
May  5 05:41:56 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     max reduction production/standby: 0/0 W, max increase: 0 W
May  5 05:41:56 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     target limit/output/state: -1 W (unchanged)/0 W/unchanged, 0 update timeouts
May  5 05:41:56 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     MPPTs AC power/DC voltage: a: 0 W/29.8 V b: 0 W/29.9 V
May  5 05:41:56 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL inverter 11xxxxxxxx60]: sending limit of 37.1 % (593 W respectively), max output is 1600 W
[…]
May  5 05:42:28 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] up 231499 s, it is day, next inverter restart at 296942 s (set to 0)
May  5 05:42:28 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] battery interface enabled, SoC 8.4 % (ignored), age 1 s (valid)
May  5 05:42:28 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] BMS: 50.82 V, MPPT: 50.81 V, inverter 11xxxxxxxx60: 50.70 
May  5 05:42:28 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] battery voltage 50.82 V, load-corrected voltage 51.42 V @ 595 W, factor 0.00100 1/A
May  5 05:42:28 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [PowerMeters::Json::Mqtt] Topic 'shellypro3em63-01/status/em:0': new value: 11.60, total: 354.70
May  5 05:42:28 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] battery discharge allowed, start 54.80 V or 80 %, stop 50.80 V or 20 %
May  5 05:42:28 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] start NOT reached, stop NOT reached, solar-passthrough disabled, use at night enabled and dormant
May  5 05:42:28 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] total max AC power is 2400 W, conduction losses are 5 %
May  5 05:42:28 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] targeting 400 W, base load is 900 W, power meter reads 354.7 W (valid)
May  5 05:42:28 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] granting 550 W from DC power bus (no battery discharge limit), solar power is 0/0 W DC/AC
May  5 05:42:28 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] requesting 550 W from 1 battery-powered inverter currently producing 595 W (diff -45 W, hysteresis 20 W)
May  5 05:42:28 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] will cover 547 W using 1 battery-powered inverter
[... 75 lines cut ...]
May  5 16:28:40 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] BMS: 53.65 V, MPPT: 53.66 V, inverter 11xxxxxxxx60: 53.90 
May  5 16:28:40 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] battery voltage 53.65 V, load-corrected voltage 53.65 V @ 0 W, factor 0.00100 1/A
May  5 16:28:40 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] battery discharge allowed, start 54.80 V or 80 %, stop 50.80 V or 20 %
May  5 16:28:40 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] start NOT reached, stop NOT reached, solar-passthrough disabled, use at night enabled and dormant
May  5 16:28:40 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] total max AC power is 2400 W, conduction losses are 5 %
May  5 16:28:40 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] targeting 400 W, base load is 900 W, power meter reads 853.1 W (valid)
May  5 16:28:40 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] granting 453 W from DC power bus (no battery discharge limit), solar power is 0/0 W DC/AC
May  5 16:28:40 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] requesting 453 W from 1 battery-powered inverter currently producing 0 W (diff 453 W, hysteresis 20 W)
May  5 16:28:40 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] will cover 453 W using 1 battery-powered inverter
May  5 16:28:40 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL inverter 11xxxxxxxx60]:
[... cut ...]
May  5 16:28:40 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL inverter 11xxxxxxxx60]: sending limit of 28.3 % (453 W respectively), max output is 1600 W
```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

AndreasBoehm, 2025-05-05:
Is it possible that your battery reached the start threshold and was discharged because of that and not because of the `Use Battery At Night Even If Only Partially Charged` option?

Asking because this log line looks like this is what happened
> May  5 05:41:56 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] start NOT reached, stop NOT reached, solar-passthrough disabled, use at night enabled and dormant

If this is the case, discharging will not stop at sunrise but when the stop threshold has been reached.

The stop at sunrise only happens when discharging started because of `Use Battery at...` but start threshold was not reached. This behaviour also matches the documentation.

It its complicated and the difference might not make too much sense to be honest.
But in general i would say that if there is energy in the battery why not use it until there is enough solar power to cover the consumption?

AndreasBoehm, 2025-05-05:
What i said above applies to all the 'issues' that you described. The battery is in a discharge cycle because the start threshold was reached and will only stop when the stop threshold is reached.

wusel42, 2025-05-05:
> Is it possible that your battery reached the start threshold and was discharged because of that and not because of the `Use Battery At Night Even If Only Partially Charged` option?

Indeed, yesterday it reached 100% SoC and OpenDTUOnBattery started the inverter. And as you mention it, the other day I wanted to start discharging before dawn and lowered the start threshold. Will ensure to have it in "discharged" mode tonight. This should do?

```
May  5 18:17:33 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] up 276806 s, it is day, next inverter restart at 296999 s (set to 0)
May  5 18:17:33 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] battery interface enabled, SoC 69.4 % (ignored), age 1 s (valid)
May  5 18:17:33 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] BMS: 53.58 V, MPPT: 53.59 V, inverter 11xxxxxxx60: 53.80 
May  5 18:17:33 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] battery voltage 53.58 V, load-corrected voltage 53.58 V @ 0 W, factor 0.00100 1/A
May  5 18:17:33 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] battery discharge restricted, start 54.80 V or 80 %, stop 54.30 V or 20 %
May  5 18:17:33 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] start NOT reached, stop reached, solar-passthrough disabled, use at night enabled and dormant
May  5 18:17:33 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] total max AC power is 2400 W, conduction losses are 5 %
May  5 18:17:33 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] targeting 400 W, base load is 900 W, power meter reads 842.5 W (valid)
May  5 18:17:33 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 443 W AC
May  5 18:17:33 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] requesting 0 W from 1 battery-powered inverter currently producing 0 W (diff 0 W, hysteresis 20 W)
May  5 18:17:33 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] will cover 0 W using 1 battery-powered inverter
May  5 18:17:33 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL inverter 11xxxxxxx60]:
May  5 18:17:33 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     battery-powered, standing by at 0 W, output included in power meter reading
May  5 18:17:33 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     lower/current/upper limit: 80/395/1200 W, output capability: 1600 W
May  5 18:17:33 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     sending commands enabled, reachable, eligible
May  5 18:17:33 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     max reduction production/standby: 0/0 W, max increase: 1200 W
May  5 18:17:33 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     target limit/output/state: -1 W (unchanged)/0 W/unchanged, 0 update timeouts
May  5 18:17:33 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     MPPTs AC power/DC voltage: a: 0 W/53.8 V b: 0 W/53.7 V c: 0 W/53.7 V d: 0 W/53.9 V
[…]
May  5 18:17:40 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] up 276813 s, it is day, next inverter restart at 296968 s (set to 0)
May  5 18:17:40 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] battery interface enabled, SoC 69.4 % (ignored), age 1 s (valid)
May  5 18:17:40 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] BMS: 53.58 V, MPPT: 53.59 V, inverter 11xxxxxxx60: 53.90 
May  5 18:17:40 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] battery voltage 53.58 V, load-corrected voltage 53.58 V @ 0 W, factor 0.00100 1/A
May  5 18:17:40 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] battery discharge restricted, start 54.80 V or 80 %, stop 50.80 V or 20 %
May  5 18:17:40 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] start NOT reached, stop NOT reached, solar-passthrough disabled, use at night enabled and dormant
May  5 18:17:40 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] total max AC power is 2400 W, conduction losses are 5 %
May  5 18:17:40 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] targeting 400 W, base load is 900 W, power meter reads 861.5 W (valid)
May  5 18:17:40 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 462 W AC
May  5 18:17:40 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] requesting 0 W from 1 battery-powered inverter currently producing 0 W (diff 0 W, hysteresis 20 W)
May  5 18:17:40 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] will cover 0 W using 1 battery-powered inverter
May  5 18:17:40 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL inverter 11xxxxxxx60]:
May  5 18:17:40 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     battery-powered, standing by at 0 W, output included in power meter reading
May  5 18:17:40 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     lower/current/upper limit: 80/395/1200 W, output capability: 1600 W
May  5 18:17:40 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     sending commands enabled, reachable, eligible
May  5 18:17:40 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     max reduction production/standby: 0/0 W, max increase: 1200 W
May  5 18:17:40 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     target limit/output/state: -1 W (unchanged)/0 W/unchanged, 0 update timeouts
[... cut ...] rial]/V` at sunset and have DPL's `Use Battery …` feature take care until sunrise to achieve my desired scenario? Basically I'll have to align OnBattery's view of charge/discharge cycles with my goal of reducing Grid consumption based on per hour (starting June: per 15 minute) kWh prices?

Re-reading the Solar Passthrough documentation, `Full Solar-Passthrough` should take care of my goal to push excessive – battery is full – solar power to the Grid? (Is it just me who finds the name confusing?)

AndreasBoehm, 2025-05-05:
I would recommend to set the DPL Mode via MQTT instead of fiddling with other values: https://opendtu-onbattery.net/firmware/mqtt_topics/#mode

This way you can use a external schedule for the DPL.

wusel42, 2025-05-06:
> The battery is in a discharge cycle because the start threshold was reached and will only stop when the stop threshold is reached.

Yes, that was my issue; after tricking OnBattery to stop the discharge cycle, ```Use Battery …``` works as expected. Sorry for this misunderstanding.
```
May  5 20:50:09 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] start NOT reached, stop NOT reached, solar-passthrough disabled, use at night enabled and active
May  6 05:40:21 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] start NOT reached, stop NOT reached, solar-passthrough disabled, use at night enabled and dormant
```

While I could use MQTT to control OpenDTU-OnBattery externally, I'd prefer to have the option to stop a discharge cycle by hitting a low battery situation OR sunrise/the start of incoming solar power (sunrise would be, I assume, a simple ```if```, so maybe more easy to implement, receiving energy from solar would be more flexible/sensible). Less moving parts is always better. Would such a change be acceptable?

AndreasBoehm, 2025-05-06:
What you basically want is `Use battery only at night` correct?
What do you think about this @schlimmchen?

For me it sounds a bit strange but i don't have dynamic prices, and if i would, i think i would like to have more than a simple only-use-at-night switch.

AndreasBoehm, 2025-05-06:
Please create a feature request for what you mentioned above @wusel42

schlimmchen, 2025-05-06:
Well, more options mean more maintenance, more docs, etc., which is why is dislike them. However, Kai makes a good point and I understand that stopping a discharge cycle in any case in the morning is of importance.

This switch would basically tie the discharge cycle to the notion of "day and night" rather than "thresholds alone", which was the source for confusion in the past anyways. So..... Maybe we make this the new default?

Anyways this is the perfect time for a unit-test of the respective function. Touching it again is outright dangerous...

wusel42, 2025-05-07:
> What you basically want is `Use battery only at night` correct?

Kind of, yes: at night (preferably: after solar power ceased), discharge until defined SoC is reached but always stop at sunrise (preferably: after solar power starts) — that's there already. What's missing: If battery is beyond a specific SoC, start DPL's discharge cycle but stop that not only when the defined SoC is reached but also at sunrise (or solar power starts).

So it's not `Use battery only at night`, more `Stop discharge cycle at sunrise`.

> For me it sounds a bit strange but i don't have dynamic prices, and if i would, i think i would like to have more than a simple only-use-at-night switch.

True; as stated, I'd like to have `Full Solar-Passthrough` without `Solar-Passthrough`: if I enable Solar-Passthrough and set thresholds for Full Solar-Passthrough, this starts the inverter directly:

```
May  6 14:17:37 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] up 348814 s, it is day, next inverter restart at 383374 s (set to 0)
May  6 14:17:37 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] battery interface enabled, SoC 70.7 % (ignored), age 5 s (valid)
May  6 14:17:37 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] BMS: 54.62 V, MPPT: 54.52 V, inverter 11xxxxxxxx60: 55.10 
May  6 14:17:37 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] battery voltage 54.62 V, load-corrected voltage 54.62 V @ 0 W, factor 0.00010 1/A
May  6 14:17:37 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] battery discharge restricted, start 55.00 V or 80 %, stop 50.80 V or 20 %
May  6 14:17:37 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] start NOT reached, stop NOT reached, solar-passthrough disabled, use at night enabled and dormant
May  6 14:17:37 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] total max AC power is 1200 W, conduction losses are 5 %
May  6 14:17:37 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] targeting 300 W, base load is 900 W, power meter reads 539.0 W (valid)
May  6 14:17:37 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] requesting 650 W from 1 solar-powered inverter currently producing 411 W (diff 239 W, hysteresis 20 W)
May  6 14:17:37 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] will cover 411 W using 1 solar-powered inverter
May  6 14:17:37 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 239 W AC
May  6 14:17:37 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] requesting 0 W from 1 battery-powered inverter currently producing 0 W (diff 0 W, hysteresis 20 W)
May  6 14:17:37 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] will cover 0 W using 1 battery-powered inverter
May  6 14:17:37 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL inverter 11xxxxxxxx60]:
May  6 14:17:37 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     battery-powered, standing by at 0 W, output included in power meter reading
May  6 14:17:37 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     lower/current/upper limit: 80/249/1200 W, output capability: 1600 W
May  6 14:17:37 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     sending commands enabled, reachable, eligible
May  6 14:17:37 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     max reduction production/standby: 0/0 W, max increase: 1200 W
May  6 14:17:37 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     target limit/output/state: -1 W (unchanged)/0 W/unchanged, 0 update timeouts
May  6 14:17:37 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     MPPTs AC power/DC voltage: a: 0 W/55.1 V b: 0 W/54.9 V c: 0 W/55.0 V d: 0 W/55.1 V
May  6 14:17:37 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL inverter 11xxxxxxxx97]:
May  6 14:17:37 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     solar-powered, producing 411 W, output included in power meter reading
May  6 14:17:37 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     lower/current/upper limit: 50/800/800 W, output capability: 800 W
May  6 14:17:37 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     sending commands enabled, reachable, eligible
May  6 14:17:37 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     max reduction production/standby: 361/361 W, max increase: 0 W
May  6 14:17:37 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     target limit/output/state: -1 W (unchanged)/411 W/unchanged, 0 update timeouts
May  6 14:17:37 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - -     MPPTs AC power/DC voltage: a: 313 W/31.6 V b: 98 W/41.4 V
[…]
May  6 14:17:58 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] up 348834 s, it is day, next inverter restart at 383394 s (set to 0)
May  6 14:17:58 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] battery interface enabled, SoC 70.8 % (ignored), age 0 s (valid)
May  6 14:17:58 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] BMS: 54.61 V, MPPT: 54.68 V, inverter 11xxxxxxxx60: 55.00 
May  6 14:17:58 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] battery voltage 54.61 V, load-corrected voltage 54.61 V @ 0 W, factor 0.00010 1/A
May  6 14:17:58 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] battery discharge restricted, start 55.00 V or 80 %, stop 50.80 V or 20 %
May  6 14:17:58 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU 4xxxxxx5 - - [DPL] full solar-passthrough dormant, start 56.00 V or 100 %, stop 53.50 V
[... cut ...]  On a cloudy day, DPL would not start, but on a sunny day, it could start in the afternoon already, consuming only solar power, so that the battery would be charged still. If that happens, I'd like to push more solar power into my house, up to the point of reaching actually net-zero from the grid.

As far as I see it, everything is already there, there "just" needs to be some decoupling to be done (alternative end for the discharge cycle, Full and non-full Solar-Passthrough as separate options).

wusel42, 2025-05-07:
> Please create a feature request for what you mentioned above [@wusel42](https://github.com/wusel42)

I think https://github.com/wusel42/OpenDTU-OnBattery-WiP/commit/4c0290358bb7bdb4df4644aa5ea4d96cf5586b8b should do the trick – a least it compiles –, but I'm currently failing to see how to set/add `BatteryRechargeAtDay` in/to the Webfrontend?

AndreasBoehm, 2025-05-07:
I don't think that another switch is a good idea. Things are getting too complicated and the testing effort is increasing as well. I think what @schlimmchen was trying to say is that `Use battery at night ....` should stop discharging at sunrise. Did i get that right @schlimmchen ?

wusel42, 2025-05-07:
> `Use battery at night ....` should stop discharging at sunrise.

I'm always reluctant to change default behaviors, especially as the current options – stop discharge of less-than-wanted charged battery at sunrise but discharge a "defined full" battery until it's "defined empty" – seem to be fine with most users. 

Anyway, https://github.com/hoylabs/OpenDTU-OnBattery/commit/a211862d2bdedfd9cbd6fc46b538714b3a160657 is now running on my OnBattery instance, should power down the battery-powered inverter in about 6 hours although started because of start threshold was reached  — *if* I did understand the code correctly ;-)

```
May  7 23:01:51 192.0.2.249  - OpenDTU-FXXXX8 OpenDTU abcdef0 - - [DPL] start NOT reached, stop NOT reached, solar-passthrough disabled, prioritize battery charge enabled, use at night enabled and dormant
```

Thank you for your work!

wusel42, 2025-05-07:
> Well, more options mean more maintenance, more docs, etc., which is why is dislike them.

Totally see your point, but less adaptivity to local needs isn't a great option either.

> However, Kai makes a good point and I understand that stopping a discharge cycle in any case in the morning is of importance.

Thanks, although I'm not certain that "stopping a discharge cycle in any case" is the way to go, given that the opposite is currently the case. If my battery would be bigger –  which it will become over time –, I would opt for "discharge as much as possible, but ensure a full charge every n days" ...

> Anyways this is the perfect time for a unit-test of the respective function. Touching it again is outright dangerous...

Did I miss something? In test/ I find a README from Captain Obvious, stating tests would make sense ;-) Spaß beiseite: if there'd be a framework already, I'd try to add a test, but as it is, I'd consider that request a bit too much ...

schlimmchen, 2025-05-09:
> I'd try to add a test, but as it is, I'd consider that request a bit too much ...

My remark was targeted to Andreas, no worries.

> I would opt for "discharge as much as possible, but ensure a full charge every n days" ...

Yeah... That's what I anticipate (whenever that might become a reality) the Battery Guard to do.

> Did i get that right @schlimmchen ?

Correct. However, reflecting again on the change of the current behavior: I don't even want that for myself^^ If I made it through the night after my battery reached the start threshold, I surely don't want to fall back to solar-passthrough power once day begins. I want to continue to use the battery, especially as this usually happens on sunny periods, and it would be ill-advised to stop the discharge cycle as the battery will be full by the end of the day anyways.

Okay, so I had to re-read most of this issue to get me back on track. Kai wants to unconditionally stop a discharge cycle in the morning, because of reasons. The commit he made makes sense to implement this, however, the naming of the switch is not to the point. And even if it was, it's still a new switch...

Given that your reasons are volatile energy prices, I don't see this switch in the DPL. What I do see long term instead are more options to control the DPL from outside logic to optimize use of a DPL-governed battery, by means of MQTT or even HTTP API.

I still don't get it... Aren't prices generally low during the night and high during the day? Shouldn't you be using your solar production as well as possible ("solar-passthrough") and only save excess energy to the battery? And if so, why should you stop the discharge cycle in the morning?

wusel42, 2025-05-09:
> If I made it through the night after my battery reached the start threshold, I surely don't want to fall back to solar-passthrough power once day begins. I want to continue to use the battery, [...]

I assume we have largely different base loads, sized batteries and Wp ;-) From a charged battery, I can currently feed ~450W from 18:00 to 6:00, batteries are down to 10ish percent charge. To recharge on a cloudy day, I had to stop any discharge and got back to about 70% of charge only, and hit the stop threshold some hours before sunrise.

The last two days, there was plenty of sun and I had to lower the start threshold to prevent the battery hit 100% (said to shorten it's lifespan) and especially the SmartSolars going into Absorbtion mode — as the HMS would shut itself off (PV-x: Eingangsüberspannung).

So, most likely I'll seek to fetch tomorrow's solar prognosis, change the start threshold in the evening in relation to the prognosis so that the batteries hit around 90% few hours before dawn.

> I still don't get it... Aren't prices generally low during the night and high during the day?

Nah, thanks to the Gigawatts of solar already installed, on a sunny Friday like today, it's dirt cheap midday, with higher prices from usually 6 to 8 and 18 to 22 hours.

[image]

(That's the price per kWh I pay; 0,20 €/kWh of that is taxes and other expenses ...)

During Winter, it's still the way you said — more depending on wind, obviously. Here's, randomly chosen, the graph of 2025-01-25:

[image]

In Winter, I mostly peak-discharge during hours of above that day's average price. (That was per manual schedule in the Shelly, I haven't looked at OnBattery back then.)  So, in "Summer" ...

>  Shouldn't you be using your solar production as well as possible ("solar-passthrough") and only save excess energy to the battery? And if so, why should you stop the discharge cycle in the morning?

... it's quite useless – in general as well as in my situation – to push solar power to the grid when there's already an excess of energy. Tomorrow, 13:00-15:00, I'll pay 0,05 €/kWh, whereas 19:00-22:00 it's 0,34 to 0,38 €/kWh. Therefore I better charge my batteries and go for net-zero via DPL from 19 to 22. According to the prognosis, I assume they will hit the start threshold around 14:00 anyway ;-)

wusel42, 2025-05-10:
> Please create a feature request for what you mentioned above [@wusel42](https://github.com/wusel42)

I wanted this for me anyway, thus: PR https://github.com/hoylabs/OpenDTU-OnBattery/pull/1956.