hoylabs/OpenDTU-OnBattery #1945: VE.Direct "OR" is a bitfield

schlimmchen, 2025-05-06:
### What happened?

See https://github.com/hoylabs/OpenDTU-OnBattery/discussions/1917#discussioncomment-13043367.

### To Reproduce Bug

Hm... How about waiting for the night or disconnecting the inputs **and** disabling using the RX pin of the VE.Direct interface?

### Expected Behavior

We need to interpret the field as a hexadecimal value and test each bit. Transporting this to MQTT and the web UI will probably be a nightmare. I guess we are fine to report the first "off reason" only. If the user needs to figure out why a solar charger is not producing, all "off reaons" can be tackled one at a time.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.04.17

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

SW-Niko, 2025-05-06:
>I guess we are fine to report the first "off reason" only.

I agree with you. I can do this fix.

Snusme83, 2025-05-06:
Also zwei Laderegler sind bei mir in der Tat BMS- gesteuert laut VictronConnect:

[image]

Ich könnte mir vorstellen, das stammt so noch aus Zeiten als ich Venus OS in Betrieb hatte!

Habe jetzt mal beide Laderegler die BMS-Steuerung zurückgesetzt im VictronConnect - dann müssten sich diese beiden Laderegler heute Abend genauso verhalten wie der eine und zwar mit "No input power"
(0X00000001).🤷🏻‍♂️

Snusme83, 2025-05-06:
[image]

Snusme83, 2025-05-06:
Bei mir passt jetzt alles. Lag echt an der BMS-Steuerung, die im VictronConnect eingestellt/ hinterlegt war. Danke euch Beiden.

[image]

closed 2025-05-09 by schlimmchen