hoylabs/OpenDTU-OnBattery #1986: Dynamic power limiting stucks at ceratin value / Maximum Total Output resets to 1000 W after disabling/enabling dynamic powerlimiting

fat-randy, 2025-05-22:
### What happened?

Hello,

My Target Grid Consumption is always -150W. 
Sometimes the Inverter Total Power stucks at a certain value. e.g. stucks at ~350 W.  
So the regulation is "not" working anymore. And the inverter is not allowed to provide more power even it is possible e.g. more thatn 1000W (I have 1780W peak and sun is shining bright)
My workaround is to switch dynamic power limiting off and on (manually via MQTT). Than it works again.
Additinally after using this workaround the Maximum Total Output in the General Tab of Dynamic Power Limiter Settings  resets to 1000 W. Was 1600W before.

Hardware: OpenDTUonbattery
Inverter: HMS 1600 4T 14A (HMS-1600/1800/2000-4T)
Dynamic Power Limiter Settings:
- General
  - Enabled
  -  Govern Inverter "HMS 1600 4T 14A": Enabled
  - Target Grid Consumption: -150 W
  - Base Load: 20 W
  - Hysteresis: 20 W
  - Maximam Total Output: 1600 W
-  HMS 1600 4T 14A (HMS-1600/1800/2000-4T)
  - PowerMeter reading includes inverter output: enabled
  - Power Source: Solar Panels
  - Compensate shaded or unused inputs: Enabled
  - Overscaling input power threshold: 98 %
  - Minimum Power Limit: 52 W
  - Maximum Power Limit: 1600 W

### To Reproduce Bug

Stuckung at certain "Inverter Total Power" Not manually reproduceable.
Resetting to 1000W of "Maximum Total Output" reproduceable by switching on/off Dynamic power limiting.

### Expected Behavior

Dynamic power limiting works without manual manipulation. Not reset of Maximum Total Output after switching on/off Dynamic power limiting.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.02.10

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

spcqike, 2025-05-22:
> 2025.02.10

please try a new release from here
https://github.com/hoylabs/OpenDTU-OnBattery/pull/1960#issuecomment-2872125524

AndreasBoehm, 2025-05-22:
@fat-randy you are using a old version. Please update to the latest release version. Do jot try the version that @spcqike linked until you tried the latest release version.

spcqike, 2025-05-22:
@AndreasBoehm why shouldn’t he use this version? I recommended it, because the only release after 25.02.10 is 25.04.12 and I wasn’t quite sure, if all scaling related PRs are in this release already. At least I haven’t found it in the release notes at a fast glance.

AndreasBoehm, 2025-05-23:
Because its not clear yet if the PR that you linked to will ever get merged.

No worries, `2025.04.17` contains all fixes for solar only setups :)

fat-randy, 2025-05-23:
Hi, firmware is updatet to official 2025.04.17 release.
If the problems appear again I will let you know. Thank you. Great job.