hoylabs/OpenDTU-OnBattery #1999: Wiederholen von Ein- bzw Ausschalten Kommandos an Wechselrichter

scharrrfi, 2025-05-29:
### What happened?

Hallo zusammen,

mir ist aufgefallen, dass es immer mal wieder Fälle gibt, in denen die Wechselrichter das Ein- bzw. Ausschalten Kommando beim DPL nicht empfangen oder ausgeführt haben. OpenDTU OnBattery erkennt das auch über dem jeweiligen Status des Wechselrichters. Der DPL macht allerdings weiter, als wäre das Kommando erfolgreich gewesen. Der Wechselrichter bleibt also an obwohl kein bedarf mehr besteht bzw. bleibt aus und wird trotzdem versucht vom DPL hochzuregeln.

Wäre es möglich das Ein- bzw Ausschalten Kommando einfach nach x Sekunden zu wiederholen, wenn sich der Wechselrichter Status nicht entsprechend geändert hat?

### To Reproduce Bug

S.o.

### Expected Behavior

Wäre es möglich das Ein- bzw Ausschalten Kommando einfach nach x Sekunden zu wiederholen, wenn sich der Wechselrichter Status nicht entsprechend geändert hat?

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025-04-17

### What firmware variant (PIO Environment)?

generic_esp32s3

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

schlimmchen, 2025-05-30:
Logs, or it didn't happen.

Das Wiederholen von (DPL) Kommandos an den WR habe ich implementiert und ich hab erstmal keinen Zweifel daran, dass das auch funktioniert, wie es soll. Daher wäre es wichtig besser zu verstehen, unter welchem Umständen das bei dir auftritt und wie sich das äußert. Anhand der bisherigen Infos muss ich das als falsch ablehnen. Insb. wird definitiv der Inverter gestartet, wenn der DPL ihn produzieren lassen möchte, egal ob das letzte Einschalt-Kommando (selbst mit Wiederholungen) nicht erfolgreich war.

scharrrfi, 2025-06-01:
Kann man die Logs irgendwie dauerhaft speichern?

Leider funktioniert natürlich das Provozieren des Fehlers nicht. Er tritt auch nur ca. einmal die Woche auf. Vielleicht liegt es daran, dass ich zwei Wechselrichter steuer?
Daher Reproduktion und Logs schwierig. Natürlich doof, wenn das implementiert ist und vermutlich auch in 99% der Fälle korrekt funktioniert.

Folgende Symptome habe ich dann. Wenn er nicht einschaltet: Netzbezug. Obwohl das korrekt übermitteilt wird und der DPL anscheinend hochschalter (Wechselrichter ist dann auf 100%, aber aus). Er wird wieder korrekt geregelt wenn ich ihn manuell einachalte oder ich OpenDTU on Battery neustarte.
Wenn er nicht ausschaltet, hab ich das Problem, dass die Batterie nicht anfängt zu laden (wird auch über open dtu gesteuert). Der Wechselrichter ist dann zwar herunter geregelt, geht aber nicht in stand by. Wenn ich ihn dann manuell dahin versetze, klappt wieder alles.

AndreasBoehm, 2025-06-03:
Der WR bleibt an weil die minimale Leistung kleiner ist als die hysterese, das hatten wir eigentlich gefixt dachte ich.

Versuch mal minimale Leistung 25W und Hysterese 20W, das sollte helfen

scharrrfi, 2025-06-03:
Hab es mal geändert und beobachte es.
Erklärt das auch das Problem in die andere Richtung?

AndreasBoehm, 2025-06-03:
> Erklärt das auch das Problem in die andere Richtung?

Das ist eine gute Frage, das müssen wir im Log sehen um die Frage zu beantworten.

scharrrfi, 2025-06-05:
So, nun ist das Problem auch andersherum aufgetreten. Allerdings ist anscheinend die Kommunikation zum Huawei Ladegerät auch abgerissen. Ggf. hängt sich deswegen der DPL auf. 

    

> MPPTs AC power/DC voltage: a: 0 W/54.2 V
> 21:56:40.114 > Websocket: [/console][7] connect
> 21:56:40.114 > Websocket: [/livedata][74] disconnect
> 21:56:40.281 > Websocket: [/huaweilivedata][64] disconnect
> 21:56:40.281 > Websocket: [/batterylivedata][59] disconnect
> 21:56:40.281 > [DPL] up 688082 s, it is day, next inverter restart at 738732 s (set to 12)
> 21:56:40.281 > [DPL] battery interface enabled, SoC 99.0 % (used), age 0 s (valid)
> 21:56:40.281 > [DPL] BMS: 54.01 V, MPPT: -1.00 V, inverter 112184725329: 54.20 
> 21:56:40.281 > [DPL] battery voltage 54.01 V, load-corrected voltage 54.01 V @ 0 W, factor 0.00100 1/A
> 21:56:40.281 > [DPL] battery discharge allowed, start 50.00 V or 20 %, stop 49.00 V or 10 %
> 21:56:40.281 > [DPL] start reached, stop NOT reached, solar-passthrough disabled, use at night enabled and dormant
> 21:56:40.335 > [DPL] total max AC power is 1800 W, conduction losses are 3 %
> 21:56:40.335 > [DPL] targeting 0 W, base load is 300 W, power meter reads 396.0 W (valid)
> 21:56:40.335 > [DPL] DC power bus usage blocked by HuaweiCan auto power
> 21:56:40.335 > [DPL] requesting 0 W from 2 battery-powered inverters currently producing 0 W (diff 0 W, hysteresis 20 W)
> 21:56:40.335 > [DPL] will cover 0 W using 2 battery-powered inverters
> 21:56:40.335 > [DPL inverter 116190748522]:
> 21:56:40.335 >     battery-powered, standing by at 0 W, output included in power meter reading
> 21:56:40.335 >     lower/current/upper limit: 52/109/1500 W, output capability: 1500 W
> 21:56:40.417 >     sending commands enabled, reachable, eligible
> 21:56:40.417 >     max reduction production/standby: 0/0 W, max increase: 1500 W
> 21:56:40.417 >     target limit/output/state: -1 W (unchanged)/0 W/unchanged, 0 update timeouts
> 21:56:40.417 >     MPPTs AC power/DC voltage: a: 0 W/54.1 V b: 0 W/54.0 V
> 21:56:40.417 > [DPL inverter 112184725329]:
> 21:56:40.417 >     battery-powered, standing by at 0 W, output included in power meter reading
> 21:56:40.417 >     lower/current/upper limit: 25/300/300 W, output capability: 300 W
> 21:56:40.417 >     sending commands enabled, reachable, eligible
> 21:56:40.417 >     max reduction production/standby: 0/0 W, max increase: 300 W
> 21:56:41.290 >     target limit/output/state: -1 W (unchanged)/0 W/unchanged, 0 update timeouts
> 21:56:41.290 >     MPPTs AC power/DC voltage: a: 0 W/54.2 V
> 21:56:41.290 > [DPL] up 688083 s, it is day, next inverter restart at 738732 s (set to 12)
> 21:56:41.368 > [DPL] battery interface enabled, SoC 99.0 % (used), age 0 s (valid)
> 21:56:41.368 > [DPL] BMS: 54.01 V, MPPT: -1.00 V, inverter 112184725329: 54.20 
> 21:56:41.368 > [DPL] battery voltage 54.01 V, load-corrected voltage 54.01 V @ 0 W, factor 0.00100 1/A
> 21:56:41.368 > [DPL] battery discharge allowed, start 50.00 V or 20 %, stop 49.00 V or 10 %
> 21:56:41.368 > [DPL] start reached, stop NOT reached, solar-passthrough disabled, use at night enabled and dormant
> 21:56:41.368 > [DPL] total max AC power is 1800 W, conduction losses are 3 %
> 21:56:41.368 > [DPL] targeting 0 W, base load is 300 W, power meter reads 396.0 W (valid)
> 21:56:41.511 > [DPL] DC power bus usage blocked by HuaweiCan auto power
> 21:56:41.511 > [DPL] requesting 0 W from 2 battery-powered inverters currently producing 0 W (diff 0 W, hysteresis 20 W)
> 21:56:41.511 > [DPL] will cover 0 W using 2 battery-powered inverters
> 21:56:41.511 > [DPL inverter 116190748522]:
> 21:56:41.511 >     battery-powered, standing by at 0 W, output included in power meter reading
> 21:56:41.511 >     lower/current/upper limit: 52/109/1500 W, output capability: 1500 W
> 21:56:41.511 >     sending commands enabled, reachable, eligible
> 21:56:41.511 >     max reduction production/standby: 0/0 W, max increase: 1500 W
> 21:56:41.616 >     target limit/output/state: -1 W (unchanged)/0 W/unchanged, 0 update timeouts
> 21:56:41.616 >     MPPTs AC power/DC voltage: a: 0 W/54.1 V b: 0 W/54.0 V
> 21:56:41.616 > [DPL inverter 112184725329]:
> 21:56:41.616 >     battery-powered, standing by at 0 W, output included in power meter reading
> 21:56:41.616 >     lower/current/upper limit: 25/300/300 W, output capability: 300 W
> 21:56:41.616 >     sending commands enabled, reachable, eligible
> 21:56:41.616 >     max reduction production/standby: 0/0 W, max increase: 300 W
> 21:56:41.616 >     target limit/output/state: -1 W (unchanged)/0 W/unchanged, 0 update timeouts
> 21:56:42.080 >     MPPTs AC power/DC voltage: a: 0 W/54.2 V
> 21:56:42.080 > [PowerMeters::Json::Http] New total: 396.00
> 21:56:42.324 > Fetch inverter: 116190748522
> 21:56:42.324 > Queue size - NRF: 1 CMT: 0
> 21:56:42.324 > [DPL] up 688084 s, it is day, next inverter restart at 738732 s (set to 12)
> 21:56:42.324 > [DPL] battery interface enabled, SoC 99.0 % (used), age 0 s (valid)
> 21:56:42.324 > [DPL] BMS: 54.01 V, MPPT: -1.00 V, inverter 112184725329: 54.20 
> 21:56:42.324 > [DPL] battery voltage 54.01 V, load-corrected voltage 54.01 V @ 0 W, factor 0.00100 1/A
> 21:56:42.324 > [DPL] battery discharge allowed, start 50.00 V or 20 %, stop 49.00 V or 10 %
> 21:56:42.324 > [DPL] start reached, stop NOT reached, solar-passthrough disabled, use at night enabled and dormant
> 21:56:42.375 > [DPL] total max AC power is 1800 W, conduction losses are 3 %
[... cut ...]
> 21:56:42.438 >     lower/current/upper limit: 25/300/300 W, output capability: 300 W
> 21:56:42.438 >     sending commands enabled, reachable, eligible
> 21:56:42.438 >     max reduction production/standby: 0/0 W, max increase: 300 W
> 21:56:42.736 >     target limit/output/state: -1 W (unchanged)/0 W/unchanged, 0 update timeouts
> 21:56:42.736 >     MPPTs AC power/DC voltage: a: 0 W/54.2 V
> 21:56:42.736 > Success
> 21:56:43.350 > [DPL] up 688085 s, it is day, next inverter restart at 738732 s (set to 12)
>

scharrrfi, 2025-06-13:
Also, ich glaube tatsächlich das die Einschaltproblematik etwas mit dem den Huawei Netzteil zu tun hat. Heute Nacht ist es wieder ausgestiegen. 
[image]

In den Logs findet sich dazu allerdings nichts sinnvolles, bis auf das das Netzteil eben nicht erreichbar ist. Das ist aber ein anderes Problem. Wäre schon toll, wenn ihr den Punkt mit dem Ausschalten bzgl. der Hysterese fixen könntet.

Entweder spinnt bei mir der MCP2515, der Slot-Adapter oder das Netzteil selbst. Weil nach dem Ein- und Ausstecken des Slotadapters (also Netzteil einmal komplett DC und AC stromlos machen) ist es normalerweise instant wieder für paar Tage erreichbar.

[opendtu-log-2025-06-13_06-46-26.txt](https://github.com/user-attachments/files/20721028/opendtu-log-2025-06-13_06-46-26.txt)

Bin mittlerweile übrigens auf der aktuellen Version, da ich dachte, wegen den ganzen Huawei fixes, steigt das Netzteil vielleicht nicht mehr aus.

closed 2025-07-12 by github-actions[bot]