hoylabs/OpenDTU-OnBattery #2004: System hat nicht mehr geregelt und hoymiles automatisch auf min

ioannis2, 2025-05-31:
### What happened?

Systemfehlerbeschreibung:

Das gesamte System hat plötzlich nicht mehr reagiert.
Konfiguration:
    Zwei Hoymiles 2000 4T sind an die Solarmodule angeschlossen.
    Eine Pylontech-Batterie ist vorhanden.
    Drei weitere Hoymiles 2000 4T sind an die Batterie angeschlossen.
    Ein Huawei-Gerät wird zum Laden der Batterie verwendet.

Das System hat grundsätzlich problemlos funktioniert, solange die Solarmodule ausreichend Strom geliefert haben. Ein Teil der Solarenergie wurde über das Huawei-Gerät zur Ladung der Batterie genutzt.
Bei einem erhöhten Energiebedarf hat die Batterieladung jedoch aufgehört. Daraufhin sind die drei an die Batterie angeschlossenen Hoymiles gestartet, danach herrschte plötzlich Stillstand: Der Energiebezug erfolgte nur noch aus dem Netz, die Hoymiles an den Solarmodulen sind auf ein Minimum zurückgefahren, und es gab keine weitere Reaktion des Systems.
Ein Software-Neustart brachte keine Besserung. Erst nach einer kompletten Trennung der Stromversorgung und anschließendem Neustart der OpenDTU (von der Batterie gespeist), funktionierte das System wieder – allerdings nur für kurze Zeit.
Zusätzlich fiel auf, dass die Aktualisierung der Hoymiles-Werte in der Benutzeroberfläche mehr als 20 Sekunden gedauert hat.

Anbei sende ich einige Log-Dateien zur weiteren Analyse.

### To Reproduce Bug

bei kleiche Verhalten ist es wieder das gleiche passiert

### Expected Behavior

bitte überprüfen

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.04.17

### What firmware variant (PIO Environment)?

generic_esp32s3

### Relevant log/trace output

```Shell

```

### Anything else?

[log3.txt](https://github.com/user-attachments/files/20534265/log3.txt)
[log2.txt](https://github.com/user-attachments/files/20534266/log2.txt)
[log1.txt](https://github.com/user-attachments/files/20534264/log1.txt)

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

ioannis2, 2025-05-31:
Ich habe jetzt die Mindestleistung für alle drei Hoymiles 2000 4T, die mit der Batterie verbunden sind, auf 150 W eingestellt. Ein Neustart über die Benutzeroberfläche hat jedoch ebenfalls keine Besserung gebracht

[log5.txt](https://github.com/user-attachments/files/20534670/log5.txt)

spcqike, 2025-05-31:
> 07:24.837 > [DPL] targeting -50 W, base load is 100 W, power meter reads 281.1 W (stale)

Meine erste Vermutung scheint auch hinzukommen. Dein powermeter sendet keine aktuellen Daten. Dadurch werden inverter an der Batterie gestoppt und Solarbetriebene inverter auf die Grundlast gefahren.

ioannis2, 2025-05-31:
Vielen Dank! Ich habe den Shelly noch einmal neu gestartet – jetzt sieht es deutlich besser aus.
Eine kurze Frage noch: Die Aktualisierungsrate liegt aktuell bei 25 Sekunden oder hoher.
Gibt es eine Möglichkeit, sie wieder auf etwa 10 Sekunden zu reduzieren?

spcqike, 2025-05-31:
> Die Aktualisierungsrate

Welche genau? 

> Gibt es eine Möglichkeit, sie wieder auf etwa 10 Sekunden zu reduzieren?

Was hast du denn seither geändert dass es nun so lange ist? 

Das Intervall des powermeters kannst du auf 1s stellen. Zumindest habe ich es so. 

Das Intervall der DTU auf 1s. 

Der DPL. Tja, der wird mit jedem inverter Träger. Das sollte aber im allgemeinen nicht schlimm sein, da theoretisch so geregelt werden sollte, dass ein Update an genau einen inverter reicht, um wieder auszuregeln. (Klappt aber nicht immer. Je nach vorhandenen invertern und anliegenden oder benötigten Leistungen)

ioannis2, 2025-06-01:
"Was genau ist denn dein Problem an den 25s?"
Es macht auf mich den Eindruck, dass die Anzeige tatsächlich der Realität entspricht und immer dann nachgeregelt wird, wenn auch eine Aktualisierung erfolgt. Vielleicht täusche ich mich aber.

Ich habe bisher keinen direkten Vergleich zwischen der Anzeige und der tatsächlichen Messung mit dem Smartmeter durchgeführt, da mein Smartmeter derzeit defekt ist. Sobald ich ein neues Gerät habe, kann ich diesen Vergleich gerne nachholen.

Für mich wäre es kein Problem, wenn es sich bei der Abweichung lediglich um eine kosmetische Anzeige im Webinterface handelt. Eine Steuerung mit einem Aktualisierungsintervall von 25 Sekunden, wie du selbst sagst, halte ich jedoch für nicht optimal.

Vielleicht kann ein Entwickler uns genauer erklären, wie sich das Programm tatsächlich verhält und in welchen Intervallen die Aktualisierung intern bei 1, 2, 3, 4 usw. Mikrowechselrichtern durchgeführt wird.

ioannis2, 2025-06-02:
Ich habe es mir heute mit einem anderen Shelly-Gerät angeschaut. Es scheint, dass etwa alle 10 Sekunden eine Aktualisierung im Hintergrund erfolgt.

AndreasBoehm, 2025-06-02:
> ```
> 1 Hoymiles: 5 Sekunden Aktualisierungsintervall
> 
> 2 Hoymiles: 10 Sekunden Aktualisierungsintervall
> 
> 3 Hoymiles: 15 Sekunden Aktualisierungsintervall
> 
> 4 Hoymiles: 20 Sekunden Aktualisierungsintervall
> 
> 5 Hoymiles: 25 Sekunden Aktualisierungsintervall
> ```

Da die WR nacheinander abgefragt werden erhöht sich das Aktualisierungsintervall entsprechend.
Wenn du tatsächlich alle 5 WR an einer DTU betreiben willst und den DPL sinnvoll nutzen möchtest dann musst du das Intervall auf 1 Sekunde reduzieren, das würde ich sowieso empfehlen.
Optimal wäre es wenn du die Solar WR und die Batterie WR an unterschiedlichen DTUs betreibst, sonst wirst du nie auf ein vernünftiges Aktualisierungsintervall kommen.

> Heute ist mir eine Auffälligkeit bei der Regelung zwischen Hoymiles und Huawei aufgefallen.
> Die Abstimmung scheint nicht optimal zu sein.

Das kann du bei 25 Sekunden Aktualisierungsintervall auch nicht erwarten.

Natürlich wird die Leistung der Solar WR langsam erhöht je mehr Leistung das Huawei sich nimmt, aber wenn du nur -50W anstrebst dauert das eben bis das Huawei nachregelt und dann auch der DPL wieder nachregelt. Wenn aber das Huawei mal überschwingt und die Leistung deshalb wieder abgeregelt wird, zieht auch der DPL die Leistung wieder zurück. Das sind einfach zwei Regelungen die gegeneinander arbeiten.

Warum die Solar WR nicht ungeregelt auf 100%? Das wäre für das laden über das Huawei am besten.

ioannis2, 2025-06-02:
Danke für deine Antwort. Ich werde die Einstellungen für die Solar-Wechselrichter auf 100 % setzen. Mein Ziel war es jedoch, möglichst eine Null-Einspeisung zu erreichen.

schlimmchen, 2025-06-15:
Nach dem Überfliegen kann ich hier kein Problem an OpenDTU-OnBattery erkennen. Daher schließe ich dieses Issue. Falls ich was übersehen habe, bitte nochmal erläutern.

closed 2025-06-15 by schlimmchen