hoylabs/OpenDTU-OnBattery #2005: OpenDTU-onBattery friert Ausgangslimit ein nach Nachtstandby!

KIngdingelIIng, 2025-06-01:
### What happened?

Sobald die DTU (esp32 s3 16mb) nach der Nacht( Standby) wieder Werte vom Wechselrichter bekommt limitiert sie die Ausgangsleistung des Wechselrichters so niedrig das Teilweise obwohl genug Sonne vorhanden ist, Sie nichtmal den geringen Hausstrom ( shelly)  bedient der weit unter dem liegt was die Module in dem moment liefern könnten!  Startet man in dem Moment die DTU neu läuft sie über den Tag wieder normal und macht was sie soll! 
Ich hatte in der letzten Dtu on bat. die ich hatte einen Menüpunkt : NEUSTART  wo ich angeben konnte das sie sich immer zb. um 0 uhr  einmal neustartet! 
Diesen Menüpunkt gibt es aber in der Aktuellen Firmware nicht! (2025.04.17) 

### To Reproduce Bug

Jeden Tag nach dem Nacht-standby muss man die DTU einmal neu starten damit sie ordentlich funktioniert!

### Expected Behavior

Bitte die Neustartfunktion wieder implementieren!!
Dann sollte das Poblem behoben sein!

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.04.17

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

spcqike, 2025-06-01:
Sry aber es gab noch nie eine Option, um die DTU regelmäßig neu zu starten. 
Die Option gabbing givt es nur für Batterie betriebene wechselrichter. 

Und:
Ohne log kann man deinen möglichen Fehler nicht nachvollziehen.

Hannes58, 2025-06-23:
@KIngdingelIIng hast du schon mal die OpenDTU-OnBattery Webconsole geöffnet ? Siehe hier ein Beispiel aus meinem Live System, du kannst über copy to clipboard das Log ziehen und dann in eine eigene log datei deines Vertrauens legen.

[image]

KIngdingelIIng, 2025-06-23:
Ja aber ich wüsste nicht wie ich genau den zeitraum erwische oder muss ich das dann 24std laufen lassen??

closed 2025-07-24 by github-actions[bot]