hoylabs/OpenDTU-OnBattery #2016: Enabling PowerMetter https + json makes openDTU frequently unresponsive

helgeerbe, 2025-06-03:
### What happened?

Hi all,

greetings from the past. It's good to see, that this nice little project is still well maintained!

After one year I decided to upgrade my installation on my espmini.

I compiled master branch on 29.05.25.

Upgrade went smooth and without any issues. But when enabling PowerMeter https + json I got crashes (see log below).

On the 2.6.25 I saw a commit https://github.com/hoylabs/OpenDTU-OnBattery/commit/970bfb65258adbf780beffd95729e0fb2ab8cdf4 on master. Maybe this error is already addressed.

I fall back to release 2025.04.17. Unfortunately I don't have any logs. But when I try to query the PowerMeter values I only got values  for around every 10th request.

My settings:

[image]

My PowerMeter instance a node red-server working as a Proxy. As my original PowerMeter delivers json, but value is text string containing  also a W. Which is not parsable by OpenDTU. Curl, and browser request works as expected.

When I enable this setting. OpenDTU becomes shaky. Lot of disconnects in home assistant via MQTT. Webinterface is offen not reachable.

As a workaround I send PowerMeter values now via MQTT and OpenDTU becomes rock solid and very responsive.

### To Reproduce Bug

Use Test-Button on PowerMeter setting to query PowerMeter via https and json.

### Expected Behavior

Every request to PowerMeter via https + json is working. Enabling this setting should not affect the stability of OpenDTU

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.04.17

### What firmware variant (PIO Environment)?

generic_esp32_4mb_no_ota

### Relevant log/trace output

```Shell
09:15:48.382 > Starting OpenDTU
09:15:48.382 > Initialize FS... done
09:15:48.532 > Reading configuration... Check for default DTU serial... done
09:15:48.533 > done
09:15:48.547 > Reading language pack... done
09:15:48.571 > Reading PinMapping... found valid mapping done
09:15:48.577 > [SerialPortManager] HW UART port 0 now in use by 'Serial Console'
09:15:48.691 > Initialize Network... done
09:15:48.712 > Setting Hostname... Configuring WiFi STA using new credentials... done
09:15:48.716 > [SyslogLogger] Disable
09:15:48.719 > [SyslogLogger] Syslog not enabled
09:15:48.722 > Initialize NTP... done
09:15:48.726 > Initialize SunPosition... done
09:15:48.736 > Initialize MqTT... done
09:15:48.766 > Initialize WebApi... done
09:15:48.768 > Initialize Display... done
09:15:48.770 > Initialize LEDs... done
09:15:48.790 > Initialize Hoymiles interface... NRF: Connection successful
09:15:48.794 >   Setting radio PA level... 
09:15:48.797 >   Setting DTU serial... 
09:15:48.800 >   Setting poll interval... 
[... 55 lines cut ...]
09:16:10.055 >   #5  0x401cd26d in mbedtls_entropy_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/entropy.c:328
09:16:10.055 >   #6  0x401c8bb5 in mbedtls_ctr_drbg_reseed_internal at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ctr_drbg.c:369
09:16:10.055 >   #7  0x401c8d0a in mbedtls_ctr_drbg_seed at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ctr_drbg.c:475
09:16:10.055 >   #8  0x401525f9 in start_ssl_client(sslclient_context*, IPAddress const&, unsigned int, char const*, int, char const*, bool, char const*, char const*, char const*, char const*, bool, char const**) at /Users/herbe/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/ssl_client.cpp:149 (discriminator 3)
09:16:10.055 >   #9  0x40151ae6 in WiFiClientSecure::connect(IPAddress, unsigned short, char const*, char const*, char const*, char const*) at /Users/herbe/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:141
09:16:10.055 >   #10 0x40151b9b in WiFiClientSecure::connect(char const*, unsigned short, char const*, char const*, char const*) at /Users/herbe/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:136
09:16:10.055 >   #11 0x40151c6a in WiFiClientSecure::connect(char const*, unsigned short) at /Users/herbe/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:117
09:16:10.055 >       (inlined by) WiFiClientSecure::connect(char const*, unsigned short) at /Users/herbe/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:113
09:16:10.055 >   #12 0x40247409 in WiFiClientSecure::connect(char const*, unsigned short, int) at /Users/herbe/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:122
09:16:10.055 >   #13 0x40166751 in HTTPClient::connect() at /Users/herbe/.platformio/packages/framework-arduinoespressif32/libraries/HTTPClient/src/HTTPClient.cpp:1162
09:16:10.055 >       (inlined by) HTTPClient::connect() at /Users/herbe/.platformio/packages/framework-arduinoespressif32/libraries/HTTPClient/src/HTTPClient.cpp:1126
09:16:10.055 >   #14 0x40167a1b in HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at /Users/herbe/.platformio/packages/framework-arduinoespressif32/libraries/HTTPClient/src/HTTPClient.cpp:601
09:16:10.055 >   #15 0x40167c73 in HTTPClient::GET() at /Users/herbe/.platformio/packages/framework-arduinoespressif32/libraries/HTTPClient/src/HTTPClient.cpp:518
09:16:10.055 >   #16 0x400dffb1 in HttpGetter::performGetRequest() at src/HttpGetter.cpp:160
09:16:10.055 >   #17 0x4013acd9 in PowerMeters::Json::Http::Provider::poll() at src/powermeter/json/http/Provider.cpp:126
09:16:10.055 >   #18 0x4013b1e2 in PowerMeters::Json::Http::Provider::pollingLoop() at src/powermeter/json/http/Provider.cpp:94
09:16:10.055 >   #19 0x4013b289 in PowerMeters::Json::Http::Provider::pollingLoopHelper(void*) at src/powermeter/json/http/Provider.cpp:72
```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

schlimmchen, 2025-06-03:
I could easily reproduce your issue using an HTTPS (not HTTP, but HTTPS) connection to an Apache2 server. Given the very large call trace, I still assumed that the stack size for the respective task was just too low. I increased it to 8k and the issue went away. As per the watermark, the task has 3400 Bytes free, so 4792 Bytes are in use. That's more than 4k and explains why you did not have success increasing it to 4k.

Please increase the stack size of that task to 6144 Bytes and try again. 6k should give sufficient margin.

This issue is larger than it appears right now: We need to double-check the MQTT client task when using TLS. It should be okay, as it still has more than 4k free. I am somewhat sure that other parts of the firmware also use the HttpGetter... Yes, the HTTP+SML PowerMeter. That also has a stack size of 3k right now, which surely is too small as well, at least with HTTPS.

Yup, crashes immediately:
```
[2025-06-03 21:59:04] Guru Meditation Error: Core  1 panic'ed (Double exception). 
[2025-06-03 21:59:04] 
[2025-06-03 21:59:04] Core  1 register dump:
[2025-06-03 21:59:04] PC      : 0x4037d876  PS      : 0x00040d36  A0      : 0x820f9fff  A1      : 0x3fcc57c0  
[2025-06-03 21:59:04] A2      : 0x00040d36  A3      : 0x00040026  A4      : 0x00000080  A5      : 0x00000000  
[2025-06-03 21:59:04] A6      : 0x00000080  A7      : 0x00000080  A8      : 0x3fcc5880  A9      : 0x00000001  
[2025-06-03 21:59:04] A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000007  A13     : 0x00000005  
[2025-06-03 21:59:04] A14     : 0x00000000  A15     : 0x00000003  SAR     : 0x00000010  EXCCAUSE: 0x00000002  
[2025-06-03 21:59:04] EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  
[2025-06-03 21:59:04] 
[2025-06-03 21:59:04] 
[2025-06-03 21:59:04] Backtrace: 0x4037d873:0x3fcc57c0 0x420f9ffc:0x00000000 |<-CORRUPTED
```

Increasing to 6k leaves 1528 Bytes free, which should be sufficient margin (and proof that 4k would also have not been enough).

helgeerbe, 2025-06-04:
I can confirm, that setting the stack size to 6144 Bytes solved the crashes, but ....

```log
09:42:55.840 > [121178][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():264]: (-32512) SSL - Memory allocation failed
09:42:55.851 > [121188][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -32512
09:42:56.301 > [PowerMeters::Json::Http] New total: -4468.00
09:43:03.603 > Admin AP remaining seconds: 120 / 180
09:43:06.252 > [PowerMeters::Json::Http] New total: -6103.00
09:43:13.638 > Admin AP remaining seconds: 130 / 180
09:43:16.261 > [PowerMeters::Json::Http] New total: -5585.00
09:43:23.671 > Admin AP remaining seconds: 140 / 180
09:43:25.252 > [150591][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():264]: (-32512) SSL - Memory allocation failed
09:43:25.263 > [150601][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -32512
09:43:25.283 > [PowerMeters::Json::Http] Value 1: HTTP Error: connection refused
09:43:34.088 > Admin AP remaining seconds: 150 / 180
09:43:36.220 > [PowerMeters::Json::Http] New total: -8050.00
09:43:42.566 > [167904][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():273]: (-10368) X509 - Allocation of memory failed
09:43:42.576 > [167915][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -10368
``` 

The allocation errors only occur, when I hit the manual request button in the UI. The normal polling works fine.

I know that security is a real pain in the ass. And the complexity of OpenDTU goes far beyond what you would normally use a microcontroller like the esp32 for. But I am always amazed at what is possible here, if you take the limitations into account and optimize them. It kind of reminds me of the old days where you fight for every byte.

schlimmchen, 2025-06-07:
> Could this be one of the issues that @tbnobody and @schlimmchen reportedly had when updating the core to v3.0.0 ?

Hm? I have never tried upgrading to v3, yet. Thomas is experimenting, as far as I know, but nothing more.

> According to this older thread answer by lbernstone it requires even 64kB of memory for a TLS client transaction:

Yes, maybe. We are talking about stack memory in the context of this issue, not heap memory.

helgeerbe, 2025-06-07:
> Is the available heap memory on ESP32 even less than on ESP32-S3, despite its SRAM being 8kB larger? 

Heap is 248 kB

> Maybe Helge has much more going on on his device, whereas I am testing with my fairly bored staging board...

Nope, I only configured my network and setup PowerMeter. Nothing else is enabled.

schlimmchen, 2025-06-08:
> Heap is 248 kB

So effectively there is 42kB more available on an ESP32-S3. That can make a difference. The difference between the background request to read the power meter and the interactive one using the we UI certainly is that the web process that triggers the interactive request is also lingering on the heap. I guess that's enough to break the interactive HTTPS request.

stefan123t, 2025-06-09:
I have just double checked with my previous DPL+PM setup, but obviously I have used `http` for querying the three endpoints on the Shelly Pro 3EM https://github.com/tbnobody/OpenDTU/issues/272#issuecomment-2219644352

Though I consider using HTTPS a best practice and common sense security conscious behaviour in general. 
But I have my doubts that TLS can be configured / used on the ESP32 as the HTTPS client may be eating through chunks of memory.

@helgeerbe can we quantify that amount of memory, i.e. can you configure once with `http` and once with `https` and compare the memory usage ?

helgeerbe, 2025-06-10:
> [@helgeerbe](https://github.com/helgeerbe) can we quantify that amount of memory, i.e. can you configure once with `http` and once with `https` and compare the memory usage ?

I could do this. But can you tell me where exactly in the code I should put the measurement points? Is free heap sufficient?

commit f34927cd (2025-06-07): Fix: increate HTTP power meter stack sizes

the previous 3k stack size was too small even for HTTP connections, due
to the usage of ESP_LOG and DTU_LOG macros. it turns out that 4k is also
way too small to perform HTTPS requests.

closes #2016.