hoylabs/OpenDTU-OnBattery #2033: DPL Regelstrategie: Rechenfehler bei Limitreduktion durch Lastabfall

AIexBV, 2025-06-09:
### What happened?

Beim Lastabfall (Reduzierung) wird das Inverterlimit um die Einspeiseleistung reduziert, was prinzipiell korrekt ist. Liegt die aktuelle Inverterleistung allerdings unter dem alten Limit, wird das neue Limit unter Umständen - sofern es auch noch über der aktuellen Inverterleistung liegt - zunächst gar nichts bewirken. In der Folge muss die Limitregelung u.U. erst mehrere Runden durchlaufen, bis die gewünschte Einspeiseleistung erreicht wird.

Richtig wäre es an der Stelle, das neue Limit anhand der aktuellen Leistung abzüglich der Einspeisung zu ermitteln.

### To Reproduce Bug

Eine höhere Last anschließen, als der Inverter versorgen kann. In der Konsole sieht man, dass das Limit hochgeregelt wird, aber nicht für die Last ausreicht. Dann die Last so reduzieren, dass eine geringe Einspeisung über der Hysterese z.B. 60W entsteht. In der Konsole sieht man dann, dass das Limit von beispielsweise 800W um 60W auf 740W reduziert wird, obwohl es eigentlich von der aktuellen Leistung von beispielsweise 300W auf 240W reduziert werden müsste. Erst nach mehreren Nachregelungen wird das gewünschte Limit erreicht.

### Expected Behavior

Bei Lastabfall soll das neue Limit die aktuelle Inverterleistung als Berechnungsgrundlage berücksichtigen, statt dem alten Limit.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.04.17

### What firmware variant (PIO Environment)?

generic_esp32_4mb_no_ota

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

spcqike, 2025-06-12:
> Liegt die aktuelle Inverterleistung allerdings unter dem alten Limit, wird das neue Limit unter Umständen - sofern es auch noch über der aktuellen Inverterleistung liegt - zunächst gar nichts bewirken.

das ist auch normal. Wenn der Inverter sein aktuelles Limit nicht erreicht, wird davon ausgegangen, dass die Eingänge die geforderte Leistung schlicht nicht liefern.
Warum sollte der Inverter also ein neues, sogar höheres Limit bekommen, wenn er es so oder so nicht erreicht?

Das Ganze ist kein Bug, sondern by Design. Damit werden unnötige Übertragungen reduziert.

Im Betrieb mit mehreren Wechselrichtern wird der Inverter nicht angesprochen. stattdessen bekommt einer, der "am Limit läuft", also tendentiell mehr liefern könnte, ein höheres Limit. 

Ist der Inverter alleine, bekommt er einfach kein neues Limit. Wozu auch?

AIexBV, 2025-06-12:
Es geht konkret um Lastreduktion, nicht um Lasterhöhung, für die das zutrifft, was du schreibst.

### Beispiel (zur Vereinfachung ohne Hysterese)
PV-Leistung 300 Watt
Last 1000 Watt
Limit 800 Watt

_Dann Lastreduzierung auf 250 Watt:_

Einspeisung ist folglich 50 Watt
Neues Limit wird jetzt um 50 Watt von 800 Watt auf 750 Watt reduziert!
Im nächsten Schritt wird das Limit von 750 Watt auf 700 Watt reduziert usw.
Erst nach mehreren unnötigen Übertragungen(!) wird das gewünschte Limit eingestellt.

_Richtig wäre es, das Limit sofort von 800 Watt auf 250 Watt zu reduzieren!_

spcqike, 2025-06-12:
das hast du so oben aber nicht geschrieben. 

> sofern es auch noch über der aktuellen Inverterleistung liegt

ich hatte verstanden

```
Limit_alt 500
PV 300
Limit_neu 400
keine Änderung
```

> Neues Limit wird jetzt um 50 Watt von 800 Watt auf 750 Watt reduziert!

Das kann ich mir so gar nicht vorstellen, kann es aber auch grade nicht bewusst erzwingen und reproduzieren. könntest du da von einen Logauszug bereitstellen?

der DPL sieht eigentlich keine Schleifen vor. er berechnet in jedem Durchgang das Limit, so wie du es gern hättest.

https://github.com/hoylabs/OpenDTU-OnBattery/blob/e36ecd45219aa4bb73733c561e0aafd37ec397b2/src/PowerLimiter.cpp#L522-L588

es wird der aktuelle powermeter Wert genommen, dazu die Summe der AC Ausgangsleistung aller Wechselrichter und das gewünschte Ziel. Jeder Durchlauf sollte, unter Berücksichtigung der Hysterese, eine Punktlandung sein. 

Eine inkrementelle Annäherung ist so nicht vorgesehen.

Das Ziel wird nur dann nicht getroffen, wenn die Eingänge des Wechselrichters ungleich belastbar sind, sie also unterschiedlich viel beisteuern können und das durch "Überskalieren" nicht kompensiert werden kann.

in deinem Beispiel sollte die Berechnung also ungefähr so sein:
```
PV_Leistung 300
Last 1000 / Limit 800
Lastreduzierung auf 250

-50W Einspeisung

neues Limit = -50 + 300 - Target (0?) = 250
```

AndreasBoehm, 2025-06-12:
> Last 1000 Watt 
> Limit 800 Watt
> PV-Leistung 300 Watt 
> 
> _Dann Lastreduzierung auf 250 Watt:_
> 
> Einspeisung ist folglich 50 Watt Neues Limit wird jetzt um 50 Watt von 800 Watt auf 750 Watt reduziert! Im nächsten Schritt wird das Limit von 750 Watt auf 700 Watt reduziert usw. Erst nach mehreren unnötigen Übertragungen(!) wird das erforderliche Limit erreicht.
> 
> _Richtig wäre es, das Limit sofort von 800 Watt auf 250 Watt zu reduzieren!_

Die grundsätzliche Frage ist erst mal warum das Limit überhaupt auf 800 W hochgeht wenn am Schluss nur 300 W vom WR kommen? Hast du Überskalierung aktiviert? Erzähl uns doch bitte mal bisschen was über dein Setup und deine Einstellungen.

AIexBV, 2025-06-12:
> Die grundsätzliche Frage ist erst mal warum das Limit überhaupt auf 800 W hochgeht wenn am Schluss nur 300 W vom WR kommen? Hast du Überskalierung aktiviert? Erzähl uns doch bitte mal bisschen was über dein Setup und deine Einstellungen.

Dass das Limit über die mögliche Produktion gesetzt wird, habe ich schon oft beobachtet. Wenn der WR genau am eingestellten Limit produziert und eine starke Last eingeschaltet wird, wird das Limit richtigerweise um dessen Leistung erhöht. Die Software weiß ja zunächst nicht, dass der WR das neue Limit gar nicht erreichen kann. Das ist m.E. so korrekt. Überskalierung verwende ich nicht.

@spcqike Ich bin kein C++ Spezi, aber aus den ersten Blick sieht das ordentlich aus.

Ich versuche nochmal ein Setup aufzusetzen, um das Phänomen zu loggen. Ich muss dazu sagen, dass bei mir 2 WR laufen. Daher ist es nicht so einfach, alles so einzustellen, dass der Effekt sichtbar wird.

AndreasBoehm, 2025-06-12:
Ich wiederhole mich: Erzähl uns doch bitte mal bisschen was über dein Setup und deine Einstellungen.

AndreasBoehm, 2025-06-12:
Nur die DPL Einstellungen und infos über dein Setup

AIexBV, 2025-06-13:
Meine PV besteht aus einem Hoymiles HMS-1800 und einem HMS-2000. Im DPL ist angestrebter Netzbezug auf 10W, Grundlast auf 110W, Hysterese auf 20W und max. Gesamtleistung auf 800W. Die beiden WR sind jeweils auf 55W minimal und 400W maximal gesetzt. "_Ungenutzte Eingänge ausgleichen_" wird nicht verwendet.

Ich konnte den Effekt jetzt reproduzieren. Oben im Bild ist der Netzbezug zunächst 1766W. Bei beiden WR ist das Limit über der Produktion, soweit also unauffällig. Nach Lastabschaltung meldet der Zähler 84W Einspeisung. Anschließend wird das Limit im ersten WR von 398W auf 304W reduziert, was aber nichts bringt, da er ohnehin nur 209W produziert. Erst nach 2 weiteren Iterationen (nicht mehr abgebildet), wird die Zielproduktion erreicht.

[image]

AIexBV, 2025-06-14:
Jetzt mit Überskalierung. Da sieht deutlich besser aus!

[image]

AndreasBoehm, 2025-06-25:
Danke fürs reinschauen @spcqike!

Ich hab einen draft PR erstellt und nutze dort `getCurrentOutputAcWatts()` als baseline für increase und reduction operationen für alle inverter typen. Lokal hab ich es noch nicht probiert wie es sich verhält.

https://github.com/hoylabs/OpenDTU-OnBattery/pull/2079#issuecomment-3005409925

Mit dem build kannst du Überskalieren wieder ausschalten @AIexBV

closed 2025-09-03 by AndreasBoehm