hoylabs/OpenDTU-OnBattery #2036: deutlich schlechtere Funkverbindung mit neueren Dev. releases

peff74, 2025-06-10:
### What happened?

mit den neueren Versionen aus dem Development habe ich mit meinem HM1500 / HM800 deutlich schlechtere Funkverbindungen was man auch eindeutig in der Funkstatistik und den Logs sieht.

Es kommt häufig zu sehr vielen Fehlern hintereinander 

```
[10:23:37.634] W (259642) hoymiles: Middle missing
[10:23:37.727] W (259786) hoymiles: Middle missing
[10:23:38.365] W (260235) hoymiles: Middle missing
[10:23:38.461] W (260403) hoymiles: Middle missing
[10:23:38.721] W (260537) hoymiles: Middle missing
```

[image]

### To Reproduce Bug

Wenn ich eine ältere Version nutze z.B. die [ea16c8a](https://github.com/hoylabs/OpenDTU-OnBattery/commits/ea16c8a) besteht das Problem nicht. 
Vom Gefühlt her, könnte das Problem seit der Einführung der neuen Logs bestehen.

### Expected Behavior

wieder besserer Emfang :-)

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

dc71c8b

### What firmware variant (PIO Environment)?

generic_esp32_8mb

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

peff74, 2025-06-10:
ich habe soeben den Menupunkt Protokollierung entdeckt.
Dort die Standard-Protokollierungsstufe auf (W) eingestellt (stand auf (V) und Magic die Verbindung steht wieder wie eine Eins!

[image]

Da scheint das Logging in die Funkverbindung reinzugrätschen.

Sorbat, 2025-06-10:
Was ist W? Und V?

tbnobody, 2025-06-10:
@schlimmchen I also saw this behavior... It began with the implementation of the changed queue handling. (The interrupt received message is there since the very beginning, so I don't think thats the impact)
Not yet sure what takes so long here.

firedp, 2025-06-12:
Wo ist denn diese Einstellung "Protokollierungseinstellungen" versteckt? Habe sie noch nicht entdeckt.

peff74, 2025-06-12:
die gab es ab einer der neueren Develop Versionen.
Hatte viel rumprobiert wegen meiner Empfangsproblemen...

spcqike, 2025-06-12:
die sind in der dev-Version oder auch, seit dieser Nacht, im neusten Release

[image]

RicoBach68, 2025-06-12:
V20250611 hat bei mir ebenfalls extrem schlechte Verbindung zu den zwei HM-1500. Zwischen 15:00 und 17:00 Uhr hat die Anlage kaum noch reagiert. 17:00 Uhr habe ich die DTU neu gestartet und die Protokollierung auf "Keine" gesetzt, der Empfang ist trotzdem sehr schlecht geblieben. Zurück zu 20250417???

[image]

[image]

peff74, 2025-06-13:
> Zurück zu 20250417???
Wenn die 2025.06.11 net rennt, natürlich zurück.
Aber da wir bei Dir was anderes faul sein, oder die Umstellung der Protokollierung hat nicht geklappt, schau ggf. in der Konsole

stefan123t, 2025-06-13:
> tbnobody: schlimmchen I also saw this behavior... It began with the implementation of the changed queue handling. (The interrupt received message is there since the very beginning, so I don't think thats the impact)
> Not yet sure what takes so long here.

I dont know exactly how the new queueing has been implemented, but I recall the command simply checks whether there is already a command of the same type in the queue, eg ActivePowerLimit in which case it may simply overwrite the old command's parameters with the new target Power Limit.

Did we introduce a race condition that way, ie the TX/RX routines colliding with the command queue ? 

Maybe we have to take special care of the head element of the command queue, eg lock/unlock it when we are already in the TX/RX process ?

But then again this should not impact the DTU inverter communication for more than a cycle. Except when this happens more than once when the command queuer hits the DTU very frequently, like always during TX/RX ... ?

@tbnobody aside from this shouldnt the Configuration.cpp migrate() Method change the **V**erbose to **W**arning level for this or the next version by default ?

Andrix82, 2025-06-17:
Kann die selbe Beobachtung bestätigen, mit vorhergehendem Release 100% Funkverbindung über Wochen, mit dem aktuellen Release und Standard Setting  <10% Transmission Rate, nach Deaktivierung der Protokollierung wieder 100%

Ähnliches bei MPPT, vorher alle 1-2 Sekunden Werteupdate, nun gerade bei hohen Leistungen bis zu ~1 Minute keine Updates.
Nach Deaktivierung der Protokollierung wieder i.O.  ( Meine MPPT Verbindung ist schon bekannt für einen eher schwachen pyhsical layer :D )

Zen-Samko, 2025-06-20:
In der Protokollierung steht 
ESP-IDF Kern auf E Fehler, 
was ich nicht geändert habe, der Rest nimmt vom oben gewählten Standard Wert!

LG

speddy2222, 2025-07-13:
Gute Morgen,
wo finde ich die Dev. Variante ich habe auch ständig Abbrüche?

Grüße

RealNBB, 2025-07-16:
> Gute Morgen, wo finde ich die Dev. Variante ich habe auch ständig Abbrüche?
> 
> Grüße

https://github.com/hoylabs/OpenDTU-OnBattery/actions/workflows/build.yml?query=branch%3Adevelopment