hoylabs/OpenDTU-OnBattery #2042: [Request]Deactivate power limiter for a while

MarkusPVHM, 2025-06-12:
### Is your feature request related to a problem? Please describe.

When battery is 100% SoC, is there a possibilty to deactivate the limiter for a while, eg 2h? At the moment the solar energy is reduced by the solar chargers. And I will have no income from the grid provider.

### Describe the solution you'd like

extra timer botton for example for deactivation

MarkusPVHM, 2025-06-12:
That would be great, but I do not see this in my configuration. Battery data is sent by MQTT. Is that the reason? I have Daly BMS....

AndreasBoehm, 2025-06-12:
It also works with the MQTT battery. 
The only requirement is that you have solar-passthrough enabled in the DPL config.
You can send a screenshot of the settings and we can take a look.