hoylabs/OpenDTU-OnBattery #2049: Update der Wiki Anleitung für Zendure SF Einbindung

ff-81, 2025-06-15:
Hallo,

per Zufall bin ich heute auf das PDF-Dokument gestoßen.

> [System-Daten_Zendure-SolarFlow_Hoymiles-HMS1600_OpenDTU-DPL_01.2-2025.pdf](https://github.com/user-attachments/files/18558870/System-Daten_Zendure-SolarFlow_Hoymiles-HMS1600_OpenDTU-DPL_01.2-2025.pdf)

Die zugehörige Wiki Anleitung sollte entsprechend aktualisiert werden.
https://github.com/hoylabs/OpenDTU-OnBattery/wiki/Zendure-SolarFlow-with-Hoymiles-Inverter

Da bei der Verwendung von allen 4 Eingängen am WR offensichtlich die Regelung der MPPTs durcheinander gerät (gefühlt streiten sich hierbei der Ausgang des Hub mit dem Eingang des WR um die Regelung), ist das Setup so nicht nutzbar. 

Es müsste daher die Abbildung entsprechend der im PDF enthaltenen Version (Verwendung von nur 3 Eingängen am WR) angepasst werden.
Inhaltlich müsste alles mit Bezug auf die gleichzeitige Verwendung von 4 WR Eingängen entfernt werden.

Ich habe als Neueinsteiger gem. der im Wiki hinterlegten Informationen im April 2025 ein entsprechendes System beschafft, aufgebaut und eingerichtet. Das sich ergebende Fehlerbild habe ich zunächst einem BUG/Defekt der konkret eingesetzten Anlage zugeschrieben und über ein Ausschlussverfahren den Anschluss von nur zwei WR Eingängen als möglichen Workarround identifiziert.

Bitte passt die Wiki-Dokumentation entsprechend an, um weiteren solarinteressierten Neueinsteigern den Einstieg zu erleichtern.

Die Anleitung selbst ist super und hat mich nicht nur dazu bewogen das System nachzubauen, sondern war auch sehr hilfreich bei der Grundkonfiguration mit OpenDTUoB!
Vielen Dank an den Ersteller @https://github.com/Josef1966 und an @Manos1966!!!

Viele Grüße

> [@RicoBach68](https://github.com/RicoBach68) Hier gibts eine ganz gute Anleitung, passend zu deinem Setup: [#1473 (comment)](https://github.com/hoylabs/OpenDTU-OnBattery/pull/1473#issuecomment-2604680142)
> 
> @AndreasBoehm Ich würde - sofern von euch gewünscht - die Anleitung gerne im Wiki hinterlegen . Ich weiß nur nicht, wie man das (korrekt) macht - gibt es hier vielleicht irgendeine Beschreibung? Ich würde dann den anhängenden pdf-File dort hinterlegen und zukünftig pflegen/updaten.
> 
> [System-Daten_Zendure-SolarFlow_Hoymiles-HMS1600_OpenDTU-DPL_01.2-2025.pdf](https://github.com/user-attachments/files/18558870/System-Daten_Zendure-SolarFlow_Hoymiles-HMS1600_OpenDTU-DPL_01.2-2025.pdf)
>  

 _Originally posted by @Josef1966 in [#1551](https://github.com/hoylabs/OpenDTU-OnBattery/issues/1551#issuecomment-2615795541)_

Manos1966, 2025-06-16:
Hi @ff-81 

vielen Dank fuer die Info und Links !

Eine Frage habe ich noch: Was hast du unter "Stromquelle" eingestellt?

[image]

ff-81, 2025-06-16:
> [@ff-81](https://github.com/ff-81)
> 
> Ausgezeichnete Idee - da ich 1:1 das gleiche Setup habe - und die gleichen Probleme - würde ich gerne ebenso noch das Thema der Wechselrichter Firmware (PDL / mit einbringen wollen - die Geschichte mit 3/4 Eingängen hat vor der 2.0.4 funktioniert - allerdings ist die Regelung mit der neuen "DPL" Firmware weit besser. Mit der 2.0.4 aber nicht mehr. Somit ist man mit dem HMS-1600 und einem Zendure Solar Hub auf ~800W AC-Pout am Limit. Sofern jemand das anders zum Laufen bekommen hat - bitte um Info! :)
> 

Hi @lpg-elek ,

bin aktuell auf WR Firmware 2.0.4 und DTUoB Version 2025.06.11.
Nachdem vier Eingänge bereits bei der initialen Einrichtung nicht stabil funktioniert hatten, habe ich seitdem 2 Eingänge am WR angeschlossen. Die Konfiguration mit 3 belegten Eingängen habe ich nie getestet, da mir die ~800W Leistung zum WR hin aktuell ausreichen.
Ich meine mich zu entsinnen, dass der DPL zuletzt die definierte Einspeiseleistung nicht korrekt berechnen konnte, wenn bestimmte Werte der WR und Batteriekonfiguration nicht gepasst haben. Denke, das hängt mit der PDL seit 2.0.4 zusammen.
Seit ich statt 1600/1200 W nun 800 W für Limit Eingangs-/Ausgangsleistung konfiguriert habe, funktioniert es wie erwartet. Im Bereich Batterie-Einstellungen habe ich dabei 1200 W Wechselrichter Eingangsleistung belassen. Vielleicht  hilft es in Deinem Fall die Werte testweise an diesen Stellen auf 1200 W (bei drei Eingängen ) festlegen?

[image]

Ich habe auch gelesen, dass möglicherweise ein Downgrade der WR Firmware zumindest temporär hilfreich sein könnte: 
[Hoymiles Firmware 2.0.4 und die hängende Regelung](https://github.com/hoylabs/OpenDTU-OnBattery/discussions/1881#discussioncomment-13112776)

Meine DTU hat seit Umstellung auf Version 2025.06.11 noch Probleme mit wiederholten Abbrüchen der Wifi Verbindung. Die Ursache habe ich noch nicht  gefunden, hier hilft vorerst ein AP-seitig ausgelöster Reconnect. 
Netzteil tauschen und Temperatur-Überwachung werde ich als nächstes angehen. Wifi Roaming habe ich auch noch im Hinterkopf...
Logging habe ich bereits von Verbose auf Warning umgestellt. 

So long!

ff-81, 2025-06-16:
> Hi [@lpg-elek](https://github.com/lpg-elek) ,
>
> bin aktuell auf WR Firmware 2.0.4 und DTUoB Version 2025.06.11. Nachdem vier Eingänge bereits bei der initialen Einrichtung nicht stabil funktioniert [...]

@lpg-elek 
Hm okay, ich habe jetzt nochmal die Werte im Hub auf 1200W gestellt. Bisher funktioniert die 0-Einspeisung wie sie soll.
Leider ist der Akku gleich voll und erst jetzt gegen Nachmittag kommt der zweite String dazu. Ich hätte gerne gesehen, ob in Summe mit Einspeisung und Akkuladung mehr von den Panels in die Anlage kommt, als 8xx Watt. Sollte der Theorie nach ja möglich sein...
Die gestern festgestellten Schwankungen bei der Einspeisung bzw. dass diese keine Prio hatte und trotz reichlich Solarleistung Grid-Strom bezogen wurde, kann ich mir dann aktuell auch nicht wirklich erklären... 
Hoffe du findest einen Workaround !

[image]

ff-81, 2025-06-16:
> ergaenzt! 👍

Vielen Dank!

Manos1966, 2025-06-17:
@ff-81 Wenn die Wiki Aenderungen ausreichend sind, schliesse bitte den Fall.

ff-81, 2025-06-17:
> [@ff-81](https://github.com/ff-81) - Nachtrag - also ich schaffe mit Eingang 1&2 zusammen am HUB - etwas mehr als 800W AC Pout - mit 3 Eingägnen (2+3&4) - nicht mehr als 718W AC - siehe Bild. Ich wechsle nochmals die Leitungen - aber es scheint wohl der WR hier nicht zu wollen. Aber sonst können wir das da zumachen.
> 
> [image]

Das ist ja nicht so berauschend, es sollten doch eigentlich 1200W drin sein.
Vielleicht nochmal testweise ein FW Downgrade in Erwägung ziehen. Version 1.1.12 haben andere hier empfohlen: [Hoymiles Firmware 2.0.4 und die hängende Regelung](https://github.com/hoylabs/OpenDTU-OnBattery/discussions/1881#discussioncomment-13112776)

Auf jeden Fall scheint es den WR mit 3 angeschlossen MPPT nicht mehr zu diesem absolut untragbaren Fehlverhalten bei der Regelung zu führen, daher würde ich den konkreten Fall in Bezug auf die Aktualisierung der Anleitung schließen.

Danke für eure Hilfe und Input!

closed 2025-06-17 by ff-81

Josef1966, 2025-06-18:
Also bei mir läuft das System wie von mir im WIKI dargestellt mit Aufteilung auf die 4 Eingänge am HMS-1600 sehr gut.

Wenn es doch mal hängt, dann gehe ich wie folgt vor:

https://github.com/hoylabs/OpenDTU-OnBattery/discussions/1799#discussioncomment-12686618

Josef1966, 2025-06-18:
> > Also bei mir läuft das System wie von mir im WIKI dargestellt mit Aufteilung auf die 4 Eingänge am HMS-1600 sehr gut.
> > Wenn es doch mal hängt, dann gehe ich wie folgt vor:
> > [#1799 (reply in thread)](https://github.com/hoylabs/OpenDTU-OnBattery/discussions/1799#discussioncomment-12686618)
> 
> Hi [@Josef1966](https://github.com/Josef1966) - welche FW verwendest Du am 1600er? mit der 2.x hatte ich kein glück mehr. Es spießt sich die MPPT Regelung mit allen 4 oder auch 3 Eingägngen (ja, mit 3 gehts so "halb" - wenn man es eben messtechnisch genau nimmt, auch nicht) - ich habe es mit der Stromzange und dem Oszi angesehen - die MPPTs regeln gleichzeitig und somit schwingt das System - dumm gemacht - könnte man aber mit einer FW-Anpannungs im WR leicht behebn! Allerdings muss das vom Hersteller kommen...

FW: 1.1.12