hoylabs/OpenDTU-OnBattery #2051: [Request] add different setting for maximum inverter power (solar/battery)

SigmaSirio, 2025-06-16:
### Is your feature request related to a problem? Please describe.

It would be nice to be able to configure the inverter's maximum power when the source is only solar (during the day) and change it when the battery is used in the evening.
In this way, if you have a battery, that suffers from high discharge power, you can preserve it by lowering the maximum absorbed power.

### Describe the solution you'd like

example with my Hoymiles 700 Inverter
During the day I have more than 400w from solar panels, and respecting zero input, I would like to be able to use all the available power if required by the house. I currently have 300W set but could use more.
In the evening, if the battery is charged, I would like to be able to lower the maximum power of the inverter to 200W so as not to stress the battery with too much discharge power. Using 300W now, it is still a bit high.

I'm not sure how I can handle the moments when the battery is charged but it gets cloudy or starts to come evening. You would need to identify a trigger that will or will not activate the new setting.
You probably already have some triggers running.

SigmaSirio, 2025-06-16:
I could configure it only by mqtt, Volt Ampere with positive values. My shut only returns positive values.
how does the limitation then act?

SigmaSirio, 2025-06-19:
At the moment I only tried to set the discharge limit, like your screen, but without configuring the reading of battery values via mqtt, I don't know if they are mandatories, but it doesn't work with this simple setting.
as soon as I can I will try to configure the reading of shunt values via mqtt as well.

schlimmchen, 2025-06-19:
@spcqike is right: Limiting the battery discharge current is the way to go here. I don't understand why that is not working for you? If you set a discharge limit and the inverter is drawing more power than expected, please observe and provide a log while the issue presents itself.

SigmaSirio, 2025-06-20:
sorry for the slow responses but to test, I have to get to the evening every time.
I confirm that the limit only in the battery menu does not work with 5A max (200W)
But I have a doubt, is the limit current calculated for each single channel of the inverter or are they added together?
Now I have set 2 A maximum, so less than half of what I would need, my inverter has 2 channels. 40V x 4A = 160W. Let's see how it behaves this evening.
Remember, I have not configured the mqtt to read the battery data.

SigmaSirio, 2025-06-22:
Websocket: [/console][2] connect
18:03:33.927 > [DPL] up 13106 s, it is day, next inverter restart at 52512 s (set to 5)
18:03:33.927 > [DPL] battery interface disabled, SoC 0.0 % (ignored), age 13106 s (stale)
18:03:33.927 > [DPL] BMS: -1.00 V, MPPT: 39.17 V, inverter 114182977390: 38.60 
18:03:33.927 > [DPL] battery voltage 39.17 V, load-corrected voltage 39.47 V @ 305 W, factor 0.00100 1/A
18:03:33.927 > [DPL] battery discharge allowed, start 40.20 V or 80 %, stop 37.70 V or 20 %
18:03:33.927 > [DPL] full solar-passthrough dormant, start 60.00 V or 100 %, stop 60.00 V
18:03:33.927 > [DPL] start NOT reached, stop NOT reached, solar-passthrough enabled, use at night disabled and dormant
18:03:33.980 > [DPL] total max AC power is 600 W, conduction losses are 0 %
18:03:33.980 > [DPL] targeting 30 W, base load is 120 W, power meter reads 632.7 W (valid)
18:03:33.980 > [DPL] granting 600 W from DC power bus (no battery discharge limit), solar power is 37/35 W DC/AC
18:03:33.980 > [DPL] requesting 600 W from 1 battery-powered inverter currently producing 305 W (diff 295 W, hysteresis 20 W)
18:03:33.980 > [DPL] will cover 305 W using 1 battery-powered inverter
18:03:33.980 > [DPL inverter 114182977390]:
18:03:33.980 >     battery-powered, producing 305 W, output excluded from power meter reading
18:03:34.789 >     lower/current/upper limit: 30/299/300 W, output capability: 700 W
18:03:34.789 >     sending commands enabled, reachable, eligible
18:03:34.789 >     max reduction production/standby: 275/305 W, max increase: 1 W
18:03:34.789 >     target limit/output/state: -1 W (unchanged)/305 W/unchanged, 0 update timeouts
18:03:34.789 >     MPPTs AC power/DC voltage: a: 153 W/38.6 V b: 153 W/38.6 V
18:03:34.789 > [PowerMeters::Json::Http] New total: 632.95
18:03:34.789 > Websocket: [/livedata][3] disconnect
18:03:34.943 > [DPL] up 13107 s, it is day, next inverter restart at 52512 s (set to 5)
18:03:34.943 > [DPL] battery interface disabled, SoC 0.0 % (ignored), age 13107 s (stale)
18:03:34.943 > [DPL] BMS: -1.00 V, MPPT: 39.18 V, inverter 114182977390: 38.60 
18:03:34.943 > [DPL] battery voltage 39.18 V, load-corrected voltage 39.49 V @ 305 W, factor 0.00100 1/A
18:03:34.943 > [DPL] battery discharge allowed, start 40.20 V or 80 %, stop 37.70 V or 20 %
18:03:34.943 > [DPL] full solar-passthrough dormant, start 60.00 V or 100 %, stop 60.00 V
18:03:34.943 > [DPL] start NOT reached, stop NOT reached, solar-passthrough enabled, use at night disabled and dormant
18:03:34.981 > [DPL] total max AC power is 600 W, conduction losses are 0 %
18:03:34.981 > [DPL] targeting 30 W, base load is 120 W, power meter reads 633.0 W (valid)
18:03:34.981 > [DPL] granting 600 W from DC power bus (no battery discharge limit), solar power is 43/40 W DC/AC
18:03:34.981 > [DPL] requesting 600 W from 1 battery-powered inverter currently producing 305 W (diff 295 W, hysteresis 20 W)
18:03:34.981 > [DPL] will cover 305 W using 1 battery-powered inverter
18:03:34.981 > [DPL inverter 114182977390]:
18:03:34.981 >     battery-powered, producing 305 W, output excluded from power meter reading
18:03:35.598 >     lower/current/upper limit: 30/299/300 W, output capability: 700 W
18:03:35.598 >     sending commands enabled, reachable, eligible
18:03:35.598 >     max reduction production/standby: 275/305 W, max increase: 1 W
18:03:35.598 >     target limit/output/state: -1 W (unchanged)/305 W/unchanged, 0 update timeouts
18:03:35.598 >     MPPTs AC power/DC voltage: a: 153 W/38.6 V b: 153 W/38.6 V
18:03:35.598 > Fetch inverter: 114182977390
18:03:35.675 > Queue size - NRF: 1 CMT: 0
18:03:35.675 > TX RealTimeRunData Channel: 23 --> 15 82 97 73 90 80 11 54 76 80 0B 00 68 55 86 5B 00 00 00 00 00 00 00 00 9C B8 1F 
18:03:35.675 > Interrupt received
18:03:35.675 > RX Channel: 61 --> 95 82 97 73 90 82 97 73 90 01 00 01 01 82 01 9E 06 3C 01 82 01 9E 06 3D 00 05 91 | -80 dBm
18:03:35.675 > Interrupt received
18:03:35.675 > RX Channel: 61 --> 95 82 97 73 90 82 97 73 90 02 65 EA 00 03 26 71 02 D6 02 C9 08 A5 13 8E 0B E9 81 | -80 dBm
18:03:35.675 > Interrupt received
18:03:35.675 > RX Channel: 40 --> 95 82 97 73 90 82 97 73 90 83 00 03 00 8A 03 E8 01 F9 00 0F FD 4E 30 | -80 dBm
18:03:35.962 > [DPL] up 13108 s, it is day, next inverter restart at 52512 s (set to 5)
18:03:36.033 > [DPL] battery interface disabled, SoC 0.0 % (ignored), age 13108 s (stale)
18:03:36.033 > [DPL] BMS: -1.00 V, MPPT: 39.16 V, inverter 114182977390: 38.60 
18:03:36.033 > [DPL] battery voltage 39.16 V, load-corrected voltage 39.47 V @ 305 W, factor 0.00100 1/A
18:03:36.033 > [DPL] battery discharge allowed, start 40.20 V or 80 %, stop 37.70 V or 20 %
18:03:36.033 > [DPL] full solar-passthrough dormant, start 60.00 V or 100 %, stop 60.00 V
18:03:36.033 > [DPL] start NOT reached, stop NOT reached, solar-passthrough enabled, use at night disabled and dormant
18:03:36.033 > [DPL] total max AC power is 600 W, conduction losses are 0 %
18:03:36.068 > [DPL] targeting 30 W, base load is 120 W, power meter reads 633.0 W (valid)
18:03:36.069 > [DPL] granting 600 W from DC power bus (no battery discharge limit), solar power is 43/40 W DC/AC
18:03:36.069 > [DPL] requesting 600 W from 1 battery-powered inverter currently producing 305 W (diff 295 W, hysteresis 20 W)
18:03:36.069 > [DPL] will cover 305 W using 1 battery-powered inverter
18:03:36.069 > [DPL inverter 114182977390]:
18:03:36.069 >     battery-powered, producing 305 W, output excluded from power meter reading
[... cut ...]
18:03:44.203 > [DPL] start NOT reached, stop NOT reached, solar-passthrough enabled, use at night disabled and dormant
18:03:44.248 > [DPL] total max AC power is 600 W, conduction losses are 0 %
18:03:44.248 > [DPL] targeting 30 W, base load is 120 W, power meter reads 640.4 W (valid)
18:03:44.248 > [DPL] granting 600 W from DC power bus (no battery discharge limit), solar power is 43/40 W DC/AC
18:03:44.248 > [DPL] requesting 600 W from 1 battery-powered inverter currently producing 308 W (diff 292 W, hysteresis 20 W)
18:03:44.248 > [DPL] will cover 308 W using 1 battery-powered inverter

AndreasBoehm, 2025-06-27:
> I can't find a way to verify that it can actually read the values. how can I do it?

Good hint. The DTU does not get your battery values. **BMS: -1.00 V**

```
[DPL] BMS: -1.00 V, MPPT: 39.18 V, inverter 114182977390: 38.60 
```

You should check the homepage of the DTU, the live view of the battery should be there. Here an example from my MQTT battery.

[image]

closed 2025-06-27 by AndreasBoehm

SigmaSirio, 2025-06-27:
I don't see what you see.
what I see about the battery is this but it looks like it's what the victron 100/20 sees and not the shunt via mqtt.

[image]

can you tell me if the syntax of the mqtt configuration is correct? since you also read from mqtt, do you use home assistant or read the battery directly?
I don't like to go by trial and error.

AndreasBoehm, 2025-06-27:
If you don't see the card `Battery: MQTT` your mqtt topics and/or JSON paths are wrongly setup.
I can not help you with that as i don't know your setup and can't tell you what to put there.
Can you share a screenshot of [mqtt-explorer](https://mqtt-explorer.com/) that shows your topics and the contents of the relevant topics?

> anyway, the limit should work anyway without the shunt reading from what you told me above.

hmm, from looking at the code this will not work without know the batteries voltage.

@schlimmchen should this work without having a battery configured? According to the screenshot https://github.com/hoylabs/OpenDTU-OnBattery/issues/2051#issuecomment-2976244816 you can enable the limit without enabling the battery interface itself.

SigmaSirio, 2025-06-27:
All clear

I will check the mqtt configuration for the shunt better.

I can confirm that in the past few days when the battery was charged, the limit was not respected.
I agree with you that even if it were not precise, it would be fine anyway, in the end the inverter knows how much it is consuming and should limit itself based on the A configured in that screen. It makes sense to me too.
I hope to give you more details this evening during the discharge. I will leave everything configured as above if you do not have other advice to give me for other tests.

SigmaSirio, 2025-06-27:
uhmm?? why was the topic closed?

AndreasBoehm, 2025-06-28:
In case you can't get your mqtt battery to work you can use the test-build from https://github.com/hoylabs/OpenDTU-OnBattery/pull/2083#issuecomment-3012974539 where i fixed the issue that the discharge limit does not work without the BMS providing it.

SigmaSirio, 2025-07-02:
There was no way I could read the data from mqtt from the shunt with Tasmota.
So I used the volts read from the Victron which has the single value and not a json string. It worked the first time.
I also deleted the reading of the Amps from the shunt leaving ONLY the volts from the Victron
By doing this I was able to verify that the battery discharge limit seems to work perfectly.
Thanks