hoylabs/OpenDTU-OnBattery #2058: Power Meter HTTP Error: read Timeout after upgrade

SvenSowa, 2025-06-17:
### What happened?

After upgrading to the latest build 2025.06.11, I am getting frequent http timeouts when reading the grid power. The power limiter will fall back to the base load and I will consume more power from the grid. This happens every other minute or within the next 5 minutes or so. Typically there is 4 timeouts in a row. I am using the http+json power meter type with a 1 second polling interval and 500ms timeout (Hichi Tasmota IR WIFI sensor). After rolling back to 2025.04.17 the timeout issue was gone. I have a second device, where I have 2025.06.11 installed. After setting the same polling interval and timeout, he is showing the same issue.

[image]

To visualize it even better, I compared last night with the night before (2025.06.11 versus 2025.04.17). You will see that the timeouts led to a very spiky regulation last night.

[image]

[image]

Even upgrading to the latest development commit [https://github.com/hoylabs/OpenDTU-OnBattery/commits/f1ee860](url) doesn't solve the problem:

[image]

My Tasmota device responds within 120ms typically. I did run some low level raw http calls in a loop for a while. No errors there.

### To Reproduce Bug

Upgrade to 2025.06.11

### Expected Behavior

Should work like 2025.04.17

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.06.11

### What firmware variant (PIO Environment)?

generic_esp32_4mb_no_ota

### Relevant log/trace output

```Shell
See screenshot
```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

AndreasBoehm, 2025-06-25:
Please try to change the logging settings under `Settings -> Logging` to `Error` or `None` for everything but the PowerMeter and let me know if this helped.

SvenSowa, 2025-07-01:
I did a longer test today and here are the results:

[image]

[image]

[image]

[opendtu-log-2025-07-01_09-56-21.txt](https://github.com/user-attachments/files/20994196/opendtu-log-2025-07-01_09-56-21.txt)

SvenSowa, 2025-07-01:
I see your point. I could switch to mqtt, but don't want a dependency on it really. My tasmota device works fine. I can poll it constantly from another computer and it responds within 80-120ms while poking it constantly for minutes. No errors there. I will do some more investigation.

closed 2025-10-16 by github-actions[bot]