hoylabs/OpenDTU-OnBattery #2065: Stromzählerwert friert plötzlich komplett ein, keine Regelung mehr möglich

Andrix82, 2025-06-19:
### What happened?

-Stromzählerwert (HTTPS+JSON von Shelly 3EM) mit aktuellstem Release ist gerade komplett eingefroren bei einem Wert von -1059,3W
-Verbose Logging vom Stromzähler liefert keinerlei Outputs
-Verbose Logging vom DPL zeigt dass dieser stets mit dem eingefrorenen Wert rechnet
-Einstellungen Tab vom Stromzähler lässt sich öffnen und der HTTP Test liefert stets dynamische korrekte Werte.
-Zustand blieb mehrere Minuten so 
-Hat sich eben nun ohne Konfig-Änderungen meinerseits und ohne SW-Reset laut Betriebszeitzähler von selbst wieder gelöst
-Verbose Logging vom Stromzähler liefert wieder Konsole-Outputs

### To Reproduce Bug

Noch unklar, erstes mal so beobachtet seit über einem Jahr

### Expected Behavior

Stromzählerwert friert nicht ein

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.06.11

### What firmware variant (PIO Environment)?

generic_esp32s3

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

Andrix82, 2025-06-19:
#2058 dürfte wohl ähnliche Beobachtung / Problem sein, auch wenn im Detail andere Auswirkung

stefan-moosi, 2025-06-22:
Hallo. Habe das gleiche Problem. Ich habe die OpenDTU  vor 2 Tagen geliefert bekommen und ich muss sie einmal am Tag Stromlos machen damit sie wieder funktioniert. Die Verbrauchsdaten vom 3em werden im Display noch angezeigt aber die dynamische Anpassung funktioniert nicht und via IP Adresse ist auch keine Verbindung über das LAN möglich.

AndreasBoehm, 2025-06-25:
@Andrix82 @Spawn-8-4 Könnt ihr mir verraten was ihr unter Einstellungen -> Protokollierung konfiguriert habt?

Spawn-8-4, 2025-06-25:
Hallo. Ich hab Standard E Fehler, der Rest wird vom Standard vererbt.

Andrix82, 2025-06-25:
Kann es leider nicht mehr genau sagen wie die Prorokollierungseinstellungen zu dem Zeitpunkt waren, gehe aber davon aus dass der Standard W oder E eingestellt wurde und fast alle diesen geerbt haben.
Das detaillierte Logging (V) als Standard musste ich ja gleich umstellen da ich sonst kaum Funkverbindung zum Wechselrichter hatte.
Das Phänomen habe ich bisher aber auch nur dieses eine mal festgestellt.

AndreasBoehm, 2025-06-26:
@Hona91 welche Wechselrichter Firmware nutzt du?

Hona91, 2025-06-27:
Gut zu wissen, welche Version benötige ich dann?
 Als ich den Wechselrichter erhalten habe, wollte ich eben bei Hoymiles ein Update damit er Power Distribution kann.

AndreasBoehm, 2025-06-27:
@Hona91 1.1.12

andreas70, 2025-07-02:
Seit dem Update auf 2025.06.11 vor 2 Tagen hatte ich jetzt leider auch schon zweimal einen Komplettabsturz der Verbindung zum Zähler. 

Zähler ist ein IR Lesekopf mit Tasmota, eingebunden über HTTP(S) + JSON. Bei Störung geht die Anzeige Leistung /Netz auf 0 und die WR auf Grundlastleistung. Test der Zählerverbindung bringt dann nach der Timeoutzeit "connetion refused"(?!). Im Browser funktioniert die HTTP Anfrage (http://<IP Adresse>/cm?cmnd=status%208) und bringt die korrekten Werte. Im IoBroker auch keine Verbindungsstörung zum Zähler, Tasmota Webinterface auch einwandfrei.

Nach Neustart über Webinterface läuft dann alles wieder einwandfrei.

Logs hab ich momentan leider keine, konnte noch kein Muster ausfindig machen wann das auftritt.