hoylabs/OpenDTU-OnBattery #2068: [Request] Automatic Inverter Reset in case of serious/problematic errors

Andrix82, 2025-06-20:
### Is your feature request related to a problem? Please describe.

Since some weeks ( don't think related to openDTU SW update, more to maybe Inverter aging OR mostly
 that I am in full solar passthrough mode frequently) I encounter daily that the inverter (HMT2250) got stuck in a limited power mode after the error "FB Überstrom(FB Overcurrent)" occurred:

[image]

In this mode the Outputpower is limited to real measured ~1,6kW ( so DC Input Power shown in picture is plausible ) and the AC-Output Power is reported with high deviation only ~1kW.

This state could only and always be left with an inverter reset.

"FB Overcurrent" seems to be related to the DC input current.

As every user with a Battery connected to the inverter DC input is operating it highly above the rated max input current I assume this is possibly affecting every user of openDTUonBattery.

### Describe the solution you'd like

Feature to automatically reset the inverter when selectable error codes were reported by the inverter.

spcqike, 2025-06-20:
> Describe alternatives you've considered

Restart inverter whenever it’s efficiency is < 90% for 1 minute. Or 5. or whatever. 

Until now I saw such low efficiency only with 24V battery systems every now and then, when the shunts saturated (over 300W)

Andrix82, 2025-06-21:
Sry there is maybe a missunderstanding but this is not an isolated hardware issue like defect or similar, it is not related to my cable connections.
It is a possible concept issue of operating an solar panel inverter on a high current input source like batteries that FB overcurrent errors can occour.
Maybe the error is only coming at max inverter loads over longer period(full solar passthgrough) with HMT2250 and high Battery capacity (15kWh)
When they occour the inverter can not operate normally as can be seen in this case, a reset solves it.

When I am the only user with this error reported then its mostly really limited to the surroundings written above , then I fully understand and agree its not worth your precious time and ressources to bring this feature, but I would maybe keep it open and under investigation since the error is more or less "to be expected" with a battery setup and could come up for other users or i.e. with newer inverter models/firmwares more often.

Thanks!

spcqike, 2025-06-23:
the screen should possible show a high production. like 90% of the limit or more, if possible :)

schlimmchen, 2025-06-24:
> It is a possible concept issue of operating an solar panel inverter on a high current input source like batteries that FB overcurrent errors can occour.

OpenDTU-OnBattery has thousands of users, most of them with a battery in their system, and I cannot remember reading from one of them that a FB overcurrent situation would be considered as "something that just happens in these setups".

> I advise to disassemble the cabling from the battery to the inverter and carefully check all the connections and maybe swap the cables or crimp them new. Especially if the error doesn't occur at low power, only at high, it might be a heating bad connection worsens and gives rise to such weird errors.

Yes.

> Cable inputs and MC4 Connectors are all fine cool also in FullSolarPassthrough over hours and symmetrical ( ~10cm 4mm² cables to the very same 63A clampblock )

Too thin. And the common terminal ("clampblock") you mention can very well have loose bolts. Trust us. I had the same issue. Couldn't believe it until I re-tightened all of the bolts and the asymmetry mostly went away.

We are not fixing hardware issues in software.

closed 2025-07-23 by AndreasBoehm