hoylabs/OpenDTU-OnBattery #2096: [Bug] Webinterface not reachable while inverter feedback loop still works very slowly (minutes instead of seconds)

gitisgreat2023, 2025-07-04:
### What happened?

This morning I noticed the the feedback loop didn't work, i.e. several minutes 2kW were drawn from the grid, very late the inverters started producing. Then the load went away but the inverters produced several minutes more too much (about 2kW). It was like, working, but with a delay of some minutes.

I thought, okay, let's have a look, start the log to figure out what is going on, are the inverters etc not reachable. However, the webinterface wasn't reachable! It was still running, the inverter power in the end still changed (readings via MQTT), so it was still functioning. 

Does anybody recognize this issue? What could be the cause? Under which circumstances can the web interface not be reached? I remember having it once in the past but though that it was a quirk. It wasn't a DNS resolve issue, I tried reaching the webinterface using the IP address as well. A network scanner also showed it as alive.

Wouldn't it an idea to add a write MQTT value in dtu which for value 1 reboots the DTU? After rebooting it worked again like a charm!
I would use that MQTT field to build in my GUI a button "reboot DTU" in case I cannot reach it. (using a smart plugin would be even more robust but is kind of overkill IMHO)

### To Reproduce Bug

Good question

### Expected Behavior

Web interface can be reached under all circumstances.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.06.11

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

AndreasBoehm, 2025-07-10:
@gitisgreat2023 Sounds similar to issue https://github.com/hoylabs/OpenDTU-OnBattery/issues/2098
What do you think? Could it be the same cause? Are you using a huawei charger?

gitisgreat2023, 2025-07-10:
I also use a Huawei yes. Though the Web-UI wasn't reachable. MQTT was still and with minutes delay also the inverters.

A solution would be to add a MQTT field to reboot the DTU once a value other than 0 is written.

AndreasBoehm, 2025-09-16:
In der aktuellen Release Version wurden ein paar Fehler behoben die einen ungewollten Neustart oder Aussetzer verursacht haben.

Bitte installier das aktuellste Release und gib Bescheid ob sich dein Problem dadurch erledigt hat.

https://github.com/hoylabs/OpenDTU-OnBattery/releases/tag/2025.09.16