hoylabs/OpenDTU-OnBattery #2134: [Request] Support Zendure SolarFlow 2400 AC

Birndlmaier, 2025-07-25:
### Is your feature request related to a problem? Please describe.

Currently, zero feed-in is not possible with the Open-DTU on Battery and the 2400 AC storage system. When the battery is full, the bypass is activated and all the power produced by the Hoymiles HMS2000, for example, is fed into the grid.

### Describe the solution you'd like

Ideally, there would be a function that checks whether the battery is full and whether the house otherwise needs any electricity.

AndreasBoehm, 2025-08-08:
@vaterlangen thats for you :)

AndreasBoehm, 2025-08-08:
Or maybe not?

Its a AC coupled battery, there is no such thing called bypass. The HMS-2000 is delivering 100% of its power and when the battery is full the surplus power will be fed into the grid.

IMHO thats easy to solve, you enable the DPL for zero feed-in and set it to e.g. -100, the AC battery should be set to e.g. 0.
As soon as you start exporting power the AC battery will start charging, feed-in will drop to 0, DPL will then increase the power of the HMS to export -100, this happens over and over until the HMS is at 100%. When the battery is full the DPL will decrease the power of the HMS to cover your current need and feed-in 100 watts.

closed 2025-08-08 by AndreasBoehm