hoylabs/OpenDTU-OnBattery #2141: [Bug] Allow a larger value for Maximum Output Power for Huawei R4850G2

gitisgreat2023, 2025-07-30:
### What happened?

As I also control my Huawei using MQTT I noticed that I could charge at "true" 3000W. First then I realized why previously with automatic power control it doesn't reach 3000W, it seems to assume a too high efficiency.

Running it at 55A and 53.35V it's efficiency is 94.63%. An elegant solution would be to divide the 3000W through 0.9463, that allows for 3170W input power and 3000W output power. Of course for lower wattages the 94.63% isn't correct.

Second solution would be to allow a higher value, instead of 3000W 3170W as maximum. At the moment I just changed the config.json manually...

[image]

### To Reproduce Bug

Set the max value of the Huawei to 3000 and look which value it actually can output.

### Expected Behavior

With max 3000W output I expected it allows for 3000W output.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.06.11

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

AndreasBoehm, 2025-08-08:
Please try the test version from the PR https://github.com/hoylabs/OpenDTU-OnBattery/pull/2142#issuecomment-3166830718 and let me know if that fixes your issue.

Manos1966, 2025-08-10:
Hi @AndreasBoehm 
kurze Verstaendnissfrage, aus eigenem Interesse, weil ich einen R4875G1 auf dem Weg habe (4000W max Leistung):
Ist ein Max Limit von 3000W zur Zeit "fest geschrieben" ?

AndreasBoehm, 2025-08-11:
@gitisgreat2023 Thank you!
@Manos1966 Aktuell ist das limit auf 3000W begrenzt, das ist für ein R4875 natürlich ungünstig. Sollten wir einfach erhöhen können.