hoylabs/OpenDTU-OnBattery #2144: Nachts werden komische MQTT Werte (UNDEF) an Openhab gesendet

Homer-Sim, 2025-08-02:
### What happened?

Ich habe Mitte Juli (genau am 17.7.) auf die 2025.06.11 geupdated. 
Erst vor ein paar Tagen habe ich bemerkt, dass es Probleme gibt. Vorher gab es diese Probleme nicht.
An meiner OpenDTU hängen je zwei HMS-800 und je zwei HMT-2250. 

Ich habe einen HMS-800 der seit genau dem Update in der Nacht via ODoB an openhab UNDEF Werte liefert und somit zu den Geraden im Diagramm führt, weil die Rechnungen abgrebrochen werden. 
Seit dieser version klappt bei einem HMT-2250 auch die Nullsetzung um Mitternacht nicht mehr. Da hab ich ein Workaround gefunden (24:00 reset des Inverters per MQTT). Das hatte ich aber nicht in Verbindung mit dem Updatre gebracht. 
[image]

Leider weiß ich nicht mehr von welcher Version ich kam, aber es war eine 2024. 
Habt ihr eine Idee ab welccher Version der Fehler gekommen sein könnte zu welcher soll/kann ich zurück?

Danke

### To Reproduce Bug

An meiner OpenDTU hängen je zwei HMS-800 und je zwei HMT-2250. 
Die Fehler treten aber nur an je einer der Beiden auf, also eine HMS-800 und eine HMT-2250

### Expected Behavior

Entweder soll die OpenDTU "0" senden oder keine Werte, wenn vom WR komischer Werte (UNDEF in openhab)

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.06.11

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

SW-Niko, 2025-08-03:
Hallo @Homer-Sim ,
Momentan gibt es ein Problem mit zu vielen Protokolleinträgen.
Du kannst mal probieren alle Module auf Stufe "Keine" oder "(E) Fehler" zu konfigurieren.
Das reduziert die Anzahl der Einträge und löst vielleicht auch dein Problem.

Homer-Sim, 2025-08-03:
Hallo @SW-Niko,
Danke für deine Antwort. Wo soll ich das einstellen? Ist das bei den WR unter String? 
Da finde ich das aber nicht so in der Form 
Danke

SW-Niko, 2025-08-03:
Unter Einstellungen -> Protokollierung falls du die WebUI auf deutsch eingestellt hast.
ansonsten Settings -> Logging

[image]

spcqike, 2025-08-03:
wie sieht es denn mit deiner Laufzeit der openDTU aus? Läuft diese stabil durch oder stürtzt sie hier und da ab/startet neu?

Homer-Sim, 2025-08-03:
@spcqike 
bisher lief die ohne Probleme komplett durch. ich habe es nur gemerkt als Werte in Diagrammen, die berechnetwurden von der Leistung eines WR an dem Speicher hängen nicht mehr "gute Werte" geliefert haben. Tagsüber läuft alles ohne Fehl und Tadel

closed 2025-09-02 by github-actions[bot]