hoylabs/OpenDTU-OnBattery #2174: Support Multiple AC Charger

Cartman586, 2025-08-27:
### Is your feature request related to a problem? Please describe.

Only one charger is supported.

### Describe the solution you'd like

I am currently using the alpha version. I am using the T2HG/T2MG charger interface. However, I have two of these chargers but can only control one. I would need this, please. A Multiple Chrager support would be great.

Cartman586, 2026-03-13:
Guten Morgen, ich wollte mich kurz nach dem Stand zu diesem Thema erkundigen. Vielen Dank

Snoopy-HSS, 2026-10-19:
habe das Thema für AC Charger via HTTP GET umgesetzt und ist im Pull Request https://github.com/hoylabs/OpenDTU-OnBattery/pull/2347