hoylabs/OpenDTU-OnBattery #2178: [Request] Victron (Carlo Gavazzi) EM24 Ethernet

dAjaY85, 2025-09-01:
### Is your feature request related to a problem? Please describe.

Is it possible to integrate the EM24 electricity meter as an additional module?

### Describe the solution you'd like

The EM24 is integrated in a similar way to the VP-3P75CT.

Sorbat, 2025-09-01:
Hi, was heißt denn ähnlich? Was ist gleich und was genau ist anders?

spcqike, 2025-09-01:
> Hast du das Gerät bereits?

Wenn nicht, würde ich auch einfach eines holen, welches bereits unterstützt wird (:

dAjaY85, 2025-09-02:
Das ist meine Integration des Modbus Stromzählers in Homeassistant, damit kann man die wichtigsten Werte in Homeassistant anzeigen lassen.

modbus:
  - name: EM24_Modbus
    type: tcp
    host: 192.168.178.xx
    port: 502
    sensors:
      ###. EM24-E1 ###
      - name: EM24_V_L1N
        unit_of_measurement: V
        scan_interval: 5
        precision: 2
        address: 0x0000
        input_type: input
        slave: 1
        scale: 0.1

      - name: EM24_V_L2N
        unit_of_measurement: V
        scan_interval: 5
        precision: 2
        address: 0x0002
        input_type: input
        slave: 1
        scale: 0.1

      - name: EM24_V_L3N
        unit_of_measurement: V
        scan_interval: 5
        precision: 2
        address: 0x0004
        input_type: input
        slave: 1
        scale: 0.1

      - name: EM24_P_L1N
        unit_of_measurement: W
        slave: 1
        address: 0x0012
        swap: word
        data_type: int32
        scale: 0.1
        precision: 2
        device_class: power
        input_type: input
        scan_interval: 1

      - name: EM24_P_L2N
        unit_of_measurement: W
        slave: 1
        address: 0x0014
        swap: word
        data_type: int32
        scale: 0.1
        precision: 2
        device_class: power
        input_type: input
        scan_interval: 1

      - name: EM24_P_L3N
        unit_of_measurement: W
        slave: 1
        address: 0x0016
        swap: word
        data_type: int32
        scale: 0.1
        precision: 2
        device_class: power
        input_type: input
        scan_interval: 1

      - name: EM24_P_SUM
        unit_of_measurement: W
        slave: 1
        address: 0x0028
        swap: word
        data_type: int32
        scale: 0.1
        precision: 2
        device_class: power
        unique_id: Total_Power
        input_type: input
        scan_interval: 1

      - name: EM24_to_Grid
        unit_of_measurement: KWh
        slave: 1
        address: 0x004E
        swap: word
        data_type: int32
        scale: 0.1
        precision: 2
        device_class: power
        unique_id: to_grid
        input_type: input
        scan_interval: 60

      - name: EM24_from_Grid
        unit_of_measurement: KWh
        slave: 1
        address: 0x0034
        swap: word
        data_type: int32
        scale: 0.1
        precision: 2
        device_class: power
        unique_id: from_grid
        input_type: input
        scan_interval: 60

      - name: EM24_Frequency
        unit_of_measurement: Hz
        slave: 1
        address: 0x0033
        input_type: input
        data_type: uint16
        scale: 0.1
        precision: 2
        device_class: frequency
        unique_id: Frequency
        scan_interval: 1

dAjaY85, 2025-09-05:
Ja ein universeller modbus Stromzähler würde Sinn machen, sind ja einige die mit modbus Daten versenden.