hoylabs/OpenDTU-OnBattery #2216: DPL Springt auf 100% wenn ein Eingang weg fällt

KarlHeinzz1, 2025-09-13:
### What happened?

Wenn einer der Speicher leer ist und der Eingang gegen 10V oder darunter geht, springt der DPL auf 100% und bleibt bei 100%. Ein Neustart der WR hilft nur kurz. Was hilft ist "verschattetet Eingänge ausgleichen" aus zu machen, aber dann wird natürlich auch falsch geregelt.

WR HMS2000 und Firmware 1.0.27 - Verschattete Eingänge ausgleichen ist an
3x B2500 Speicher an den 4 Eingängen des HMS2000 (einer an 2 Eingängen) + ein Speicher am HM600

[image]

[image]

### To Reproduce Bug

Einen Eingang weg nehmen

### Expected Behavior

Es soll weiter richtig geregelt werden auf Zielwert

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

d2e0ad0

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell
[09:12:22.569] W (171533869) hoymiles: Nothing received, resend count exeeded
[09:12:22.569] I (171533968) hoymiles: Fetch inverter: 114184216052
[09:12:22.852] I (171533968) hoymiles: Request SystemConfigPara
[09:12:22.852] I (171533969) hoymiles: Queue size - NRF: 2 CMT: 0
[09:12:22.852] D (171533973) hoymiles: TX RealTimeRunData Channel: 23 --> 15 84 21 60 52 80 14 47 32 80 0B 00 68 C5 19 55 00 00 00 00 00 00 00 00 B3 3F 85
[09:12:22.852] D (171534248) powerMeter: [HTTP/JSON] New total: -1074.00
[09:12:22.901] D (171534249) dynamicPowerLimiter: [Controller] up 171534 s, it is day, next inverter restart at 224825 s (set to 0)
[09:12:22.901] D (171534249) dynamicPowerLimiter: [Controller] targeting 0 W, base load is 75 W, power meter reads -1074.0 W (valid)
[09:12:22.901] D (171534250) dynamicPowerLimiter: [Controller] requesting 151 W from 1 smart-buffer-powered inverter currently producing 1225 W (diff -1074 W, hysteresis 10 W)
[09:12:22.901] D (171534251) dynamicPowerLimiter: [Inverter 1164a00ba154] expected AC power per MPPT 485 W
[09:12:22.901] D (171534253) dynamicPowerLimiter: [Inverter 1164a00ba154] 3/4 mppts are not-producing/shaded, scaling 2000 W
[09:12:22.954] D (171534253) dynamicPowerLimiter: [Controller] will cover 151 W using 1 smart-buffer-powered inverter
[09:12:22.954] D (171534253) dynamicPowerLimiter: [Controller] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 0 W AC
[09:12:22.954] V (171534253) dynamicPowerLimiter: [Inverter 1164a00ba154] State Details
[09:12:22.954] V (171534253) dynamicPowerLimiter: [Inverter 1164a00ba154]     smart-buffer-powered, producing 1225 W, output included in power meter reading
[09:12:22.955] V (171534253) dynamicPowerLimiter: [Inverter 1164a00ba154]     lower/current/upper limit: 40/2000/2100 W, output capability: 2000 W
[09:12:23.058] V (171534254) dynamicPowerLimiter: [Inverter 1164a00ba154]     sending commands enabled, reachable, eligible
[09:12:23.058] V (171534254) dynamicPowerLimiter: [Inverter 1164a00ba154]     max reduction production/standby: 1185/1185 W, max increase: 0 W
[09:12:23.058] V (171534254) dynamicPowerLimiter: [Inverter 1164a00ba154]     target limit/output/state: 2000 W (update)/151 W/production, 0 update timeouts
[09:12:23.058] V (171534255) dynamicPowerLimiter: [Inverter 1164a00ba154]     MPPTs AC power/DC voltage: a: 507 W/38.1 V b: 0 W/9.1 V c: 345 W/37.7 V d: 372 W/36.4 V
[09:12:23.111] V (171534255) dynamicPowerLimiter: [Inverter 114184216052] State Details
[09:12:23.111] V (171534255) dynamicPowerLimiter: [Inverter 114184216052]     smart-buffer-powered, standing by at 0 W, output included in power meter reading
[09:12:23.111] V (171534256) dynamicPowerLimiter: [Inverter 114184216052]     lower/current/upper limit: 40/45/700 W, output capability: 600 W
[09:12:23.111] V (171534256) dynamicPowerLimiter: [Inverter 114184216052]     sending commands enabled, offline, disqualified (unreachable)
[09:12:23.111] V (171534256) dynamicPowerLimiter: [Inverter 114184216052]     max reduction production/standby: 0/0 W, max increase: 0 W
[09:12:23.162] V (171534256) dynamicPowerLimiter: [Inverter 114184216052]     target limit/output/state: -1 W (unchanged)/0 W/unchanged, 0 update timeouts
[09:12:23.162] V (171534257) dynamicPowerLimiter: [Inverter 114184216052]     MPPTs AC power/DC voltage: a: 0 W/14.8 V b: 0 W/14.8 V
[09:12:23.163] I (171534257) dynamicPowerLimiter: [Inverter 1164a00ba154] sending limit of 100.0 % (2000 W respectively), max output is 2000 W
[09:12:23.163] D (171534261) hoymiles: TX ActivePowerControl (01) 868.25 MHz --> 51 A0 0B A1 54 80 14 47 32 81 0B 00 03 E8 00 01 D0 40 1E
[09:12:23.163] I (171534294) dynamicPowerLimiter: [Controller] waiting for a start/stop/restart/limit command to complete
[09:12:23.262] V (171534329) hoymiles: Interrupt received
[09:12:23.262] D (171534332) hoymiles: RX 868.25 MHz --> D1 A0 0B A1 54 80 14 47 32 81 00 00 0B 00 14 07 F7 | -53 dBm
[09:12:23.262] I (171534511) hoymiles: RX Period End
[09:12:23.595] W (171534512) hoymiles: All missing
[09:12:23.595] W (171534512) hoymiles: Nothing received, resend whole request
[09:12:23.595] D (171534513) hoymiles: TX RealTimeRunData Channel: 40 --> 15 84 21 60 52 80 14 47 32 80 0B 00 68 C5 19 55 00 00 00 00 00 00 00 00 B3 3F 85
[... 497 lines cut ...]
[09:12:42.129] D (171553428) hoymiles: TX SystemConfigPara Channel: 75 --> 15 84 21 60 52 80 14 47 32 80 05 00 68 C5 19 66 00 00 00 00 00 00 00 00 8C 70 C8
[09:12:42.336] I (171553549) hoymiles: RX Period End
[09:12:42.336] W (171553549) hoymiles: Middle missing
[09:12:42.336] I (171553549) hoymiles: Request retransmit: 4
[09:12:42.336] D (171553549) hoymiles: TX RequestFrame 868.25 MHz --> 15 A0 0B A1 54 80 14 47 32 84 2E
[09:12:42.336] V (171553622) hoymiles: Interrupt received
[09:12:42.336] D (171553625) hoymiles: RX 868.25 MHz --> 95 A0 0B A1 54 80 14 47 32 04 00 D5 09 3C 13 87 2F DD 00 0A 02 06 03 E7 00 B2 F0 | -52 dBm
[09:12:42.336] I (171553667) hoymiles: RX Period End
[09:12:42.336] W (171553667) hoymiles: All missing
[09:12:42.336] W (171553667) hoymiles: Nothing received, resend whole request
[09:12:42.603] D (171553668) hoymiles: TX SystemConfigPara Channel: 3 --> 15 84 21 60 52 80 14 47 32 80 05 00 68 C5 19 66 00 00 00 00 00 00 00 00 8C 70 C8
[09:12:42.604] I (171553705) hoymiles: RX Period End
[09:12:42.604] I (171553706) hoymiles: Success
[09:12:42.604] I (171553907) hoymiles: RX Period End
[09:12:42.604] W (171553907) hoymiles: All missing
[09:12:42.604] W (171553907) hoymiles: Nothing received, resend count exeeded
[09:12:42.604] I (171554007) hoymiles: Fetch inverter: 114184216052
```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

AndreasBoehm, 2025-09-13:
Hallo @KarlHeinzz1,

das sollte nicht passieren.

Kannst du es mal mit der Version https://github.com/hoylabs/OpenDTU-OnBattery/actions/runs/17684022077 probieren?

Eventuell ist es damit schon gelöst.

AndreasBoehm, 2025-09-16:
Vielen Dank @KarlHeinzz1

closed 2025-09-16 by AndreasBoehm