hoylabs/OpenDTU-OnBattery #2244: [Request] Ändern der DPL Mode in Abhängigkeit des SoC

hans61, 2025-09-24:
### Is your feature request related to a problem? Please describe.

Das Problem besteht beim zusammenwirken von OpenDTU-OnBattery mit einem unabhängig davon arbeitendem AC gekoppelten Akkusystem (Azzurro 3000SP (=SOFAR Solar ME 3000-SP) + 2x Pylontech (US300C+US5000)), welches solange der Akku nicht voll/leer ist versucht auf einen 0 Bezug/Einspeisung aus dem Netz zu regeln.

### Describe the solution you'd like

Mein Wunsch ist bei SoC >=95% DPL Mode 0 und bei <90% DPL Mode. Wobei wahrscheinlich einstellbare Werte sinnvoll wären. Die oben genannten sind meine Erfahrungswerte mit dem Azzurro 3000SP.

### Describe alternatives you've considered

Meine alternative ist aktuell die Umsetzung obiger Funktion mit meinem Smarthome Server (Raspberry Pi + FHEM und MQTT-Broker)

In FHEM:
```
define DPL_ON DOIF ([OpenDTU_10716304:mode] eq 2 and [OpenDTU_10716304:stateOfCharge] > 95) (set OpenDTU_10716304 0)

define DPL_OFF DOIF ([OpenDTU_10716304:mode] eq 0 and [OpenDTU_10716304:stateOfCharge] < 90) (set OpenDTU_10716304 2)
```

Capteinjack, 2026-08-27:
Hallo zusammen,
Genauso ist es bei mir auch!

Ich habe zwei Hoymiles Wechselrichter, die mit der DTU einwandfrei geregelt werden. Sobald allerdings mein Akku voll ist, es ist ein Zendure4000 AC+ , müsste die DTUshop.de mit aktiviertem DPL bei noch vorhandener PV Energie soweit runter regeln, dass der Grundverbrauch des Hauses noch gedeckelt ist. Solange eben Energie da ist.  Toll wäre natürlich, wenn dann mit Schwinden der Sonne der Akku aus der Reserve langsam kommt.

Im Moment ist es so das mit aktiviertem DPL bei vollem Akku, eine quasi null Einspeisung stattfindet. Obwohl Hausseitig noch Energie benötigt würde.

Die Idee mit -800 W als Zielwert einzutragen, werde ich mal ausprobieren!

Schönen Abend allerseits

AndreasBoehm, 2026-08-28:
> Im Moment ist es so das mit aktiviertem DPL bei vollem Akku, eine quasi null Einspeisung stattfindet. Obwohl Hausseitig noch Energie benötigt würde.

Das verstehe ich nicht. Wenn der DPL regelt, dann wird der DPL den Zähler auf deinen Konfigurierten Netzbezug regeln. Licht an, Wechselrichter Limit wird erhöht, Licht aus, Wechselrichter Limit wird reduziert.

Starfoxfs, 2026-08-28:
Ich habe z.b. eine Hoymiles AC1920 Hibattery da funktioniert das nicht mit dem DPL in herkömmlicher Weise.

Auch nicht wenn ich als Stromquelle eine Smart Buffer Batterie angebe.

Der Ladezustand der Batterie (SOC) wird nicht mit einbezogen, was bedeutet wenn ich DPL aktiviere wird auf 0 Watt geregelt aber die Batterie bekommt keinen Strom ab weil der DPL einfach nicht weiß das eine Batterie vorhanden ist bzw. welchen Ladezustand diese hat.

Toll wäre wie oben schon geschrieben eine Funktion im DPL die den SOC abfragt und bei 100% SOC dann erst auf 0 Watt Nulleinspeisung regelt, eben in Abhängigkeit vom SOC Wert die DPL regelt.

AndreasBoehm, 2026-08-28:
Der AC1920 wird auf 0W regeln, dann stellst du den DPL auf -10W und schon ist das Problem gelöst. Der AC1920 sieht das es Überschuss gibt, beginnt zu laden und der DPL regelt immer weiter nach.

hans61, 2026-08-28:
Als Initiator möchte ich mich auch noch einmal dazu melden. DPL auf -10W würde ich für zuwenig halten, das hängt etwas vom Messsystem ab. Bei einem Shelly würde ich die Messgenauigkeit auf ca. +/- 30 W schätzen.
Seit einem Jahr wechsle ich die Betriebsarten über meinen Smarthome-Server (FHEM). Bei >=97% SOC schalte den DPL ein und bei <=95% wieder aus. Damit bin ich sehr zufrieden.
Wenn der DPL aus ist regelt mein Azzurro 3000SP bei konstantem Sonnenschein Lastschwankungen in <30 Sekunden nach. Bei eingeschaltetem DPL (Firmware 2025.10.11, DPL auf -150W) braucht das Gesamtsystem mehrere Minuten (bis 5Min), bis es sich wieder eingeschwungen hat (konstante Last, konstante Sonne, noch keine schnell wechselnden Bedingungen).
Ich habe 5 Hoymiles Wechselrichter, wenn jetzt mehr Leistung benötigt wird, dann regelt der DPL auch nicht alle 5 Wechselrichter gleichzeitig hoch, sondern schön "langsem" einen nach dem anderen. Das dauert mir bei leerem Akku einfach zulange.
Im übrigen schließe ich mich hier auch der Meinung von @httpneo an. Ich möchte dem nächst auch von FHEM auf Home Assistant umsteigen.
Aus meiner Sicht wäre es schön, wenn OpenDTU-OnBattery diesen Part übernehmen könnte und ich nicht noch einen weiteren Mitspieler benötige.
Was ist eigentlich das Problem einen einfachen Schwellwertschalter für den DPL einzubauen?

AndreasBoehm, 2026-08-28:
> Was ist eigentlich das Problem einen einfachen Schwellwertschalter für den DPL einzubauen?

Zeit und andere Prioritäten.
DC Speicher stehen bei diesem Projekt im Vordergrund. Für diese bedarf es einer anderen Regelung.

Starfoxfs, 2026-08-28:
Ja aber was passiert wenn SOC 100% ist wenn der DPL auf Beispielsweise -10W steht werden ab SOC 100% dauerhaft -10W eingespeist was die eigentlich DPL Regelung nämlich Nulleinspeisung ja auch wieder überflüssig macht.

Oder bin ich da falsch der Annahme ?

Dann könnte man den DPL auch dauerhaft deaktiviert lassen die Batterie regelt das sowieso zumindest bis 100% und dann fängt halt wieder die volle Netzeinspeisung an (was ich ja nicht haben will).

Die anderen Prioritäten gehen natürlich vor.

AndreasBoehm, 2026-08-28:
Korrekt, es werden 10W eingespeist wenn der Akku nichts mehr abnimmt.

Einspeisung kannst du nicht vermeiden, egal was du einstellst, jegliche Regelung wird zu langsam sein. Also warum nicht 10W einspeisen und gut is.

spcqike, 2026-08-28:
> Bei >=97% SOC schalte den DPL ein und bei <=95% wieder aus.

Setze den DPL per MQTT lieber in den passenden Mode (0, 1 oder 2) 

An und aus schalten, wie es über die Web Oberfläche und APi geschieht, erzeugt schreibzugriffe auf dem eeprom

Starfoxfs, 2026-08-29:
Naja Schreibzugriffe auf den eeprom dann kommt irgendwann einfach ein neuer ESP32-S3 rein und gut. 

Finde die Weboberfläche komfortabler und dafür ist se ja auch da.

Harisjems, 2026-08-29:
Es  betrifft aber  Flash von WR😉
ESP gibt nur Befehl.

Starfoxfs, 2026-08-29:
Heißt als Klartext wenn ich den DPL über die Weboberfläche normal aktiviere wird bei jeder Setzung des Limits durch diesen am Wechselrichter der Flashspeicher des Wechselrichter belastet ?

Starfoxfs, 2026-08-29:
Naja also Mode 2 ist genauso wie wenn ich den DPL abschalte und den Inverter auf 100% setze.

Bringt keinen Vorteil wenn die Smartbuffer Batterie 100% hat wird der Rest des Solarertrags voll eingespeist.

Mode 1 ist Inverter aus

Mode 0 ist Standard nach Stromzähler regeln. Was dann den Flash Speicher des Inverters belastet (wenn das so ist) ?!?

Also ist damit nix gewonnen, man kann den DPL komplett abschalten dann wird die AC Batterie von 10%-100% geladen und der Rest geht halt ins Netz....

Harisjems, 2026-08-29:
Ich denke, ich habe gelogen, was Flash von WR betrifft; ich habe das falsch verstanden.😉

Es ist manchmal erforderlich oder möglich, sich damit zu arrangieren, dass nicht alles in einem bestimmten Setup geregelt werden kann.Ja, es wäre nötig gewesen, sich im Vorfeld Gedanken zu machen.
Oder wenn, dann ist der Aufwand so groß, dass es für mich persönlich wie verlorene Zeit erscheint.
HA! Automatisierungen, Skripte usw. Hier spricht man derzeit von einem Einspeisung von 90-120 kWh pro Jahr, den ich verschenke beim 14KWp.
5 min .Vollgas auf Autobahn mit aktuellen Sprit peisen😁 als Vergleich.
Zb.bei Marstek Venus A und Überschuss / Einspeisung , Aktuell.
Wenn der Überschuss mehr als 200 W beträgt, Länge als 3 min.schalte ich Verbraucher X über Tasmota Leserkopf ein. Das ist die einfache Lösung, die ich umsetze.
Wenn der Zukauf länger als 3 Minuten über 400 W ist, schalte diesen Verbraucher wieder über Tasmota aus.

SW-Niko, 2026-08-30:
Ich habe einen Energie Optimierer für DC-Batterie Systeme in der Erprobungsphase.
Der kann unter anderem auch den Bezug in Abhängigkeit von dem SoC Wert regeln (4 Stufen).

>Aus meiner Sicht wäre es schön, wenn OpenDTU-OnBattery diesen Part übernehmen könnte und ich nicht noch einen weiteren Mitspieler benötige. Was ist eigentlich das Problem einen einfachen Schwellwertschalter für den DPL einzubauen?

Aber mir geht es wie Andreas 😞 

>Zeit und andere Prioritäten.

<br>

[image]