hoylabs/OpenDTU-OnBattery #2250: Update to 2025.09.16: OpenDTU don't get Data

dAjaY85, 2025-09-29:
### What happened?

After Update to 2025.09.16: OpenDTU don't update Data with a uptime of 2 days+

Communication to Inverter (nRF24), Pylontech (CAN) an Victron is not working.

To solve this issue, I need to power cycle OpenDTU.

Maybe a memory issue?

### To Reproduce Bug

it comes after some time

### Expected Behavior

openDTU shoul communicate with other components

[image]

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.09.16

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Anything else?

[image]

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

dAjaY85, 2025-11-28:
Moin,
hab jetzt ein Log am PC erstellen können. Sieht nach einem Problem aus, wenn der Stromwert mittels MQTT übermittelt wird.
Dynamic Power Limiter, Victron und Pylontech sind abgeschaltet und deaktiviert.

Der Stromzähler ist folgendermaßen eingerichtet:

[image]

 
[OpenDTU-Log.log](https://github.com/user-attachments/files/23818971/OpenDTU-Log.log)

SW-Niko, 2026-08-31:
Hallo @dAjaY85 

Ich sehe in deinem Log das du hintereinander 2 Commandos sendest. Das könnte genau in den deadlock laufen den ich beschrieben habe. 🤔 

_D (5655213) mqtt: Received MQTT message on topic 'solar/powerlimiter/cmd/threshold/soc/stop' (Bytes 1-2/2)
D (5655440) mqtt: Received MQTT message on topic 'solar/powerlimiter/cmd/threshold/soc/start' (Bytes 1-2/2)_

dAjaY85, 2026-09-24:
Naja, mann muss ja beide SOC Werte ändern, ansonsten bei änderung des start Wertes, wird die Solarproduktion angeschmissen und stoppt nicht mehr zum gewünschten Zeitpunkt.

Hab aber meine Regelung etwas umgebaut und nutze jetzt "solar/powerlimiter/cmd/mode" damit funktioniert es auch ziemlich gut und ich muss die SOC Werte nicht anpassen.

Wenn es interessiert, nutze diese Regelung:
```
alias: OpenDTU DPL Preissteuerung
triggers:
  - entity_id: sensor.tibber_price_level
    trigger: state
  - entity_id: sensor.pylon_state_of_charge_soc
    trigger: state
actions:
  - choose:
      - conditions:
          - condition: numeric_state
            entity_id: sensor.pylon_state_of_charge_soc
            above: 80
        sequence:
          - action: mqtt.publish
            data:
              topic: solar/powerlimiter/cmd/mode
              payload: '0'
      - conditions:
          - condition: state
            entity_id: sensor.tibber_price_level
            state: NORMAL
          - condition: numeric_state
            entity_id: sensor.pylon_state_of_charge_soc
            below: 50
        sequence:
          - action: mqtt.publish
            data:
              topic: solar/powerlimiter/cmd/mode
              payload: '1'
      - conditions:
          - condition: state
            entity_id: sensor.tibber_price_level
            state:
              - CHEAP
              - VERY_CHEAP
        sequence:
          - action: mqtt.publish
            data:
              topic: solar/powerlimiter/cmd/mode
              payload: '1'
    default:
      - action: mqtt.publish
        data:
          topic: solar/powerlimiter/cmd/mode
          payload: '0'
mode: restart
```

AndreasBoehm, 2026-10-01:
Closing this as a duplicate of #1629. All of these reports have the same cause: a deadlock in the configuration `WriteGuard` when two config writes (MQTT, Home Assistant or the web UI) overlap. The DTU then stops polling the inverter, battery and charger, while the web UI stays reachable.

Please follow #1629 for the fix and its release status. If you still see the freeze on a build that includes the fix, please comment there.

closed 2026-10-01 by AndreasBoehm